Koenkk/zigbee2mqtt #8915: Inoperative devices

jfc715, 2021-09-29:
Hi.

At first, i am very rookie with zigbee or home assistant. I am not even sure am i in right place to ask these things, hope so.

Almost every day it lost control of devices. Sometimes all, sometimes just a few. Most of the time it works really laggy. At the moment it was first laggy, and now nothing works. Usually it fixes by removing the power cable from Raspberry, soft reboot from hassio wont help. 

Last autumn i give up and about month ago i´ve had again interest to look whats wrong and i found out that the mqtt broker was dead (Red led) so i ordered new one and configure everything from the beginning. Everything worked great, but just for a few hours...

**Info about the mqtt usb-stick:**
[image]

**Log from zigbee2mqtt:**
Zigbee2MQTT:info  2021-09-29 19:55:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Poreal. C GU10' failed: 'Error: Command 0x588e81fffe15694b/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"Poreal. C GU10"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:info  2021-09-29 20:09:18: Starting network scan (includeRoutes 'false')
Zigbee2MQTT:error 2021-09-29 20:09:29: Failed to execute LQI for 'Coordinator'
Zigbee2MQTT:error 2021-09-29 20:09:40: Failed to execute LQI for 'MH Pöytävalo vas.'
Zigbee2MQTT:error 2021-09-29 20:09:51: Failed to execute LQI for 'MH Pöytävalo Oik.'
Zigbee2MQTT:error 2021-09-29 20:10:02: Failed to execute LQI for 'OH Peili Lednauha'
Zigbee2MQTT:error 2021-09-29 20:10:13: Failed to execute LQI for 'Zigbee repeater'
Zigbee2MQTT:error 2021-09-29 20:10:24: Failed to execute LQI for 'Ulkorak. MH GU10'
Zigbee2MQTT:error 2021-09-29 20:10:35: Failed to execute LQI for 'Ulkorak. MAIN 1-2-3'
Zigbee2MQTT:error 2021-09-29 20:10:46: Failed to execute LQI for 'Poreal. B GU10'
Zigbee2MQTT:error 2021-09-29 20:10:57: Failed to execute LQI for 'Poreal. A GU10'
Zigbee2MQTT:error 2021-09-29 20:11:08: Failed to execute LQI for 'Poreal. C GU10'
Zigbee2MQTT:error 2021-09-29 20:11:19: Failed to execute LQI for 'Ulkorak. Lednauha 1'
Zigbee2MQTT:error 2021-09-29 20:11:30: Failed to execute LQI for 'Ulkorak. MH Lednauha'
Zigbee2MQTT:error 2021-09-29 20:11:41: Failed to execute LQI for 'Ulkorak. Lednauha 2'
Zigbee2MQTT:error 2021-09-29 20:11:52: Failed to execute LQI for 'Ulkorak. MH valokatk.'
Zigbee2MQTT:error 2021-09-29 20:12:03: Failed to execute LQI for 'Saareke Lednauha'
Zigbee2MQTT:error 2021-09-29 20:12:14: Failed to execute LQI for 'Välitila Lednauha'
Zigbee2MQTT:error 2021-09-29 20:12:25: Failed to execute LQI for 'Aqara kytkin A'
Zigbee2MQTT:error 2021-09-29 20:12:36: Failed to execute LQI for 'Aqara kytkin B'
Zigbee2MQTT:info  2021-09-29 20:12:36: Network scan finished
Zigbee2MQTT:info  2021-09-29 20:12:36: MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{"routes":false,"type":"raw","value":{"links":[],"nodes":[{"definition":null,"failed":["lqi"],"friendlyName":"Coordinator","ieeeAddr":"0x00124b001cde52ea","lastSeen":1609143040835,"networkAddress":0,"type":"Coordinator"},{"definition":{"description":"TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white","model":"LED1733G7","supports":"light (state, brightness, color_temp, color_temp_startup), effect, linkquality","vendor":"IKEA"},"failed":["lqi"],"friendlyName":"MH Pöytävalo vas.","ieeeAddr":"0x588e81fffe117eec","lastSeen":1632845161616,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI bulb E14 WS opal 600lm","networkAddress":21019,"type":"Router"},{"definition":{"description":"TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white","model":"LED1733G7","supports":"light (state, brightness, color_temp, color_temp_startup), effect, linkquality","vendor":"IKEA"},"failed":["lqi"],"friendlyName":"MH Pöytävalo Oik.","ieeeAddr":"0x680ae2fffef06d7b","lastSeen":1632250414965,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI bulb E14 WS opal 600lm","networkAddress":15579,"type":"Router"},{"definition":{"description":"Smart light switch - 1 gang","model":"TS0011","supports":"switch (state), linkquality","vendor":"TuYa"},"friendlyName":"KHH valokatk.","ieeeAddr":"0xbc33acfffef4d2c9","lastSeen":1632932203589,"manufacturerName":"_TZ3000_9hpxg80k","modelID":"TS0011","networkAddress":46145,"type":"EndDevice"},{"definition":{"description":"Zigbee LED Controller RGB+CCT (1 ID)","model":"GL-C-008-1ID","supports":"light (state, brightness, color_temp, color_temp_startup, color_xy), effect, linkquality","vendor":"Gledopto"},"failed":["lqi"],"friendlyName":"OH Peili Lednauha","ieeeAddr":"0x00124b001d110579","lastSeen":1632889673467,"manufacturerName":"GLEDOPTO","modelID":"GL-C-008","networkAddress":31117,"type":"Router"},{"definition":{"description":"Smart light switch - 3 gang without neutral wire","model":"TS0013","supports":"switch (state), linkquality","vendor":"TuYa"},"friendlyName":"Terassi MAIN. 1-2-3","ieeeAddr":"0xec1bbdfffe66f852","lastSeen":1632928643775,"manufacturerName":"_TYZB01_xiuox57i","modelID":"TS0013","networkAddress":62236,"type":"EndDevice"},{"definition":{"description":"TRADFRI signal repeater","model":"E1746","supports":"linkquality","vendor":"IKEA"},"failed":["lqi"],"friendlyName":"Zigbee repeater","ieeeAddr":"0x588e81fffe61c8eb","lastSeen":1632848859780,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI Signal Repeater","networkAddress":49669,"type":"Router"},{"definition":{"description":"TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum","model":"LED1537R6/LED1739R5","supports":"light (state, brightness, color_temp, color_temp_startup), effect, linkquality","vendor":"IKEA"},"failed":["lqi"],"friendlyName":"Ulkorak. MH GU10","ieeeAddr":"0x588e81fffe00e879","lastSeen":1632677189214,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI bulb GU10 WS 400lm","networkAddress":26660,"type":"Router"},{"definition":{"description":"Smart light switch - 3 gang with neutral wire","model":"ZM-L03E-Z","supports":"switch (state), linkquality","vendor":"Zemismart"},"failed":["lqi"],"friendlyName":"Ulkorak. MAIN 1-2-3","ieeeAddr":"0xbc33acfffe5010c4","lastSeen":1632933819866,"manufacturerName":"_TYZB01_b8cr31hp","modelID":"TS0003","networkAddress":39416,"type":"Router"},{"definition":{"description":"TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum","model":"LED1537R6/LED1739R5","supports":"light (state, brightness, color_temp, color_temp_startup), effect, linkquality","vendor":"IKEA"},"failed":["lqi"],"friendlyName":"Poreal. B GU10","ieeeAddr":"0x588e81fffe33d1f0","lastSeen":1632931675054,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI bulb GU10 WS 400lm","networkAddress":20069,"type":"Router"},{"definition":{"description":"TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum","model":"LED1537R6/LED1739R5","supports":"light (state, brightness, color_temp, color_temp_startup), effect, linkquality","vendor":"IKEA"},"failed":["lqi"],"friendlyName":"Poreal. A GU10","ieeeAddr":"0x588e81fffe0317e8","lastSeen":1632933819931,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI bulb GU10 WS 400lm","networkAddress":14691,"type":"Router"},{"definition":{"description":"TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum","model":"LED1537R6/LED1739R5","supports":"light (state, brightness, color_temp, color_temp_startup), effect, linkquality","vendor":"IKEA"},"failed":["lqi"],"friendlyName":"Poreal. C GU10","ieeeAddr":"0x588e81fffe15694b","lastSeen":1632931674598,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI bulb GU10 WS 400lm","networkAddress":37638,"type":"Router"},{"definition":{"description":"Zigbee LED Controller WW/CW","model":"GL-C-006","supports":"light (state, brightness, color_temp, color_temp_startup), effect, linkquality","vendor":"Gledopto"},"failed":["lqi"],"friendlyName":"Ulkorak. Lednauha 1","ieeeAddr":"0x00124b002114cd84","lastSeen":1632677182547,"manufacturerName":"GLEDOPTO","modelID":"GL-C-006","networkAddress":35670,"type":"Router"},{"definition":{"description":"Zigbee LED Controller RGB+CCT (1 ID)","model":"GL-C-008-1ID","supports":"light (state, brightness, color_temp, color_temp_startup, color_xy), effect, linkquality","vendor":"Gledopto"},"failed":["lqi"],"friendlyName":"Ulkorak. MH Lednauha", [... cut ...]
**List of devices (Unsupported devices are Aqara Opple):**
[image]

## How to reproduce it (minimal and precise)
No idea, this time it went down when i tried to add Aqara Opple switches (WXCJKG13LM, for some reason these are not work either, just saying Unsupported). In reality, there is no logic that causes the fault. The fault may also go away on its own but usually not. 

## Debug info
Zigbee2MQTT version:  1.18.1-1
Adapter hardware: CC2531
Adapter firmware version: Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20201127.hex (Stand 3/2021)

jfc715, 2021-09-29:
Additional screenshot from Zigbee net after rebooting Raspberry. Now everything works except few lagging devices. 
[image]

Log:

Zigbee2MQTT:info  2021-09-29 21:50:16: MQTT publish: topic 'zigbee2mqtt/Ulkorak. MAIN 1-2-3', payload '{"linkquality":65,"state_center":"OFF","state_left":"ON","state_right":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:17: MQTT publish: topic 'zigbee2mqtt/Terassi MAIN. 1-2-3', payload '{"linkquality":63,"state_center":"OFF","state_left":"OFF","state_right":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:17: MQTT publish: topic 'zigbee2mqtt/Terassi MAIN. 1-2-3', payload '{"linkquality":65,"state_center":"OFF","state_left":"OFF","state_right":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:19: MQTT publish: topic 'zigbee2mqtt/Ulkorak. MAIN 1-2-3', payload '{"linkquality":65,"state_center":"ON","state_left":"ON","state_right":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:19: MQTT publish: topic 'zigbee2mqtt/Terassi MAIN. 1-2-3', payload '{"linkquality":63,"state_center":"OFF","state_left":"ON","state_right":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:19: MQTT publish: topic 'zigbee2mqtt/Terassi MAIN. 1-2-3', payload '{"linkquality":63,"state_center":"OFF","state_left":"ON","state_right":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:20: MQTT publish: topic 'zigbee2mqtt/Ulkorak. MAIN 1-2-3', payload '{"linkquality":65,"state_center":"OFF","state_left":"ON","state_right":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:26: MQTT publish: topic 'zigbee2mqtt/Ulkorak. MAIN 1-2-3', payload '{"linkquality":65,"state_center":"ON","state_left":"ON","state_right":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:26: MQTT publish: topic 'zigbee2mqtt/Ulkorak. MAIN 1-2-3', payload '{"linkquality":63,"state_center":"ON","state_left":"ON","state_right":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Poreal. B GU10","ieee_address":"0x588e81fffe33d1f0"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-09-29 21:50:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Poreal. B GU10"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-09-29 21:50:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Poreal. A GU10","ieee_address":"0x588e81fffe0317e8"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-09-29 21:50:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Poreal. A GU10"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-09-29 21:50:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Poreal. C GU10","ieee_address":"0x588e81fffe15694b"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-09-29 21:50:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Poreal. C GU10"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-09-29 21:50:29: MQTT publish: topic 'zigbee2mqtt/Saareke Lednauha', payload '{"brightness":254,"color":{"x":0.419534951958697,"y":0.397341851333969},"color_temp":306,"state":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:30: MQTT publish: topic 'zigbee2mqtt/Välitila Lednauha', payload '{"brightness":254,"color":{"x":0.526676280311873,"y":0.41329727450763},"color_temp":500,"state":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:31: MQTT publish: topic 'zigbee2mqtt/Saareke Lednauha', payload '{"brightness":254,"color":{"x":0.419534951958697,"y":0.397341851333969},"color_temp":306,"state":"ON"}'
Zigbee2MQTT:info  2021-09-29 21:50:31: MQTT publish: topic 'zigbee2mqtt/Välitila Lednauha', payload '{"brightness":254,"color":{"x":0.526676280311873,"y":0.41329727450763},"color_temp":500,"state":"ON"}'
Zigbee2MQTT:info  2021-09-29 21:50:51: MQTT publish: topic 'zigbee2mqtt/Ulkorak. Lednauha 1', payload '{"brightness":254,"color_temp":250,"linkquality":63,"state":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:52: MQTT publish: topic 'zigbee2mqtt/Ulkorak. Lednauha 2', payload '{"linkquality":63,"state":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:52: MQTT publish: topic 'zigbee2mqtt/Ulkorak. MH valokatk.', payload '{"linkquality":63,"state":"OFF"}'
Zigbee2MQTT:info  2021-09-29 21:50:55: MQTT publish: topic 'zigbee2mqtt/Ulkorak. Lednauha 1', payload '{"brightness":254,"color_temp":250,"linkquality":65,"state":"ON"}'
Zigbee2MQTT:info  2021-09-29 21:50:55: MQTT publish: topic 'zigbee2mqtt/Ulkorak. Lednauha 2', payload '{"linkquality":63,"state":"ON"}'
Zigbee2MQTT:info  2021-09-29 21:50:56: MQTT publish: topic 'zigbee2mqtt/Ulkorak. MH valokatk.', payload '{"linkquality":63,"state":"ON"}'
Zigbee2MQTT:info  2021-09-29 21:50:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Ulkorak. MH GU10","ieee_address":"0x588e81fffe00e879"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-09-29 21:50:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Ulkorak. MH GU10"},"type":"device_announced"}'

r-vit, 2021-10-01:
that looks like a lot of devices for a CC2531 stick, maybe that's the culprit?

jfc715, 2021-10-04:
Any recommendation for better stick? How can i figure out is there too much devices?
About 5 days from last reboot and today working "somehow" but right now nothing works. About hour ago backyard lights are turned on by automation.
Strange. And much more devices is coming... 

Map:
[image]

**Log at the moment:**
(node:329) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1367)
(node:329) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":238,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,1,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
(node:329) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1368)
(node:329) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":239,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,2,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
(node:329) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1369)
(node:329) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":240,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,3,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
(node:329) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1370)
(node:329) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":241,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,4,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
(node:329) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1371)
(node:329) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":242,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,5,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
[... cut ...]
Zigbee2MQTT:info  2021-10-04 21:28:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Ulkorak. Lednauha 1' failed: 'Error: Command 0x00124b002114cd84/11 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"Ulkorak. Lednauha 1"},"type":"zigbee_publish_error"}'

r-vit, 2021-10-04:
here's you can check the info about the coordinators limits: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

personally, I've got the one based on CC2652R chip, it works pretty well with about 60 devices (keep in mind that there's quite a few devices which can be routers which helps with the big networks). I saw recently Sonoff presented a device based on that chip, so if I didn't have mine I would have definitely bought that one: https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/ - for that price you can't find anything comparable.

jfc715, 2021-11-03:
Hi.

Just arrived one of Sonoff 3.0 Dongle Plus, but only problem is that i cant get in configured.

First ZHA with Sonoff instructions, no luck. Then Z-wave JS and finally zigbee2mqtt but no.
My Rasp is now booting from usb-hdd and totally fresh install. 

[image]

**Log from zigbee2mqtt:**

panID or channel
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:172:23
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.18.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-11-02T22_17_00_975Z-debug.log
[00:17:01] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.18.1 start /app
> node index.js
Zigbee2MQTT:error 2021-11-03 11:44:41: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'SRSP - AF - interPanCtl after 6000ms'
Zigbee2MQTT:error 2021-11-03 11:44:53: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'Touchlink operation already in progress'
Zigbee2MQTT:error 2021-11-03 11:44:59: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'Touchlink operation already in progress'
Zigbee2MQTT:error 2021-11-03 11:45:51: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Cannot execute command, in Inter-PAN mode'
Zigbee2MQTT:error 2021-11-03 11:45:53: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Cannot execute command, in Inter-PAN mode'
Zigbee2MQTT:error 2021-11-03 11:45:57: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Cannot execute command, in Inter-PAN mode'
Zigbee2MQTT:error 2021-11-03 11:46:00: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Cannot execute command, in Inter-PAN mode'
Zigbee2MQTT:error 2021-11-03 12:00:17: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Cannot execute command, in Inter-PAN mode'
Zigbee2MQTT:error 2021-11-03 12:00:19: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Cannot execute command, in Inter-PAN mode'
Zigbee2MQTT:error 2021-11-03 12:00:47: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Cannot execute command, in Inter-PAN mode'
Zigbee2MQTT:error 2021-11-03 12:04:29: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Cannot execute command, in Inter-PAN mode'

jfc715, 2021-11-03:
For some reason, after restart zigbee2mqtt not giving any errors anymore, nevertheless its not discovering any devices.

[image]

Is that new Sonoff dongle broken or am i missing something?

**Another log from ZHA:**

Logger: homeassistant.components.zha.core.gateway
Source: components/zha/core/gateway.py:152
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 15.40.54 (1 occurrences)
Last logged: 15.40.54

Couldn't start ZNP = Texas Instruments Z-Stack ZNP protocol: CC253x, CC26x2, CC13x2 coordinator
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 395, in request
    response = await response_future
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 152, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 69, in new
    await app.startup(auto_form)
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 196, in startup
    return await self._startup(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 209, in _startup
    await znp.connect()
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 116, in connect
    self.capabilities = (await self.request(c.SYS.Ping.Req())).Capabilities
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 395, in request
    response = await response_future
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 55, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 92, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

closed 2021-12-12 by github-actions[bot]