Koenkk/zigbee2mqtt #8920: Can't discover any devices

thehiddentruth, 2021-09-30:
I Installed Mosquitto and Zigbee2MQTT and using C2531 Adapter. But I can't see any devices on the network. The CC2531 green light in constant on. I start the system by running the "npm start" command, but I can't see any available Zigbee devices. I have a Aqara Mini Wireless, which i make enter the "pairing mode" but still don't discover it.

## Debug Info
Zigbee2MQTT version: 1.21.1
Adapter hardware: CC2531
Adapter firmware version: No idea. According to the seller it should be flashed with "latest firmware version". Is there a way to check this without CC debugger?

danieledwardgeorgehitchcock, 2021-09-30:
Hi,

Have have you added the device to your configuration.yaml and set up your MQTT client?

You can find instructions on how to do that [here](https://www.zigbee2mqtt.io/information/configuration.html)

Please make sure that all of the mandatory entries are populated, your MQTT server is accepting connections, and your device is accessible by Zigbee2MQTT.

thehiddentruth, 2021-09-30:
**This is what confirmationguration.yaml looks like**
`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.141.184:1883
  user: MQTT
  password: admin
serial:
  port: /dev/ttyACM0
\n\nadvanced:\n    network_key: GENERATE
frontend:
  port: 8080
  host: 0.0.0.0
advanced:
  channel: 11
  log_level: debug`

I don't know what to do else. I followed this exact guide on [Zigbee2MQTT](https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html)

danieledwardgeorgehitchcock, 2021-09-30:
You’ve got the advanced key declared twice and a bunch of new line characters which likely wouldn’t help

thehiddentruth, 2021-09-30:
So multiple things

### First try
**When my configuration.yaml looks like this:**
`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.141.184:1883
  user: MQTT
  password: admin
serial:
  port: /dev/ttyACM0
frontend:
  port: 8080
  host: 0.0.0.0
`

**I get this in the terminal:** 
`Zigbee2MQTT:info  2021-09-30 15:29:00: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-09-30.15-28-59' filename: log.txt
Zigbee2MQTT:info  2021-09-30 15:29:00: Starting Zigbee2MQTT version 1.21.1 (commit #4a51e0c0)
Zigbee2MQTT:info  2021-09-30 15:29:00: Starting zigbee-herdsman (0.13.138)
Zigbee2MQTT:info  2021-09-30 15:29:02: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2021-09-30 15:29:02: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:info  2021-09-30 15:29:02: Currently 0 devices are joined:
Zigbee2MQTT:warn  2021-09-30 15:29:02: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2021-09-30 15:29:02: Allowing new devices to join.
Zigbee2MQTT:warn  2021-09-30 15:29:02: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2021-09-30 15:29:02: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2021-09-30 15:29:02: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2021-09-30 15:29:02: Connecting to MQTT server at mqtt://192.168.141.184:1883
Zigbee2MQTT:info  2021-09-30 15:29:02: Connected to MQTT server
Zigbee2MQTT:info  2021-09-30 15:29:02: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2021-09-30 15:29:03: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"4a51e0c0","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.21.1"}'
`

### Second try
**When I add the advanced network generate:**
`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.141.184:1883
  user: MQTT
  password: admin
serial:
  port: /dev/ttyACM0
frontend:
  port: 8080
  host: 0.0.0.0
advanced:
  network_key: GENERATE 
`

I can't get to start the service when I run npm start. I get this error log:

`> zigbee2mqtt@1.21.1 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2021-09-30 15:35:00: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-09-30.15-34-59' filename: log.txt
Zigbee2MQTT:info  2021-09-30 15:35:00: Starting Zigbee2MQTT version 1.21.1 (commit #4a51e0c0)
Zigbee2MQTT:info  2021-09-30 15:35:00: Starting zigbee-herdsman (0.13.138)
Zigbee2MQTT:error 2021-09-30 15:35:02: Configuration is not consistent with adapter state/backup!
Zigbee2MQTT:error 2021-09-30 15:35:02: - PAN ID: configured=6754, adapter=6754
Zigbee2MQTT:error 2021-09-30 15:35:02: - Extended PAN ID: configured=00124b001949d6a8, adapter=dddddddddddddddd
Zigbee2MQTT:error 2021-09-30 15:35:02: - Network Key: configured=25a0c367e94c8746da7c15882cc559d0, adapter=01030507090b0d0f00020406080a0c0d
Zigbee2MQTT:error 2021-09-30 15:35:02: - Channel List: configured=11, adapter=11
Zigbee2MQTT:error 2021-09-30 15:35:02: Please update configuration to prevent further issues.
Zigbee2MQTT:error 2021-09-30 15:35:02: If you wish to re-commission your network, please remove coordinator backup at /opt/zigbee2mqtt/data/coordinator_backup.json.
Zigbee2MQTT:error 2021-09-30 15:35:02: Re-commissioning your network will require re-pairing of all devices!
Zigbee2MQTT:error 2021-09-30 15:35:02: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-09-30 15:35:02: Failed to start zigbee
Zigbee2MQTT:error 2021-09-30 15:35:02: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-09-30 15:35:02: Exiting...
Zigbee2MQTT:error 2021-09-30 15:35:02: Error: startup failed - configuration-adapter mismatch - see logs above for more information
    at ZnpAdapterManager.determineStrategy (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:229:35)
    at ZnpAdapterManager.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:63:26)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.js:66:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.js:110:27)
    at start (/opt/zigbee2mqtt/index.js:95:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.21.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.21.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-09-30T13_35_02_640Z-debug.log`

When I take a look at the configuration.yaml again, now it looks like this: 

`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.141.184:1883
  user: MQTT
  password: admin
serial:
  port: /dev/ttyACM0
frontend:
  port: 8080
  host: 0.0.0.0
advanced:
  network_key:
    - 37
    - 160
    - 195
    - 103
    - 233
    - 76
    - 135
    - 70
    - 218
    - 124
    - 21
    - 136
    - 44
    - 197
    - 89
    - 208`

danieledwardgeorgehitchcock, 2021-09-30:
Ok - It looks like your first try should have worked.

It has connected to the MQTT server, and has generated a network key for you.

I do notice that your coordinator firmware isn't as up-to-date as the seller would have you believe (2019-06-08 Z-Zstack 1.2)

Were you able to join any devices on the first try?

thehiddentruth, 2021-09-30:
No. I press the aqara mini wireless button on the reset button to enter pairing mode, but it still don’t show up as a device.

thehiddentruth, 2021-10-01:
Any ideas on what I can do?

danieledwardgeorgehitchcock, 2021-10-01:
Have you got any other decives other than Aqara? They can be a bit difficult to get connected.

If all you have is Aqara, make sure that once Zigbee2MQTT is in join mode, and your Aqara device is also in join mode, to keep the device awake by periodically pressing the button.

You could also try enabling the Zigbee2MQTT frontend GUI in your settings which would be a lot more user-friendly than the command line.

As a final thing, you could aslo enable debug logging and set it to route to the command line so you can see all messages logged by Zigbee2MQTT

thehiddentruth, 2021-10-01:
I will try with my IKEA Tradfri E27 later tonight, and see if that's any better.

Actually I've already enabled frontend, but still nothing appears in the overview of devices.

Can you please explain on how to "_As a final thing, you could aslo enable debug logging and set it to route to the command line so you can see all messages logged by Zigbee2MQTT_"? A tutorial would also be very good if there's any. Thanks in advance.

danieledwardgeorgehitchcock, 2021-10-01:
All of this can be found in the configuration guidance on the [zigbee2mqtt.io](zigbee2mqtt.io) website. See [Configuration](https://www.zigbee2mqtt.io/information/configuration.html) and [Debug](https://www.zigbee2mqtt.io/information/debug.html)

Under your `advanced` section in your config file, you will need to add the following:

```yaml
  log_level: debug
  log_output:
    - console
    - file
```

thehiddentruth, 2021-10-01:
So I added the following to my configuration.yaml file
`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.141.184:1883
  user: MQTT
  password: admin
serial:
  port: /dev/ttyACM0
frontend:
  port: 8080
  host: 0.0.0.0
advanced:
 log_level: debug
 log_output:
    - console
    - file
`

And started the service again, and still nothing. And I've tried both with IKEA Tradfri bulb and Aqara button, and stil nothing shows up on the frontend or anywhere else... This is what console looks like 

`> zigbee2mqtt@1.21.1 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:debug 2021-10-01 14:44:10: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2021-10-01 14:44:10: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-10-01.14-44-10' filename: log.txt
Zigbee2MQTT:debug 2021-10-01 14:44:10: Removing old log directory '/opt/zigbee2mqtt/data/log/2021-09-30.15-32-46'
Zigbee2MQTT:info  2021-10-01 14:44:10: Starting Zigbee2MQTT version 1.21.1 (commit #4a51e0c0)
Zigbee2MQTT:info  2021-10-01 14:44:10: Starting zigbee-herdsman (0.13.138)
Zigbee2MQTT:debug 2021-10-01 14:44:10: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2021-10-01 14:44:13: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2021-10-01 14:44:13: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:debug 2021-10-01 14:44:13: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2021-10-01 14:44:13: Currently 0 devices are joined:
Zigbee2MQTT:warn  2021-10-01 14:44:13: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2021-10-01 14:44:13: Allowing new devices to join.
Zigbee2MQTT:warn  2021-10-01 14:44:13: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2021-10-01 14:44:13: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2021-10-01 14:44:13: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2021-10-01 14:44:13: Connecting to MQTT server at mqtt://192.168.141.184:1883
Zigbee2MQTT:debug 2021-10-01 14:44:13: Using MQTT login with username: MQTT
Zigbee2MQTT:info  2021-10-01 14:44:13: Connected to MQTT server
Zigbee2MQTT:info  2021-10-01 14:44:13: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:debug 2021-10-01 14:44:13: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"4a51e0c0","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_entity_attributes":true,"homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/opt/zigbee2mqtt/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{},"experimental":{"output":"json"},"external_converters":[],"frontend":{"host":"0.0.0.0","port":8080},"groups":{},"homeassistant":false,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":false,"server":"mqtt://192.168.141.184:1883","user":"MQTT"},"ota":{"disable_automatic_update_check":false,"update_check_interval":1440},"passlist":[],"permit_join":true,"serial":{"disable_led":false,"port":"/dev/ttyACM0"},"whitelist":[]},"config_schema":{"definitions":{"device":{"properties":{"debounce":{"description":"Debounces messages of this device","title":"Debounce","type":"number"},"debounce_ignore":{"description":"Protects unique payload values of specified payload properties from overriding within debounce time","examples":["action"],"items":{"type":"string"},"title":"Ignore debounce","type":"array"},"filtered_attributes":{"description":"Filter attributes from publish payload.","examples":["temperature","battery","action"],"items":{"type":"string"},"title":"Filtered publish attributes","type":"array"},"filtered_optimistic":{"description":"Filter attributes from optimistic publish payload when calling /set. (This has no effect if optimistic is set to false).","examples":["color_mode","color_temp","color"],"items":{"type":"string"},"title":"Filtered optimistic attributes","type":"array"},"friendly_name":{"description":"Used in the MQTT topic of a device. By defa [... cut ...] 1"}'
Zigbee2MQTT:debug 2021-10-01 14:44:13: Received MQTT message on 'zigbee2mqtt/bridge/extensions' with data '[]'
Zigbee2MQTT:info  2021-10-01 14:44:13: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"4a51e0c0","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.21.1"}'`

thehiddentruth, 2021-10-01:
Suggestions on what to try next?

thehiddentruth, 2021-10-02:
I ended up solving the issue by change the channel in confirguration.yaml and coordination backup.json from channel 11 to Chanel 25, and that seems to have solved the issue. I can now see all devices.

closed 2021-10-02 by thehiddentruth