Koenkk/zigbee2mqtt #8921: Hue Dimmer Switch 3241310926211 not sending anything

hmrc87, 2021-09-30:
## What happened
I successfully paired my remote https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621 with my deconz-zigbe2mqtt installation.

The debug log shows nothing related to this device whenever I click something.

## What did you expect to happen
When I click a button on the switch an Action should be send and be visible in MQTT

## How to reproduce it (minimal and precise)
- Pair the switch
- press any button

## Debug info
Zigbee2MQTT version
1.21.1 commit: eb5750d
Coordinator type ConBee/RaspBee
Coordinator revision 0x26400500
Frontend version0.6.0

danieledwardgeorgehitchcock, 2021-09-30:
Hi,

It sounds like your device might have been added to Zigbee2MQTT but may have not completed its interview, or is too far away from a node that can route back to the co-ordinator.

If you have access to the gui, or the filesystem for Zigbee2MQTT, you can check the interview by clicking on the device and looking in the about screen. If you don't have the gui, you can do this by opening the database.db file in a text editor, then searching for the device.

If you struggle, please post your database.db entry here and we can have a look!

hmrc87, 2021-09-30:
Thanks, Daniell but the Device finished its interview
[image]

hmrc87, 2021-10-01:
I tested with two of my Tradfri (round) remotes and see the same behaviour. No data is transmitted from those devices. Lights and Switches though work perfectly fine

BigT-CZE, 2021-10-02:
Hello @hmrc87 I successfully paired Philips 324131092621. I first reseted it by holding setup button for 10s, waited to join to Zigbee2MQTT and then pressed all buttons for 4s on a front. Then I received message, that device was successfully paired. Now all button presses are logged properly into zigbee2mqtt log

hmrc87, 2021-10-02:
> Hello @hmrc87 I successfully paired Philips 324131092621. I first reseted it by holding setup button for 10s, waited to join to Zigbee2MQTT and then pressed all buttons for 4s on a front. Then I [...]

Thanks for the hint - i tried it like 3 times. It pairs, the interview is completed but the intersting thing is that the device has no state in stage.json and does not appear in MQTT.

Now I see the following in the log:
` Failed to configure 'HUE DIMMER SWITCH2', attempt 1 (Error: Bind 0x0017880106eabacb/1 genOnOff from '0x00212effff041184/1' failed (waiting for response TIMEOUT)`

closed 2021-11-09 by github-actions[bot]

hmrc87, 2022-04-06:
I actually never got this running. After getting rid of zigbe2mqtt and using the Remote directly with the Phoscon App it works perfectly.

shanelord01, 2022-07-09:
I have this issue with one of my Hue Dimmer Switches but not the other. 

Tried everything I did with the other one. This unit though I get no battery state, no bind to coordinator. Nothing.

The other one everything shows correctly.