Koenkk/zigbee2mqtt #8923: Touchlink fails on CC1352P2

elastic-paul, 2021-09-30:
Lastest firmware on CC1352P2
Latest Zigbee2MQTT

Try to start Touchlink, get error:

```
publish zigbee2mqtt/bridge/request/touchlink/scan

Sep 30 12:00:57 pizigbee npm[535]: Zigbee2MQTT:info  2021-09-30 12:00:57: Start Touchlink scan
Sep 30 12:01:11 pizigbee npm[535]: Zigbee2MQTT:error 2021-09-30 12:01:11: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'SRSP - AF - interPanCtl after 6000ms'
Sep 30 12:01:11 pizigbee npm[535]: Zigbee2MQTT:info  2021-09-30 12:01:11: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{},"error":"SRSP - AF - interPanCtl after 6000ms","status":"error"}'
Sep 30 12:02:25 pizigbee npm[535]: (node:603) UnhandledPromiseRejectionWarning: Error: Cannot execute command, in Inter-PAN mode
Sep 30 12:02:25 pizigbee npm[535]:     at ZStackAdapter.checkInterpanLock (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1023:19)
Sep 30 12:02:25 pizigbee npm[535]:     at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:197:18)
Sep 30 12:02:25 pizigbee npm[535]:     at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:42)
Sep 30 12:02:25 pizigbee npm[535]:     at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:21:18
Sep 30 12:02:25 pizigbee npm[535]:     at new Promise (<anonymous>)
Sep 30 12:02:25 pizigbee npm[535]:     at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:19:16)
Sep 30 12:02:25 pizigbee npm[535]:     at ZStackAdapter.permitJoin (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:196:26)
Sep 30 12:02:25 pizigbee npm[535]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:227:36)
Sep 30 12:02:25 pizigbee npm[535]:     at listOnTimeout (internal/timers.js:554:17)
Sep 30 12:02:25 pizigbee npm[535]:     at processTimers (internal/timers.js:497:7)
Sep 30 12:02:25 pizigbee npm[535]: (node:603) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
Sep 30 12:02:25 pizigbee npm[535]: (node:603) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.```

rindlerblabla, 2021-09-30:
Guess it would help if you specified what firmware and USB-stick you are running. You could find that information under Settings- About in the UI.

Koenkk, 2021-09-30:
Duplicate of #5356

closed 2021-09-30 by Koenkk