Koenkk/zigbee2mqtt #8927: BEGA device not supported

sbmotor, 2021-09-30:
I recently bought the BEGA 70049 Zigbee control module that provides a Dali interface to control my light. It is not supported by  out-of-the-box Zigbee2mqtt. I started to follow the link https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html 

I get stuck because Zigbee model  is empty ''. Can somebody support how to setup new device when zigbee model is empty; _Warning Received message from unsupported device with Zigbee model '' and manufacturer name 'BEGA Gantenbrink-Leuchten KG'_

Information from the vendor can be found at https://www.bega.com/en/products/zigbee-control-module-dali-70049/#/70049/info

Included the log:
Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"i7ha6-2","value":true}'
Info Zigbee: allowing new devices to join.
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"i7ha6-2"}'
Info Device '0x78f5e5000a000298' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x78f5e5000a000298","ieee_address":"0x78f5e5000a000298"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x78f5e5000a000298"},"type":"device_connected"}'
Debug Device '0x78f5e5000a000298' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x78f5e5000a000298","ieee_address":"0x78f5e5000a000298"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x78f5e5000a000298"},"type":"device_announced"}'
Debug Received Zigbee message from '0x78f5e5000a000298', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":402870288,"imageType":144,"manufacturerCode":4357}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '' and manufacturer name 'BEGA Gantenbrink-Leuchten KG'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Debug Received Zigbee message from '0x78f5e5000a000298', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":402870288,"imageType":144,"manufacturerCode":4357}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '' and manufacturer name 'BEGA Gantenbrink-Leuchten KG'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Debug Received Zigbee message from '0x78f5e5000a000298', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":402870288,"imageType":144,"manufacturerCode":4357}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '' and manufacturer name 'BEGA Gantenbrink-Leuchten KG'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Debug Received Zigbee message from '0x78f5e5000a000298', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":402870288,"imageType":144,"manufacturerCode":4357}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '' and manufacturer name 'BEGA Gantenbrink-Leuchten KG'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Debug Received Zigbee message from '0x78f5e5000a000298', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":402870288,"imageType":144,"manufacturerCode":4357}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '' and manufacturer name 'BEGA Gantenbrink-Leuchten KG'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

sbmotor, 2021-10-03:
Okay, got it connected. Not there yet. Will keep posting. Any help and tricks would be welcom.

Used the Tuya example to create external converter; 
fingerprint: [
        {
            modelID: '',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: 'BEGA Gantenbrink-Leuchten KG'
        },
    ],

Now need to start the additional configuration.. As I cannot find any external input on the device itself I will dig into the examples see if it will respond to some input.. Any suggestions would be great for this. 
The vendor indicates its supports Home Automation Profile 1.2.1.

sbmotor, 2021-10-03:
Think I got it working.. Device seems similar to the Osram Zigbee 3.0 DALI CONV LI. Haven't connected the light yet, but turning light on/off gives the familiar 'tick'. Hopefully dimming also works.

External converter now reads. The below. WIll post if it works:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
	{
		fingerprint: [
			{
				modelID: '',
				// empty model name - not provided by vendor.
				manufacturerName: 'BEGA Gantenbrink-Leuchten KG'
			},
		],
		model: '70049', // Vendor model number, look on the device for a model number
		vendor: 'Bega', // Vendor of the device (only used for documentation and startup logging)
		description: 'Zigbee Control module Dali CC/TW', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
		extend: extend.ledvance.light_onoff_brightness(),
	},
];

sbmotor, 2021-10-03:
Yup, the external converter works. How can this be included in the next release?

Koenkk, 2021-10-03:
Can you provide the data/database.db entry of this device?

sbmotor, 2021-10-04:
{"id":17,"type":"Router","ieeeAddr":"0x78f5e5000a000298","nwkAddr":2613,"manufId":4357,"manufName":"BEGA Gantenbrink-Leuchten KG","powerSource":"Mains (single phase)","modelId":"","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,8,9,768,769,64733],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"","manufacturerName":"BEGA Gantenbrink-Leuchten KG","powerSource":1,"zclVersion":1,"appVersion":24,"stackVersion":80,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":70}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":24,"stackVersion":80,"hwVersion":1,"dateCode":"","zclVersion":1,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1633275956522}

sbmotor, 2021-10-04:
[image]

And how could you include the picture of the device?

Koenkk, 2021-10-04:
Added the device. Picture will be available after the next release (1 November).

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-10-04 by Koenkk