Koenkk/zigbee2mqtt #8930: Temperature and Pressure Calibration do not work as expected

TheRealWaldo, 2021-10-01:
## What happened
`temperature_calibration` and `pressure_calibration` only work 'sometimes'.

## What did you expect to happen
Both should work at all times.

## How to reproduce it (minimal and precise)

```
'0x00158d0003231617':
  friendly_name: Pressure Sensor
  device_options:
    temperature_calibration: -0.4
    pressure_calibration: 120
```

After several minutes and a few updates from the device, the calibration was applied.  After a few dozen updates (approximately 40 minutes later),  calibration stopped working.

I noticed this while calibrating sensors for pressure; spikes when it was working, followed by a dip when it wasn't.  It was most evident on the pressure calibration (as you can see, it was 120hPA off).

I could find nothing in logs as to why.

Also, it would be great to be able to calibrate humidity!

## Debug info
Zigbee2MQTT version: 1.21.1-1 (hass.io addon)
Adapter hardware:  Conbee II

TheRealWaldo, 2021-10-01:
I may have figured out that the config is supposed to be:

```
'0x00158d0003231617':
  friendly_name: Pressure Sensor
  temperature_calibration: -0.4
  pressure_calibration: 120
```

But it's not very clear from the documentation.

It also appears that humidity calibration is available but undocumented.

I am still having some issues with the values not seeming accurate, but I will have to wait a while for subsequent reports to be sure.

ralong777, 2021-10-03:
For me the any of the custom options doesnt seem to work:
```
'0x00158d0006f78330':
  friendly_name: Aqara PIR sensor2
  temperature_calibration: -2
  temperature_precision: 3
  occupancy_timeout: 60
  ```
  
either the precision or the calibration or timeout

closed 2021-11-10 by github-actions[bot]