Koenkk/zigbee2mqtt #8931: GZCGQ01LM not working

mkania11, 2021-10-01:
## What happened
GZCGQ01LM not working, illuminance_lux show null
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II GZCGQ01LM MiJia light intensity sensor
Adapter firmware version: 2019www.

kocmocgim, 2021-10-01:
same problem
after zigbee network restart

kocmocgim, 2021-10-01:
[image]

kocmocgim, 2021-10-01:
[image]
re add both properties

closed 2021-11-08 by github-actions[bot]