Koenkk/zigbee2mqtt #8934: TS0601 - _TZE200_chy_chyvmhay Thermostat not supported - please add if possible

gmarchart, 2021-10-01:
hello, 

can you please add support for this device - a thermostat device - Lidl Sylvercrest. can be added in zigbee2mqtt but its not supported - 
adding with documentation how to add a new device was not sucessfully.

Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_chyvmhay'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Debug Received Zigbee message from 'Thermostat-VZ', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,155],"type":"Buffer"},"datatype":2,"dp":35,"fn":0,"status":0,"transid":35}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_chyvmhay'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Zigbee Modell
TS0601
zigbee Hersteller
_TZE200_chy_chyvmhay

Zigbee2mqtt
Current version: 1.21.1-1
Homeassistant HassOS - 
Betriebssystem
Home Assistant OS 6.4
Supervisor
Version
supervisor-2021.09.6

Core
Version
core-2021.9.7

Conbee 2 on Raspi with HassOS

Thanks, 
Br Guenther

Lavisx, 2021-10-05:
I'm not sure if it help for add this device but I attache the tuya.dump.txt from my device.
[tuya.dump.txt](https://github.com/Koenkk/zigbee2mqtt/files/7284124/tuya.dump.txt)

drob3k, 2021-10-05:
yes

Lavisx, 2021-10-06:
I'm detect some values:
```
                switch (dp) {
                // heatingSetpoint:
                case 16: // DPID that we added to common
                    return {current_heating_setpoint: (value / 2).toFixed(1)}; 
                // localTemp:
                case 24:
                    return {local_temperature: parseFloat((value / 10).toFixed(1))};
                // childLock:
                case 40:
                    return {child_lock: value ? 'LOCK' : 'UNLOCK'};
                // tempCalibration:
                case 104:
                    return {local_temperature_calibration: parseFloat((value / 10).toFixed(1))};
```

gmarchart, 2021-10-06:
Hello, 
it seems to be a mix of this moes thermostat, maybe  ?! 

datapoints from moes and moesS Thermostat

// Moes Thermostat
    moesHold: 2,
    moesScheduleEnable: 3,
    moesHeatingSetpoint: 16,
    moesMaxTempLimit: 18,
    moesMaxTemp: 19,
    moesDeadZoneTemp: 20,
    moesLocalTemp: 24,
    moesTempCalibration: 27,
    moesValve: 36,
    moesChildLock: 40,
    moesSensor: 43,
    moesSchedule: 101,
    etopErrorStatus: 13,
    // MoesS Thermostat
    moesSsystemMode: 1,
    moesSheatingSetpoint: 2,
    moesSlocalTemp: 3,
    moesSboostHeating: 4,
    moesSboostHeatingCountdown: 5,
    moesSreset: 7,
    moesSwindowDetectionFunktion_A2: 8,
    moesSwindowDetection: 9,
    moesSchildLock: 13,
    moesSbattery: 14,
    moesSschedule: 101,
    moesSvalvePosition: 104,
    moesSboostHeatingCountdownTimeSet: 103,
    moesScompensationTempSet: 105,
    moesSecoMode: 106,
    moesSecoModeTempSet: 107,
    moesSmaxTempSet: 108,
    moesSminTempSet: 109,

Lavisx, 2021-10-06:
It look like rebrand of this model moes SH4-ZB
https://zigbee.blakadder.com/Moes_SH4-ZB.html

Lavisx, 2021-10-08:
In the converter are some changes to correct modes and presets

Lavisx, 2021-10-09:
I'm sorry but not now. I'm not sure that I can create it.
This is only my amateur code and missing there scheduling and away settings and two bool values are unknown for me.

synekvl, 2021-10-09:
Thks Lavisx, finally I have some data from the valve. Unfortunately, I had to commented out the line 299 ( e.battery_voltage() ) because it made some error in my installation causing zigbee2mqtt doesn't start.

Cheers
Vladimir

Lavisx, 2021-10-11:
I'm sorry but the valve don't send any position or any information if it is heating or idle

Lavisx, 2021-10-15:
The scheduler has a strange way of setting.  There are 9 temperature values and only 8 time settings.
I don't tested it but it looks like the temperature following the time is assigned to the time.
Now I'm not sure what mean the first temperature. Means it temperature after midnight? That's just my guess without any tests.

lukasf98, 2021-10-15:
For me saving the schedule works. Maybe you have the same issue I had: https://github.com/Lavisx/lidl-valve-z2m/issues/1

gmarchart, 2021-10-15:
> @gmarchart How do you set the schedule, via zigbee2mqtt frontend (which is not working for me, I cannot change/save schedule settings), or on the device itself?

yes via zigbee2mqtt frontend, works without issue - i change the hour and minute and temp - then the blue button (Anwenden in german- submit maybe in english ?! - i use a german interface) of the day i changed the setting and then after few seconds the settings are saved. works.

nicx, 2021-10-15:
> I'm sorry but the valve don't send any position or any information if it is heating or idle

@Lavisx are you sire? as of the zigbee2mqtt documentation the tuya valve exposes the position value in %. Is that not the case here?

kasik96, 2021-10-31:
I just tested this on the dev channel, and it does not respond to any changes. It only shows the set the temperature correctly. When I set temperature, change mode, or do other things, nothing will happen. I have two of these with the same problem.

13r1ckz, 2021-11-03:
for me no interaction with the device or any return info I can use.
if there is any interesting info I could supply please let me know.
here is the debug log on the device joining the network:
`Warning Device 'ETRV' left the network
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"ETRV","ieee_address":"0x0c4314fffe619f2f"},"type":"device_leave"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x0c4314fffe619f2f"},"type":"device_removed"}'
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"xlmeh-1","value":true}'
Info Zigbee: allowing new devices to join.
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"xlmeh-1"}'
Info Device 'ETRV' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"ETRV","ieee_address":"0x0c4314fffe619f2f"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"ETRV"},"type":"device_connected"}'
Debug Device 'ETRV' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"ETRV","ieee_address":"0x0c4314fffe619f2f"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"ETRV"},"type":"device_announced"}'
Debug Retrieving state of 'ETRV' after reconnect
Debug Device 'ETRV' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"ETRV","ieee_address":"0x0c4314fffe619f2f"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"ETRV"},"type":"device_announced"}'
Debug Received Zigbee message from 'ETRV', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":116,"fn":0,"status":0,"transid":116}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/ETRV', payload '{"auto_lock":"MANUAL","away_mode":null,"away_preset_days":null,"away_preset_temperature":null,"battery_low":null,"boost_time":null,"child_lock":null,"comfort_temperature":null,"eco_temperature":null,"force":null,"linkquality":111,"max_temperature":null,"min_temperature":null,"position":null,"update_available":null,"valve_detection":null,"week":null,"window_detection":null}'
Debug Received Zigbee message from 'ETRV', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":116,"fn":0,"status":0,"transid":116}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/ETRV', payload '{"auto_lock":"MANUAL","away_mode":null,"away_preset_days":null,"away_preset_temperature":null,"battery_low":null,"boost_time":null,"child_lock":null,"comfort_temperature":null,"eco_temperature":null,"force":null,"linkquality":108,"max_temperature":null,"min_temperature":null,"position":null,"update_available":null,"valve_detection":null,"week":null,"window_detection":null}'
Debug Received Zigbee message from 'ETRV', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":117,"fn":0,"status":0,"transid":117}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/ETRV', payload '{"auto_lock":"MANUAL","away_mode":null,"away_preset_days":0,"away_preset_temperature":null,"battery_low":null,"boost_time":null,"child_lock":null,"comfort_temperature":null,"eco_temperature":null,"force":null,"linkquality":108,"max_temperature":null,"min_temperature":null,"position":null,"update_available":null,"valve_detection":null,"week":null,"window_detection":null}'
Debug Received Zigbee message from 'ETRV', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":117,"fn":0,"status":0,"transid":117}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/ETRV', payload '{"auto_lock":"MANUAL","away_mode":null,"away_preset_days":0,"away_preset_temperature":null,"battery_low":null,"boost_time":null,"child_lock":null,"comfort_temperature":null,"eco_temperature":null,"force":null,"linkquality":93,"max_temperature":null,"min_temperature":null,"position":null,"update_available":null,"valve_detection":null,"week":null,"window_detection":null}'
Debug Received Zigbee message from 'ETRV', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1,34,24,42,36,34,68,42,92,34,96,42,96,34,96,42,96,34],"type":"Buffer"},"datatype":0,"dp":109,"fn":0,"status":0,"transid":109}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/ETRV', payload '{"auto_lock":"MANUAL","away_mode":null,"away_preset_days":0,"away_preset_temperature":null,"battery_low":null,"boost_time":null,"child_lock":null,"comfort_temperature":null,"eco_temperature":null,"force":null,"linkquality":108,"max_temperature":null,"min_temperature":null,"position":{"0":1,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"update_available":null,"valve_detection":null,"week":null,"window_detection":null}'
Debug Retrieving state of 'ETRV' after reconnect
[... cut ...] ,"min_temperature":null,"position":{"0":1,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"update_available":null,"valve_detection":null,"window_detection":null,"workdays":[{"hour":4,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}]}'`

theip4nda, 2021-11-05:
I testet with iobroker.. Object are all createt but i cant control anything :-( You need any infos for adding in database correct?

Eddiedigits, 2021-12-01:
Firstly, thanks for all the work on adding this device!

Some extra info:
dp: 16 sets the temp for manual mode. Currently if I set the temp from HA in the Climate widget with the device set to Auto. It sends the new temp on dp: 16 which is wrong. It should use dp: 105.

the schedule for a day is not in time temp pairs. the device switches to the first temp at midnight and if all the times are less than 24:00 then the last temp [temp9] is used from the last time [time8] until midnight.
00:00 [temp1, time1, temp2, time2....,  temp8, time8, temp9] 24:00

Other than that I think all the dp's I found have already been identified, but for reference here is what I had before I realised someone else was already on it:
    scAutoHeatingSetpoint: 105, // Temp in auto mode
    scLocalTemp: 24, // Local temp reading at thermostat
    scAutoLock: 3,
    scBatteryVoltage: 35,
    scChildLock: 40,
    scMode: 2, // ['AUTO', 'MANUAL', 'OFF']
    scManualHeatingSetpoint: 16, // Temp in manual mode
    scScheduleMonday: 109, // datatype: 0
    scScheduleTuesday: 110, // datatype: 0
    scScheduleWednesday: 111, // datatype: 0
    scScheduleThursday: 112, // datatype: 0
    scScheduleFriday: 113, // datatype: 0
    scScheduleSaturday: 114, // datatype: 0
    scScheduleSunday: 115, // datatype: 0

ciB89, 2021-12-01:
There are still two unkown binaries, one of them is the state of the "open window detection", so wheather is has detected that a window is open or closed. Also, according to the Lidl app, you can enable to be reminded about an open window, so that should be the other binary.

gmarchart, 2021-12-02:
> There are still two unkown binaries, one of them is the state of the "open window detection", so wheather is has detected that a window is open or closed. Also, according to the Lidl app, you can enable to be reminded about an open window, so that should be the other binary.

Hello, 
it can also be that the one of the two unknown binary is the setting for automatic Summer/Wintertime. In the Valve-Manual is such option described to set it on the device. It's the Valve Menu Point "dSt" - can be set OFF or ON.

ciB89, 2021-12-02:
> it can also be that the one of the two unknown binary is the setting for automatic Summer/Wintertime. In the Valve-Manual is such option described to set it on the device. It's the Valve Menu Point "dSt" - can be set OFF or ON.
> 
Right, that also makes sense. 
> the z2m get may restarts since i use the new version (1.22.1-1) approx. every 2-3 minutes !!! - always a new logfile will be created and the devices are unknown for the restart for few seconds- i will try to reboot HA, maybe it is then better or fixed.
> 
> Additional when i manualy change the mode in HA with the climate widget of this Valve - "auto" to "manual" or "off" z2m restarts every time i do the change. no matter from which mode to another, always get a restart.

I also got the same issue. I already pointed that out in the other PR #9509

Koenkk, 2021-12-03:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)