Koenkk/zigbee2mqtt #8937: Message from device processed after next message from other device received

slavik73, 2021-10-01:
# Bug Report
## What happened
I have at the moment Mijia door sensor and Xiaomi cube connected. The message from door sensor is seen by the driver/mqtt only after I'm doing something with the cube. I think it start to happen only with the latest versions of zigbee2mqtt as everything worked about 2 months ago and before I upgraded docker images.

## What did you expect to happen
I want to get reaction to door sensor immediately. Here door was opened 5 secs before cube was flipped.

## How to reproduce it (minimal and precise)

## Debug Info
2021-10-01T14:43:34.019Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,219,197,1,1,0,135,0,231,141,50,0,0,7,24,5,10,0,0,16,1,219,197,29,27,254,28,68,129,0,0,18,0,203,192,2,1,0,165,0,68,88,62,0,0,8,24,16,10,85,0,33,2,0,203,192,29,46]

2021-10-01T14:43:34.020Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,219,197,1,1,0,135,0,231,141,50,0,0,7,24,5,10,0,0,16,1,219,197,29,27,254,28,68,129,0,0,18,0,203,192,2,1,0,165,0,68,88,62,0,0,8,24,16,10,85,0,33,2,0,203,192,29,46]

2021-10-01T14:43:34.020Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,219,197,1,1,0,135,0,231,141,50,0,0,7,24,5,10,0,0,16,1,219,197,29] - 27

2021-10-01T14:43:34.021Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":50651,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":135,"securityuse":0,"timestamp":3313127,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,5,10,0,0,16,1]}}

2021-10-01T14:43:34.024Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":5,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":1}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":50651,"endpoint":1,"linkquality":135,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'

Zigbee2MQTT:debug 2021-10-01 17:43:34: Received Zigbee message from 'ToiletDoor_Slava', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0

2021-10-01T14:43:34.027Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,18,0,203,192,2,1,0,165,0,68,88,62,0,0,8,24,16,10,85,0,33,2,0,203,192,29,46]

2021-10-01T14:43:34.027Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,18,0,203,192,2,1,0,165,0,68,88,62,0,0,8,24,16,10,85,0,33,2,0,203,192,29] - 46

2021-10-01T14:43:34.028Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":18,"srcaddr":49355,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":165,"securityuse":0,"timestamp":4085828,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,16,10,85,0,33,2,0]}}

2021-10-01T14:43:34.028Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":16,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":85,"dataType":33,"attrData":2}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":49355,"endpoint":2,"linkquality":165,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'

Zigbee2MQTT:debug 2021-10-01 17:43:34: Received Zigbee message from 'Cube', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 2 with groupID 0

2021-10-01T14:43:34.029Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

Zigbee2MQTT:info  2021-10-01 17:43:34: MQTT publish: topic 'zigbee2mqtt/ToiletDoor_Slava', payload '{"battery":100,"contact":false,"linkquality":135,"temperature":30,"voltage":3315}'

Zigbee2MQTT:info  2021-10-01 17:43:34: MQTT publish: topic 'zigbee2mqtt/Cube', payload '{"action":"wakeup","action_angle":null,"action_from_side":null,"action_side":null,"action_to_side":null,"angle":29.99,"battery":100,"linkquality":165,"side":5,"voltage":3005}'

Zigbee2MQTT:info  2021-10-01 17:43:34: MQTT publish: topic 'zigbee2mqtt/Cube', payload '{"action":"","action_angle":null,"action_from_side":null,"action_side":null,"action_to_side":null,"angle":29.99,"battery":100,"linkquality":165,"side":5,"voltage":3005}'

Zigbee2MQTT:info  2021-10-01 17:43:34: MQTT publish: topic 'zigbee2mqtt/Cube/action', payload 'wakeup'

2021-10-01T14:43:34.953Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,18,0,203,192,2,1,0,183,0,125,60,63,0,0,8,24,17,10,85,0,33,105,0,203,192,29,10]

2021-10-01T14:43:34.954Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,18,0,203,192,2,1,0,183,0,125,60,63,0,0,8,24,17,10,85,0,33,105,0,203,192,29,10]

2021-10-01T14:43:34.954Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,18,0,203,192,2,1,0,183,0,125,60,63,0,0,8,24,17,10,85,0,33,105,0,203,192,29] - 10

2021-10-01T14:43:34.955Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":18,"srcaddr":49355,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":183,"securityuse":0,"timestamp":4144253,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,17,10,85,0,33,105,0]}}

2021-10-01T14:43:34.956Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":17,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":85,"dataType":33,"attrData":105}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":49355,"endpoint":2,"linkquality":183,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'

Zigbee2MQTT:debug 2021-10-01 17:43:34: Received Zigbee message from 'Cube', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":105}' from endpoint 2 with groupID 0

2021-10-01T14:43:34.959Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

Zigbee2MQTT:info  2021-10-01 17:43:34: MQTT publish: topic 'zigbee2mqtt/Cube', payload '{"action":"flip90","action_angle":null,"action_from_side":5,"action_side":1,"action_to_side":1,"angle":29.99,"battery":100,"linkquality":183,"side":1,"voltage":3005}'

Zigbee2MQTT:info  2021-10-01 17:43:34: MQTT publish: topic 'zigbee2mqtt/Cube', payload '{"action":"","action_angle":null,"action_from_side":null,"action_side":null,"action_to_side":null,"angle":29.99,"battery":100,"linkquality":183,"side":1,"voltage":3005}'

Zigbee2MQTT:info  2021-10-01 17:43:34: MQTT publish: topic 'zigbee2mqtt/Cube/action', payload 'flip90'

Zigbee2MQTT version: 1.21.1/latest
Adapter hardware: zzh!
Adapter firmware version: CC2652R_coordinator_20210708.hex

Koenkk, 2021-10-01:
Probably due to https://github.com/Koenkk/zigbee2mqtt/issues/8663

slavik73, 2021-10-01:
Thanks A LOT. Upgraded kernel 5.11.something -> 5.14 and it looks much better. Will reconnect other devices and check if it's completely solved.

slavik73, 2021-10-03:
After one more day - works OK for my sensors

closed 2021-10-03 by slavik73