Koenkk/zigbee2mqtt #8940: Badly recognized AwoX GU10 bulb

MysterePMS, 2021-10-01:
Hello,
I have 10 AwoX GU10 bulbs (Eglo Crosslink-z 33946 PAR10 GU10).
Some are recognized as: "manufacturer Id": 4417, "manufacturer Name": "AwoX", "model": "33944"
And others in: "manufacturer Id": 4417, "manufacturer Name": "AwoX", "model": "33951/33948".
While on the 10 bulbs they are in 33946.
The problem is that this recognized in 33951/33948 by zigbee2mqtt does not offer the change of colors.
[image]
[image]

Koenkk, 2021-10-02:
Could you provide the data/database.db entry of the problematic devices?

MysterePMS, 2021-10-10:
> Pourriez-vous fournir l'entrée data/database.db des appareils problématiques ?

Ah, je ne savais pas que vous parlez français, ca aurait été plus simple pour exposé le problème :p

Dite moi comment récupérer le database et je vous fais ca ;-)

MysterePMS, 2021-10-11:
> Pourriez-vous fournir l'entrée data/database.db des appareils problématiques ?

`{"id":29,"type":"Router","ieeeAddr":"0xa4c138fe5b4793b6","nwkAddr":39350,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1633952203793}
`

Koenkk, 2021-10-12:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

MysterePMS, 2021-10-13:
I have 10 bulbs 33946.
8 are detected in 33944 and 2 were detected in 33951/33948 but thanks to your update the 2 have increased to 33943/33946.
However the 8 remained in 33944 while they are identical to the 10.
[image]

MysterePMS, 2021-10-21:
Hello,
I allow myself to reopen the subject because since the last update of Zigbee2mqtt, my  bulbs 33946 which are recognized good have all returned to 33951/33948.
Therefore I can no longer manage the change of colors and white / yellow.
[image]