Koenkk/zigbee2mqtt #8942: Version 1.21.2 not update topic zigbee2mqtt/bridge/config

IgorSimic, 2021-10-01:
After update to version 1.21.2 and restart I noticed that topic zigbee2mqtt/bridge/config did not changed, field version remains 1.21.1
I am not absolutelly sure, but I think that this was updated automatically in the past, yet I am sure taht running version is 1.21.2

Log:
Oct  1 20:29:51 hass npm[668]: #033[32mZigbee2MQTT:info #033[39m 2021-10-01 20:29:51: Disconnecting from MQTT server
Oct  1 20:29:51 hass npm[668]: #033[32mZigbee2MQTT:info #033[39m 2021-10-01 20:29:51: Stopping zigbee-herdsman...
Oct  1 20:30:00 hass npm[668]: #033[32mZigbee2MQTT:info #033[39m 2021-10-01 20:30:00: Stopped zigbee-herdsman
Oct  1 20:30:00 hass npm[668]: #033[32mZigbee2MQTT:info #033[39m 2021-10-01 20:30:00: Stopped Zigbee2MQTT
Oct  1 20:30:00 hass npm[668]: #033[34mZigbee2MQTT:debug#033[39m 2021-10-01 20:30:00: Loaded state from file /opt/zigbee2mqtt/data/state.json
Oct  1 20:30:00 hass npm[668]: #033[32mZigbee2MQTT:info #033[39m 2021-10-01 20:30:00: Logging to console only'
Oct  1 20:30:00 hass npm[668]: #033[32mZigbee2MQTT:info #033[39m 2021-10-01 20:30:00: Starting Zigbee2MQTT version 1.21.2 (commit #44922120)
Oct  1 20:30:00 hass npm[668]: #033[32mZigbee2MQTT:info #033[39m 2021-10-01 20:30:00: Starting zigbee-herdsman (0.13.149)
Oct  1 20:30:00 hass npm[668]: #033[34mZigbee2MQTT:debug#033[39m 2021-10-01 20:30:00: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":16,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/zigazah"}}'
Oct  1 20:30:12 hass npm[668]: #033[32mZigbee2MQTT:info #033[39m 2021-10-01 20:30:12: zigbee-herdsman started (resumed)
Oct  1 20:30:12 hass npm[668]: #033[32mZigbee2MQTT:info #033[39m 2021-10-01 20:30:12: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
Oct  1 20:30:12 hass npm[668]: #033[34mZigbee2MQTT:debug#033[39m 2021-10-01 20:30:12: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Oct  1 20:30:12 hass npm[668]: #033[32mZigbee2MQTT:info #033[39m 2021-10-01 20:30:12: Set transmit power to '5'
Oct  1 20:30:12 hass npm[668]: #033[32mZigbee2MQTT:info #033[39m 2021-10-01 20:30:12: Currently 38 devices are joined:

IgorSimic, 2021-10-02:
It looks like zigbee2mqtt/bridge/config is completely thrown out, but all subtopics which it contained are present in zigbee2mqtt/bridge/info.
So, if you have sensors relaying on zigbee2mqtt/bridge/config like I do, it is possible just switch to zigbee2mqtt/bridge/info.
If throwing out zigbee2mqtt/bridge/config was intention, than this issue can be closed.

Koenkk, 2021-10-02:
`zigbee2mqtt/bridge/config` is legacy, but it should still be published to. Only when you have the following set in your `configuration.yaml` it is disabled:

```yaml
advanced:
  legacy_api: false
```

IgorSimic, 2021-10-02:
Thank you very much.
Yes I have legacy_api: false but until 1.21.2  zigbee2mqtt/bridge/config was published.
However, because all of it is present in  zigbee2mqtt/bridge/info I don't need it anymore, I just needed to change topic for my version sensors in HA.
Best regards.

closed 2021-10-02 by IgorSimic