Koenkk/zigbee2mqtt #8944: TOGGLE command not working correctly on some wall switches without neutral (MOES 1gang)

ndrewpj, 2021-10-01:
## What happened
mqtt command for on / off work as expected with the relay changing state. However, with toggle the zigbee gateway (?) reports back a change of state but the switch/relay does not!
## What did you expect to happen
the power state of the switch / relay should toggle / change state with the "toggle" command

Here is what happens in the log:

```
Debug Received MQTT message on 'zigbee2mqtt/0x84fd27fffe6032f8/set' with data '{"state": "TOGGLE"}'
Debug Publishing 'set' 'state' to '0x84fd27fffe6032f8'
Debug Received Zigbee message from '0x84fd27fffe6032f8', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":44}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x84fd27fffe6032f8', payload '{"last_seen":"2021-10-01T22:02:23+03:00","linkquality":47,"state":"OFF"}'
Debug Received MQTT message on 'zigbee2mqtt/0x84fd27fffe6032f8/set' with data '{"state": "TOGGLE"}'
Debug Publishing 'set' 'state' to '0x84fd27fffe6032f8'
Debug Received Zigbee message from '0x84fd27fffe6032f8', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":45}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x84fd27fffe6032f8', payload '{"last_seen":"2021-10-01T22:02:27+03:00","linkquality":47,"state":"OFF"}'
Debug Saving state to file /config/zigbee2mqtt/state.json
Debug Received MQTT message on 'zigbee2mqtt/0x84fd27fffe6032f8/set' with data '{"state": "TOGGLE"}'
Debug Publishing 'set' 'state' to '0x84fd27fffe6032f8'
Debug Received Zigbee message from '0x84fd27fffe6032f8', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":46}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x84fd27fffe6032f8', payload '{"last_seen":"2021-10-01T22:02:48+03:00","linkquality":44,"state":"OFF"}'
```

## How to reproduce it (minimal and precise)
Set a simple script with payload of "TOGGLE" command as per device documentation:
https://www.zigbee2mqtt.io/devices/ZTS-EU_1gang.html#moes-zts-eu_1gang
Run script - no toggle.

## Debug info
Zigbee2MQTT version: 1.21.2
Adapter hardware: CC2531
Adapter firmware version:  20190425 zStack30x

olili, 2021-10-14:
same with me as well. 
Any news on this? O.

ndrewpj, 2021-11-21:
> Confirming I experience the same using zzh as adapter with latest FW. My device is different than the pictured but still recognized as `TS0601` `_TZE200_amp6tsvy`. [Here's](https://www.moeshouse.com/collections/zigbee/products/zigbee-smart-push-button-switch-and-zigbee-scene-switch-kit-l-only-no-neutral-wire-or-l-n-wiring-optional-no-capacitor-required-tuya-zigbee-hub-required?variant=39364751622225) what mine looks like.

Yes, I have the same device from Moes.

imjoel, 2021-11-26:
any resolution to this?

closed 2022-01-04 by github-actions[bot]