Koenkk/zigbee2mqtt #8946: Yale YMC-420 D Support

nieror, 2021-10-01:
Hi there, I just bought a new Yale YMC 420 D and looks like there's a little information regarding this model. As far as I could check, it was released about a month ago, so...I couldn't find anything "relatable".
I've followed all steps to "Add a new device" taking as model the YMF40. After setting the .js file, I was able to set up the lock on zigbee2mqtt and also the integration with HA. 
BUT.... as noted below, I can only check the lock status, the battery value isn't reporting correctly and there are no information about which method was used to unlock.
Pretty much the only "correct" information as far as I can see is the LSI...
I'm pretty new to all this stuff, so if one of you could please guide me thru it, I would really appreciate.

**yale.js**
{
        zigbeeModel: ['YMC 420 D'],
        model: 'YMC 420 D',
        vendor: 'Yale',
        description: 'YMC 420 D - Digital Lock',
        extend: lockExtend,
        meta: {timeout: 20000},
    },

**MQTT Info**
zigbee2mqtt/YMC420D
10 most recently received message(s)
Received 16:05:26
QoS: 0
Payload: battery: 192
linkquality: 164
lock_state: locked
state: LOCK

**zigbee2mqtt Debug Log**
Debug Received Zigbee message from 'YMC420D', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"battery":190,"linkquality":148,"lock_state":"locked","state":"LOCK"}'
Debug Received Zigbee message from 'YMC420D', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":190}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"battery":190,"linkquality":180,"lock_state":"locked","state":"LOCK"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"YMC420D"},"status":"ok","transaction":"ncxl4-1"}'
Debug No converter available for 'YMC 420 D' with cluster 'closuresDoorLock' and type 'raw' and data '{"data":[9,35,32,0,2,1,0,0,227,28,234,40],"type":"Buffer"}'
Debug Received Zigbee message from 'YMC420D', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"battery":190,"linkquality":172,"lock_state":"unlocked","state":"UNLOCK"}'
Info MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"battery":190,"linkquality":172,"lock_state":"unlocked","state":"UNLOCK"}'
Info MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"battery":190,"linkquality":152,"lock_state":"locked","state":"LOCK"}'
Info MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"battery":190,"linkquality":148,"lock_state":"locked","state":"LOCK"}'
Debug No converter available for 'YMC 420 D' with cluster 'closuresDoorLock' and type 'raw' and data '{"data":[9,47,32,2,10,0,0,0,242,28,234,40],"type":"Buffer"}'

## Information about the device + link
I've tried to find an international link, but also unable......

https://www.yalehome.com.br/pt-BR/produtos1/products/fechaduras-digitais-biometricas/fechadura-digital-ymc420d/

## data/database.db entry of the device
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d0006dbf814","nwkAddr":30626,"manufId":4151,"manufName":"Yale","powerSource":"Battery","modelId":"YMC 420 D","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2064,"inClusterList":[0,4,3,4111,257],"outClusterList":[0,4,3,4111,257],"clusters":{"genBasic":{"attributes":{"modelId":"YMC 420 D","manufacturerName":"Yale","powerSource":3,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20160301","swBuildId":"02.01.03"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":184}},"closuresDoorLock":{"attributes":{"lockState":1,"operatingMode":1}}},"binds":[{"cluster":257,"type":"endpoint","deviceIeeeAddress":"0x804b50fffe0ea2ed","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x804b50fffe0ea2ed","endpointID":1}],"configuredReportings":[{"cluster":257,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20160301","swBuildId":"02.01.03","zclVersion":2,"interviewCompleted":true,"meta":{"configured":923668394},"lastSeen":1633115328213}

nieror, 2021-11-08:
After some research (and broken things) I was able to get new stats from the Lock such as the unlock method.
I've put together some info from another devices and was able to return the values.

1st discovery - Unlike the other locks, the 420D is returning its values as type RAW.
2nd discovery - For some reason commandOperationEventNotification, commandProgrammingEventNotification, commandGetPinCodeRsp, commandGetUserStatusRsp are not returning AT ALL from the lock.

Tweaking the js file as below, I was able to return a few more values than before. But yet, I cannot return anything related to the action_user...

**_const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const constants = require('zigbee-herdsman-converters/lib/constants');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    action: {
        cluster: 'closuresDoorLock',
        type: 'raw',
        convert: (model, msg, publish, options, meta) => {
            const lookup = {                
                0: 'keypad_unlock',
                1: 'zigbee_unlock',
		2: 'manual_lock',                
                3: 'rfid_unlock',
		4: 'finger_unlock',                
                10: 'auto_lock',
                14: 'inside_unlock',
            };
            };
            const value = lookup[msg.data[4]];
            if (value == 'lock' || value == 'inside_unlock') {
                return {action: value};
            } else {
                return {action: lookup[msg.data[3]]};
            }
        },
    },
}

const lockExtend = (meta) => {
    return {
        fromZigbee: [fz.lock_operation_event, fz.lock_programming_event, fz.lock_pin_code_response, fz.lock_user_status_response, fz.lock, fz.battery,  
            fzLocal.action],
        toZigbee: [tz.lock, tz.pincode_lock, tz.lock_userstatus],
        meta: {pinCodeCount: 250, ...meta},
        exposes: [e.lock(), e.battery(), e.pincode()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await reporting.lockState(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
        },
    };
};

module.exports = [    
	{
        zigbeeModel: ['YMC 420 D'],
        model: 'YMC 420 D',
        vendor: 'Yale',
        description: 'YMC 420 D - Digital Lock',
        extend: lockExtend({timeout: 20000}),
    },
];_**

[image]

[... 9 lines cut ...]
Zigbee2MQTT:debug 2021-11-08 18:26:02: Received Zigbee message from 'YMC420D', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-11-08 18:26:02: MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"battery":86,"linkquality":140,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:debug 2021-11-08 18:26:03: Received Zigbee message from 'YMC420D', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-11-08 18:26:03: MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"battery":86,"linkquality":164,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:debug 2021-11-08 18:26:03: Received Zigbee message from 'YMC420D', type 'raw', cluster 'closuresDoorLock', data '{"data":[9,224,32,2,7,0,0,0,107,82,28,41],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-11-08 18:26:03: MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"action":"unlock","battery":86,"linkquality":156,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:info  2021-11-08 18:26:03: MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"action":"","battery":86,"linkquality":156,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:info  2021-11-08 18:26:03: MQTT publish: topic 'zigbee2mqtt/YMC420D/action', payload 'unlock'
Zigbee2MQTT:debug 2021-11-08 18:26:16: Received Zigbee message from 'YMC420D', type 'raw', cluster 'closuresDoorLock', data '{"data":[9,231,32,3,2,1,0,0,121,82,28,41],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-11-08 18:26:16: MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"action":"rfid_unlock","battery":86,"linkquality":96,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:info  2021-11-08 18:26:16: MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"action":"","battery":86,"linkquality":96,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:info  2021-11-08 18:26:16: MQTT publish: topic 'zigbee2mqtt/YMC420D/action', payload 'rfid_unlock'
Zigbee2MQTT:debug 2021-11-08 18:26:17: Received Zigbee message from 'YMC420D', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-11-08 18:26:17: MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"battery":86,"linkquality":156,"lock_state":"unlocked","state":"UNLOCK"}'
Zigbee2MQTT:debug 2021-11-08 18:26:17: Received Zigbee message from 'YMC420D', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
[... cut ...]
Zigbee2MQTT:info  2021-11-08 18:26:23: MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"action":"","battery":86,"linkquality":172,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:info  2021-11-08 18:26:23: MQTT publish: topic 'zigbee2mqtt/YMC420D/action', payload 'unlock'
Zigbee2MQTT:debug 2021-11-08 18:26:32: Received Zigbee message from 'Sensor Porta da Sala', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0

nieror, 2021-11-11:
hey @Koenkk I know you're probably swamped there, but....we could use a hand or two :)

oguime, 2021-11-18:
I would be interested too. Seems like a nice update to the YRL226L with the addition of card and fingerprint authentication for an equivalent price, if you consider that the zigbee module comes standard...

dudu631, 2021-11-26:
I'm up for this as well!

Koenkk, 2021-11-27:
@nieror where are you currently stuck? (what doesn't work yet?)

nieror, 2021-12-01:
hello @Koenkk thank you for you reply. (sorry for the delay... I'm in a stressfull work related week so I wasn't able to reply).

Summing up, pretty much anything related to PIN operations. 
I am able to retrieve lock status, unlock type (e.g.: rfid, finger, etc) 
I just noticed that I can change an user's PIN via the "Exposes" page, but I don't get any info from there, like, if it was successful or not....I had to check the logs to make sure the SET operation completed and on the lock itself if really changed the PIN.

So pretty much I'm stuck with retrieving the PIN/User that unlocked the door in any way. As I've mentioned in my lastest post, I can't see on the debug anything related to the commandOperationEventNotification, commandProgrammingEventNotification, commandGetPinCodeRsp, commandGetUserStatusRsp events.

I think I forgot to mention, I'm using a Sonoff ZB Bridge, not sure if it would interfere....

Koenkk, 2021-12-01:
>  I'm using a Sonoff ZB Bridge

This could be the issue, I expect this locks works with the same config as all other Yale locks. I'm not sure if the sonoff ZB bridge forwards all messages to z2m (the EZSP integration is very experimental). Could you try with a recommended adapter: https://www.zigbee2mqtt.io/guide/adapters/#recommended

rrlevy, 2021-12-01:
I just purchased this lock, and I'll test with different adapters to confirm (I have a Conbee II now, and a zzh not being used)

But from what I've heard from other people that have this product, this lock is still not working with Zigbee2Mqtt, only directly with ZHA.

I'll post my results here as soon as I get it installed

rrlevy, 2021-12-04:
I just installed mine here using Conbee II. It looks just like the Sonoff example above.

Here are my logs for a finger unlock:

[image]

Zigbee2MQTT:info  2021-12-04 16:44:40: MQTT publish: topic 'zigbee2mqtt/Yale YMC420D', payload '{"action":"finger_unlock","battery":0,"linkquality":255,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:info  2021-12-04 16:44:40: MQTT publish: topic 'zigbee2mqtt/Yale YMC420D', payload '{"action":"","battery":0,"linkquality":255,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:info  2021-12-04 16:44:40: MQTT publish: topic 'homeassistant/device_automation/0x00158d0006d6d31d/action_finger_unlock/config', payload '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x00158d0006d6d31d"],"manufacturer":"Yale","model":"YMC 420 D - Digital Lock (YMC 420 D)","name":"Yale YMC420D","sw_version":"02.01.03"},"payload":"finger_unlock","subtype":"finger_unlock","topic":"zigbee2mqtt/Yale YMC420D/action","type":"action"}'
Zigbee2MQTT:info  2021-12-04 16:44:41: MQTT publish: topic 'zigbee2mqtt/Yale YMC420D/action', payload 'finger_unlock'
Zigbee2MQTT:info  2021-12-04 16:44:42: MQTT publish: topic 'zigbee2mqtt/Yale YMC420D', payload '{"battery":0,"linkquality":255,"lock_state":"unlocked","state":"UNLOCK"}'
Zigbee2MQTT:info  2021-12-04 16:44:42: MQTT publish: topic 'zigbee2mqtt/Yale YMC420D', payload '{"battery":0,"linkquality":255,"lock_state":"unlocked","state":"UNLOCK"}'

I used the external converter as posted above:

```
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const constants = require("zigbee-herdsman-converters/lib/constants");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const extend = require("zigbee-herdsman-converters/lib/extend");
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
  action: {
    cluster: "closuresDoorLock",
    type: "raw",
    convert: (model, msg, publish, options, meta) => {
      const lookup = {
        0: "keypad_unlock",
        1: "zigbee_unlock",
        2: "manual_lock",
        3: "rfid_unlock",
        4: "finger_unlock",
        10: "auto_lock",
        14: "inside_unlock",
      };
      const value = lookup[msg.data[4]];
      if (value == "lock" || value == "inside_unlock") {
        return { action: value };
      } else {
        return { action: lookup[msg.data[3]] };
      }
    },
  },
};

const lockExtend = (meta) => {
  return {
    fromZigbee: [
      fz.lock_operation_event,
      fz.lock_programming_event,
      fz.lock_pin_code_response,
      fz.lock_user_status_response,
      fz.lock,
      fz.battery,
      fzLocal.action,
    ],
    toZigbee: [tz.lock, tz.pincode_lock, tz.lock_userstatus],
    meta: { pinCodeCount: 250, ...meta },
    exposes: [e.lock(), e.battery(), e.pincode()],
    configure: async (device, coordinatorEndpoint, logger) => {
      const endpoint = device.getEndpoint(1);
      await reporting.bind(endpoint, coordinatorEndpoint, [
        "closuresDoorLock",
        "genPowerCfg",
      ]);
      await reporting.lockState(endpoint);
      await reporting.batteryPercentageRemaining(endpoint);
    },
  };
};

module.exports = [
  {
    zigbeeModel: ["YMC 420 D"],
    model: "YMC 420 D",
    vendor: "Yale",
    description: "YMC 420 D - Digital Lock",
    extend: lockExtend({ timeout: 20000 }),
  },
];

```

rrlevy, 2021-12-04:
These are the logs for LOCKING and UNLOCKING via Zigbee2MQTT:

Info MQTT publish: topic 'zigbee2mqtt/Yale YMC 420D', payload '{"action":"zigbee_unlock","battery":null,"linkquality":255,"lock_state":"locked","state":"LOCK"}'
Info MQTT publish: topic 'zigbee2mqtt/Yale YMC 420D', payload '{"action":"","battery":null,"linkquality":255,"lock_state":"locked","state":"LOCK"}'
Info MQTT publish: topic 'zigbee2mqtt/Yale YMC 420D/action', payload 'zigbee_unlock'
Info MQTT publish: topic 'zigbee2mqtt/Yale YMC 420D', payload '{"battery":null,"linkquality":255,"lock_state":"unlocked","state":"UNLOCK"}'
Info MQTT publish: topic 'zigbee2mqtt/Yale YMC 420D', payload '{"battery":null,"linkquality":255,"lock_state":"locked","state":"LOCK"}'

Koenkk, 2021-12-05:
@rrlevy looks all is working?

rrlevy, 2021-12-05:
@Koenkk only the lock/unlock is working, but as @nieror mentioned, we can't get any information about the user unlocking the door.

Koenkk, 2021-12-05:
@rrlevy can you check if the configuring goes OK? You can force it via the frontend -> press yellow button on the device page. Before doing this make sure to wake up the lock otherwise it might fail due to the lock being asleep.

rrlevy, 2021-12-06:
I just tried pairing it again (I had uninstalled and removed it from my network).

Here is the log for the pairing process:

```
Info Device '0x00158d0006d6d31d' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0006d6d31d","ieee_address":"0x00158d0006d6d31d"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d0006d6d31d"},"type":"device_connected"}'
Info Starting interview of '0x00158d0006d6d31d'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0006d6d31d","ieee_address":"0x00158d0006d6d31d","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d0006d6d31d"},"type":"pairing"}'
Info Successfully interviewed '0x00158d0006d6d31d', device has successfully been paired
Info Device '0x00158d0006d6d31d' is supported, identified as: Yale YMC 420 D - Digital Lock (YMC 420 D)
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"YMC 420 D - Digital Lock","exposes":[{"features":[{"access":7,"description":"State of the lock","name":"state","property":"state","type":"binary","value_off":"UNLOCK","value_on":"LOCK"},{"access":1,"description":"Actual state of the lock","name":"lock_state","property":"lock_state","type":"enum","values":["not_fully_locked","locked","unlocked"]}],"type":"lock"},{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"features":[{"access":2,"description":"User ID to set or clear the pincode for","name":"user","property":"user","type":"numeric"},{"access":2,"description":"Type of user, unrestrictied: owner (default), (year|week)_day_schedule: user has ability to open lock based on specific time period, master: user has ability to both program and operate the door lock, non_access: user is recognized by the lock but does not have the ability to open the lock","name":"user_type","property":"user_type","type":"enum","values":["unrestricted","year_day_schedule","week_day_schedule","master","non_access"]},{"access":2,"description":"Whether the user is enabled/disabled","name":"user_enabled","property":"user_enabled","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"Pincode to set, set pincode to null to clear","name":"pin_code","property":"pin_code","type":"numeric"}],"name":"pin_code","property":"pin_code","type":"composite"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"YMC 420 D","options":[{"access":2,"description":"Expose pin of this lock in the published payload (default false).","name":"expose_pin","property":"expose_pin","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Yale"},"friendly_name":"0x00158d0006d6d31d","ieee_address":"0x00158d0006d6d31d","status":"successful","supported":true},"type":"device_interview"}'
Info Configuring '0x00158d0006d6d31d'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"YMC 420 D - Digital Lock","friendly_name":"0x00158d0006d6d31d","model":"YMC 420 D","supported":true,"vendor":"Yale"},"type":"pairing"}'
Info MQTT publish: topic 'homeassistant/lock/0x00158d0006d6d31d/lock/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x00158d0006d6d31d/set","device":{"identifiers":["zigbee2mqtt_0x00158d0006d6d31d"],"manufacturer":"Yale","model":"YMC 420 D - Digital Lock (YMC 420 D)","name":"0x00158d0006d6d31d","sw_version":"02.01.03"},"json_attributes_topic":"zigbee2mqtt/0x00158d0006d6d31d","name":"0x00158d0006d6d31d","state_locked":"LOCK","state_topic":"zigbee2mqtt/0x00158d0006d6d31d","state_unlocked":"UNLOCK","unique_id":"0x00158d0006d6d31d_lock_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x00158d0006d6d31d/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0006d6d31d"],"manufacturer":"Yale","model":"YMC 420 D - Digital Lock (YMC 420 D)","name":"0x00158d0006d6d31d","sw_version":"02.01.03"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x00158d0006d6d31d","name":"0x00158d0006d6d31d battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0006d6d31d","unique_id":"0x00158d0006d6d31d_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x00158d0006d6d31d/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0006d6d31d"],"manufacturer":"Yale","model":"YMC 420 D - Digital Lock (YMC 420 D)","name":"0x00158d0006d6d31d","sw_version":"02.01.03"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00158d0006d6d31d","name":"0x00158d0006d6d31d linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0006d6d31d","unique_id":"0x00158d0006d6d31d_linkquality_zigbee2mqtt","unit_of_measurement": [... cut ...] qi","value_template":"{{ value_json.linkquality }}"}'
Info MQTT publish: topic 'zigbee2mqtt/0x00158d0006d6d31d', payload '{"battery":0,"linkquality":255,"state":null}'
Info Successfully configured '0x00158d0006d6d31d'
```

And here the log for the forced configuration after paired:
```
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x00158d0006d6d31d"},"status":"ok","transaction":"5h6sl-3"}'
```

I used the same external converter as posted above

nieror, 2021-12-06:
Even tho I work with IT Infrastructure, I'm not a dev, but as far as I understand this codes, what really calls to return the PIN are these functions:
commandOperationEventNotification, commandProgrammingEventNotification, commandGetPinCodeRsp, commandGetUserStatusRsp, and for some reason I cannot see them anywhere....
Not sure that since this lock has the ZB Module integrated, if for some reason they've changed some key elements of the communication...
I think that the 1st indication of this "change" is that this lock does not report the regular commandOperationEventNotification. Instead is reports as 'RAW".
Not sure if my "insight" helps, but....that's what I've found so far....

@Koenkk I've enabled the herdsman-converters log and this is what I get unlocking the door with my fingerprint. 
Can't see any indication that the PIN / user is showing up anywhere on the logs.

Zigbee2MQTT:debug 2021-12-06 14:59:02: Received Zigbee message from 'YMC420D', type 'raw', cluster 'closuresDoorLock', data '{"data":[9,225,32,4,2,1,0,0,228,11,65,41],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-12-06 14:59:02: MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"action":"fingerprint","battery":88,"linkquality":128,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:info  2021-12-06 14:59:02: MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"action":"","battery":88,"linkquality":128,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:info  2021-12-06 14:59:02: MQTT publish: topic 'zigbee2mqtt/YMC420D/action', payload 'fingerprint'
2021-12-06T17:59:04.006Z zigbee-herdsman:adapter:ezsp:uart <-- [6035b1a97d312a15b658954b24ab5593499c9e5b104fb99874fade8683fc7d5e0fa5ed8e7e] [96,53,177,169,125,49,42,21,182,88,149,75,36,171,85,147,73,156,158,91,16,79,185,152,116,250,222,134,131,252,125,94,15,165,237,142,126]
2021-12-06T17:59:04.006Z zigbee-herdsman:adapter:ezsp:uart Recv DATA frame (6,0,0): 6035b1a9112a15b658954b24ab5593499c9e5b104fb99874fade8683fc7e0fa5ed8e7e0000
2021-12-06T17:59:04.006Z zigbee-herdsman:adapter:ezsp:uart Send ACK  frame (7)
2021-12-06T17:59:04.006Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e] [135,0,159,126]
2021-12-06T17:59:04.007Z zigbee-herdsman:adapter:ezsp:log <=== Frame: 77900145000004010101010100010000d07cbba277ffff0718e50a000030020643
2021-12-06T17:59:04.007Z zigbee-herdsman:adapter:ezsp:log <=== Application frame 69 (incomingMessageHandler) received: 0004010101010100010000d07cbba277ffff0718e50a000030020643
2021-12-06T17:59:04.008Z zigbee-herdsman:adapter:ezsp:log <=== Application frame 69 (incomingMessageHandler)   parsed: 0,EmberApsFrame: {"profileId":260,"clusterId":257,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":208},124,-69,30626,255,255,�
0
2021-12-06T17:59:04.008Z zigbee-herdsman:adapter:ezsp processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":257,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":208},"lqi":124,"rssi":-69,"sender":30626,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,229,10,0,0,48,2]}}
2021-12-06T17:59:04.014Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":229,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":48,"attrData":2}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":30626,"endpoint":1,"linkquality":124,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2021-12-06 14:59:04: Received Zigbee message from 'YMC420D', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-12-06 14:59:04: MQTT publish: topic 'zigbee2mqtt/YMC420D', payload '{"battery":88,"linkquality":124,"lock_state":"unlocked","state":"UNLOCK"}'
2021-12-06T17:59:04.892Z zigbee-herdsman:adapter:ezsp:uart <-- [7035b1a97d312a15b658954b24ab5593499c9f7b7d384fb99874face8583fc7d5e0fa5e8b97e] [112,53,177,169,125,49,42,21,182,88,149,75,36,171,85,147,73,156,159,123,125,56,79,185,152,116,250,206,133,131,252,125,94,15,165,232,185,126]
2021-12-06T17:59:04.892Z zigbee-herdsman:adapter:ezsp:uart Recv DATA frame (7,0,0): 7035b1a9112a15b658954b24ab5593499c9f7b184fb99874face8583fc7e0fa5e8b97e000000
2021-12-06T17:59:04.892Z zigbee-herdsman:adapter:ezsp:uart Send ACK  frame (0)
2021-12-06T17:59:04.893Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e] [128,112,120,126]
2021-12-06T17:59:04.893Z zigbee-herdsman:adapter:ezsp:log <=== Frame: 77900145000004010101010100010000d15cb3a277ffff0708e60a000030020374a0
2021-12-06T17:59:04.893Z zigbee-herdsman:adapter:ezsp:log <=== Application frame 69 (incomingMessageHandler) received: 0004010101010100010000d15cb3a277ffff0708e60a000030020374a0
2021-12-06T17:59:04.894Z zigbee-herdsman:adapter:ezsp:log <=== Application frame 69 (incomingMessageHandler)   parsed: 0,EmberApsFrame: {"profileId":260,"clusterId":257,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":209},92,-77,30626,255,255,�
0
2021-12-06T17:59:04.894Z zigbee-herdsman:adapter:ezsp processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":257,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":209},"lqi" [... cut ...]
2021-12-06T17:59:05.607Z zigbee-herdsman:adapter:ezsp:uart --> [8430fc7e] [132,48,252,126]
2021-12-06T17:59:05.608Z zigbee-herdsman:adapter:ezsp:log <=== Frame: 799001800042a277
2021-12-06T17:59:05.609Z zigbee-herdsman:adapter:ezsp:log <=== Application frame 128 (incomingRouteErrorHandler) received: 42a277
2021-12-06T17:59:05.609Z zigbee-herdsman:adapter:ezsp:log <=== Application frame 128 (incomingRouteErrorHandler)   parsed: 66,30626
2021-12-06T17:59:05.610Z zigbee-herdsman:adapter:ezsp:driver handleRouteError: nwk=30626, status=66

rrlevy, 2021-12-06:
One interesting thing that I also noticed is that after I pair the lock, it looses all the user configurations that were previously made. (fingerprints, codes, cards, etc). I have to add it all again.

I'm not sure if this is something being done by the lock itself or if it's related to the external converter code posted.

nieror, 2021-12-06:
> One interesting thing that I also noticed is that after I pair the lock, it looses all the user configurations that were previously made. (fingerprints, codes, cards, etc). I have to add it all [...]

Noticed that as well, but as far as I understand that would be a default behavior. Like, switching from local hardware management to a remote software management.

In which I just noticed that it's also something that is not reporting.
I may be wrong, but shouldn't any changes made to the lock configuration also be logged? really...not sure...just asking :)

rrlevy, 2021-12-06:
I got an answer for a comment from a Youtuber sponsored by Yale that this specific lock should receive a firmware update from Yale.
Maybe we need that for it to send those information we are missing.

[https://youtu.be/jd1HS4YryBs](https://youtu.be/jd1HS4YryBs
)

Koenkk, 2021-12-06:
@rrlevy that could be it indeed, if we have the firmware file we I can enable OTA updates from it from z2m.

nieror, 2021-12-06:
> @rrlevy that could be it indeed, if we have the firmware file we I can enable OTA updates from it from z2m.

"Problem" is that this lock is so new that its not even listed on Yale Access... so not sure where we could find it...

oguime, 2021-12-27:
I'm buying one hopping integration comes soon, because prices are going up...

dudu631, 2022-01-27:
Any news on this?

joaofelipes, 2022-02-03:
I just bought one of those and it will be great if it worked with z2m!

closed 2022-03-14 by github-actions[bot]

rrlevy, 2022-04-12:
This should still be open since it's still not working

cefranco76, 2022-04-19:
Hi everyone. I've just bought mine and I hope that at least a pairing I was suposed to do, but...
My Zigbee2MQTT didn't even noticed the lock and then a question has raised: All that work you guys had acomplished until now, are you using a yale hub to communicate to zigbee2MQTT or the connection happens directly between the lock and zigbee2mqtt?

cefranco76, 2022-05-04:
Hi guys, that's my solution.

I'm not an expert on that but I just realized that msg has on data[5] the action's User ID
Then I've created an attribute called userid with the returning value of msg.data[5].

And as a plus, I transformed the ID in names....

```

const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const constants = require("zigbee-herdsman-converters/lib/constants");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const extend = require("zigbee-herdsman-converters/lib/extend");
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
  action: {
    cluster: "closuresDoorLock",
    type: "raw",
    convert: (model, msg, publish, options, meta) => {
      const lookup = {
        0: "keypad_unlock",
        1: "zigbee_unlock",
        2: "manual_lock",
        3: "rfid_unlock",
        4: "finger_unlock",
        10: "auto_lock",
        14: "inside_unlock",
      };
      const value = lookup[msg.data[4]];
      if (value == "lock" || value == "inside_unlock") {
        return { action: value };
      } else {
        return { action: lookup[msg.data[3]] };
      }
    },
  },
  userid: {
    cluster: "closuresDoorLock",
    type: "raw",
    convert: (model, msg, publish, options, meta) => {
    return { userid: msg.data[5] };
    },
  },

  username: {
    cluster: "closuresDoorLock",
    type: "raw",
    convert: (model, msg, publish, options, meta) => {
      const userNames = {
        0: "",
        1: "Eduardo",
        2: "Ana Maria",
        94: "Teste UserID 94",
      };
    return { username: userNames[msg.data[5]] };
    },
  },
};

const lockExtend = (meta) => {
  return {
    fromZigbee: [
      fz.lock_operation_event,
      fz.lock_programming_event,
      fz.lock_pin_code_response,
      fz.lock_user_status_response,
      fz.lock,
      fz.battery,
	  fzLocal.userid,
	  fzLocal.username,
      fzLocal.action,
    ],
    toZigbee: [tz.lock, tz.pincode_lock, tz.lock_userstatus],
    meta: { pinCodeCount: 250, ...meta },
    exposes: [e.lock(), e.battery(), e.pincode(), e.lock_action()],
    configure: async (device, coordinatorEndpoint, logger) => {
      const endpoint = device.getEndpoint(1);
      await reporting.bind(endpoint, coordinatorEndpoint, [
        "closuresDoorLock",
        "genPowerCfg",
      ]);
      await reporting.lockState(endpoint);
      await reporting.batteryPercentageRemaining(endpoint);
    },
  };
};

module.exports = [
  {
    zigbeeModel: ["YMC 420 D"],
    model: "YMC 420 D",
    vendor: "Yale",
    description: "YMC 420 D - Digital Lock",
    extend: lockExtend({ timeout: 20000 }), //
  },
];
```

[image]

nieror, 2022-05-05:
booooooa! Way to go! 
Thanks for the addition. I'll be testing that as well.

dudu631, 2022-05-12:
> It seems that new models are coming with a firmware that works with zigbee2mqtt. Can anyone confirm this information?

Nope. Received a new model and it doesn't recognize. Will try @cefranco76 solution!

gregolin, 2022-05-20:
> Hi guys, that's my solution.
>
> I'm not an expert on that but I just realized that msg has on data[5] the action's User ID Then I've created an attribute called userid with the returning value of [...]

This script worked well, thanks. But I have noticed that the Zigbee2MQTT receives 3 or 4 messages for each action (lock or unlock), and they seem don't have any pattern to be identified (understanding which of them is the valid). Have you seen this behavioral?
Thanks.

cefranco76, 2022-05-20:
> This script worked well, thanks. But I have noticed that the Zigbee2MQTT receives 3 or 4 messages for each action (lock or unlock), and they seem don't have any pattern to be identified [...]

I presume that there's a sequence of messages like "Previous State", "Action Empty", "Action Command" and "Current State".

But nonetheless irrelevant to wellfunction. I would not worry about that. If the lock, hass, integration, all works fine, that's allright for me.

cefranco76, 2022-05-20:
Happy to help on something that I don't know. lol.

cefranco76, 2022-05-20:
> > > This script worked well, thanks. But I have noticed that the Zigbee2MQTT receives 3 or 4 messages for each action (lock or unlock), and they seem don't have any pattern to be identified (understanding which of them is the valid). Have you seen this behavioral? Thanks.
> > 
> > 
> > I presume that there's a sequence of messages like "Previous State", "Action Empty", "Action Command" and "Current State".
> > But nonetheless irrelevant to wellfunction. I would not worry about that. If the lock, hass, integration, all works fine, that's allright for me.
> 
> In fact it is working well and I can capture lock or unlock on Zibbee2MQTT... I only would like to capture who is unlocking as well, in order to send a notification to my cellphone, and these 3-4 messages in sequence are making the task harder :-)
> 
> Where do you see the usernames? I couldn't find them anywhere. I'd like to know who is unlocking it (even out of Zigbee2MQTT). Thanks.

This is already done. Look at this picture: 

[image]

It's my LOCK ENTITY and has its attributes, with "userid" that I entered in the LOCK and "username" that the EXTERNAL CONVERTER filled.
The "userid" is the same ID of user on LOCK. Look for it in the attributes of YOUR LOCK ENTITY.

To see it working, configure the user 001 on your LOCK, then, in the EXTERNAL CONVERTER replace your name where you see "Eduardo" and then, in "Developer Tools / Template", write it as it follows:

{{ state_attr('lock.REPLACE_BY_YOUR_LOCK_ENTITY','userid') }}
{{ state_attr('lock.REPLACE_BY_YOUR_LOCK_ENTITY','username') }}

gregolin, 2022-06-06:
> meta: { pinCodeCount: 250, ...meta },
>     exposes: [e.lock(), e.battery(), e.pincode(), e.lock_action()],

Hi @cefranco76 .
The script shows me error on these two lines (83 and 84). Is this correct?
Thanks.

gregolin, 2022-06-07:
Two images below:

[image]
[image]

Thanks.

cefranco76, 2022-06-08:
> Two images below:
> 
> [image]
> 
> [image]
> 
> Thanks.

@gregolin I didn't cacht yet what you're doing, but as far as I know we don't need to compile this code outside zigbee2mqtt. The only thing I did was save this code in a JS file, place it at zigbee2mqtt folder, add a external converter in the configs and  restart the addon.

gregolin, 2022-06-08:
I'm not compiling it. This screen comes from HA file editor :-)

cefranco76, 2022-06-08:
> I'm not compiling it. This screen comes from HA file editor :-)

ok, I misunderstood...
But I really don't know why you see this errors, as I mentioned before, I placed the js file, restart the addon and everything worked well.

gregolin, 2022-06-08:
Here everything is working with this model (YMC 420D), but I'm having problems with another Yale lock (YDF40), which is listed officially as supported but I'm having issues during the interview. I don't know if this script has been causing any issue or other problem... so I'm investigating everything, including these errors.

cefranco76, 2022-06-08:
If you remove this external converter the YDF40 works as expected?

> Here everything is working with this model (YMC 420D), but I'm having problems with another Yale lock (YDF40), which is listed officially as supported but I'm having issues during the interview. I [...]

diogofm7, 2022-07-04:
besides the sensors, this converter allows me to lock and unlock it by zigbee2mqtt???

is it working fine, or do we still have some bug??

dsantos21, 2022-08-31:
Hi @Koenkk ! Could this code integrate the main Zigbee2mqtt base? :D It was tested by everyone here, but unfortunately it wasn't committed :D

I was able to test it too. I think that the username translation shouldn't make to the final code, as its not possible for end-user's to edit it later.

The working code is here: https://github.com/Koenkk/zigbee2mqtt/issues/8946#issuecomment-1117818080

Thanks again!!!

dsantos21, 2022-08-31:
I've made some more tests and the main differences I could gather from the converter 'Yale.js' to the code proposed here is that most of the message is now RAW, instead of, for example,:

    lock_operation_event: {
        cluster: 'closuresDoorLock',
        type: 'commandOperationEventNotification',

With type = 'raw' it just works. And, of course, the mapping model is also different (as shown in the comment code block above).

I've tried my best to do this PR! Feel free to change everything :D. In the end, I believe that this PR will be worse to review than to do from scratch... :-D.

dsantos21, 2022-08-31:
Hi @nieror the proposed code doesn't include any hard code user mapping! The code block shown above was just my test as external converter, base to PR. The proposed code to merge is here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4581

Thanks for your help!

datsrt02, 2022-12-27:
> Hey everyone! I do agree that this code could be merged and integrated to the native app.
> 
> About the code above the main difference with mine is that I'm not "hardcoding" users. I'm using Home Assistant with Node Red to "convert" the userid. I thought it would be easier to manage and alter things if needed - I understand that for a residential use, there's not much to be altering, but for other uses, I think it would be better to do it so. Something like this:
> 
> userid: { cluster: "closuresDoorLock", type: "raw", convert: (model, msg, publish, options, meta) => { return { userid: msg.data[5] }; }, },
> 
> [image]
> 
> Anyway I`ll try to merge this code with the one I`m running to check the difference!

Hi @MrNiero , can u share your .json file in node red ?, i want to try with your solution.

MrNiero, 2022-12-27:
@datsrt02 here you go:

[{"id":"e40ec2dc94d12ffd","type":"tab","label":"Test","disabled":false,"info":"","env":[]},{"id":"c0d188bd36124da5","type":"mqtt in","z":"e40ec2dc94d12ffd","name":"YMC420D","topic":"zigbee2mqtt/YMC420D","qos":"2","datatype":"json","broker":"d86f5a43b71f7bd6","nl":false,"rap":true,"rh":0,"inputs":0,"x":100,"y":260,"wires":[["c0e9785084099ae3"]]},{"id":"2e0c4568054bad9a","type":"switch","z":"e40ec2dc94d12ffd","name":"Action","property":"payload.action","propertyType":"msg","rules":[{"t":"eq","v":"senha","vt":"str"},{"t":"eq","v":"aplicativo","vt":"str"},{"t":"eq","v":"cartao_RFID","vt":"str"},{"t":"eq","v":"digital","vt":"str"},{"t":"eq","v":"inside_unlock","vt":"str"},{"t":"eq","v":"manual_lock","vt":"str"},{"t":"eq","v":"auto_lock","vt":"str"}],"checkall":"false","repair":false,"outputs":7,"x":790,"y":260,"wires":[["ebab0e5231d7c8a0"],["ebab0e5231d7c8a0"],["ebab0e5231d7c8a0"],["ebab0e5231d7c8a0"],["ebab0e5231d7c8a0"],["9620350cd093c531"],["208906c7651de7cb"]]},{"id":"208906c7651de7cb","type":"function","z":"e40ec2dc94d12ffd","name":"Porta Trancada Automaticamente","func":"msg.payload = \n\n\"Porta da Sala Trancada Automaticamente.\"\n\nreturn msg","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":1100,"y":320,"wires":[[]]},{"id":"9620350cd093c531","type":"function","z":"e40ec2dc94d12ffd","name":"Porta Trancada Manualmente","func":"msg.payload = \n\n\"Porta da Sala Trancada Manualmente.\"\n\nreturn msg","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":1090,"y":280,"wires":[[]]},{"id":"ebab0e5231d7c8a0","type":"function","z":"e40ec2dc94d12ffd","name":"Porta Destrancada","func":"msg.payload = \n\nmsg.payload.username + \" destrancou a porta via \" + msg.payload.action + \".\"\n\nreturn msg","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":1050,"y":240,"wires":[[]]},{"id":"c0e9785084099ae3","type":"switch","z":"e40ec2dc94d12ffd","name":"UserID","property":"payload.action_user","propertyType":"msg","rules":[{"t":"eq","v":"0","vt":"str"},{"t":"eq","v":"1","vt":"str"},{"t":"eq","v":"2","vt":"str"},{"t":"eq","v":"3","vt":"str"},{"t":"eq","v":"4","vt":"str"}],"checkall":"false","repair":false,"outputs":5,"x":380,"y":260,"wires":[["dc1cf2809c54938f"],["e8cc6a10a1594229"],["4454d48746c8f9af"],["35b3c78e7b60a5cc"],["f52424a9a01c2c10"]]},{"id":"dc1cf2809c54938f","type":"change","z":"e40ec2dc94d12ffd","name":"Admin","rules":[{"t":"set","p":"payload.username","pt":"msg","to":"Admin","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":570,"y":180,"wires":[["2e0c4568054bad9a"]]},{"id":"e8cc6a10a1594229","type":"change","z":"e40ec2dc94d12ffd","name":"Renan","rules":[{"t":"set","p":"payload.username","pt":"msg","to":"Renan","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":570,"y":220,"wires":[["2e0c4568054bad9a"]]},{"id":"4454d48746c8f9af","type":"change","z":"e40ec2dc94d12ffd","name":"Helo","rules":[{"t":"set","p":"payload.username","pt":"msg","to":"Helo","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":570,"y":260,"wires":[["2e0c4568054bad9a"]]},{"id":"35b3c78e7b60a5cc","type":"change","z":"e40ec2dc94d12ffd","name":"Roberto","rules":[{"t":"set","p":"payload.username","pt":"msg","to":"Roberto","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":580,"y":300,"wires":[["2e0c4568054bad9a"]]},{"id":"f52424a9a01c2c10","type":"change","z":"e40ec2dc94d12ffd","name":"Vanderlei","rules":[{"t":"set","p":"payload.username","pt":"msg","to":"Vanderlei","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":580,"y":340,"wires":[["2e0c4568054bad9a"]]},{"id":"d86f5a43b71f7bd6","type":"mqtt-broker","name":"Mosquitto","broker":"localhost","port":"1883","clientid":"","autoConnect":true,"usetls":false,"protocolVersion":"5","keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","birthMsg":{},"closeTopic":"","closeQos":"0","closePayload":"","closeMsg":{},"willTopic":"","willQos":"0","willPayload":"","willMsg":{},"sessionExpiry":""}]

airtonac, 2023-07-16:
> This script worked well, thanks. But I have noticed that the Zigbee2MQTT receives 3 or 4 messages for each action (lock or unlock), and they seem don't have any pattern to be identified [...]

Hi there,
I know it's been some time... but
What did you do about receiving multiple mqtt messages. For some reason I don't se attributes on my lock (which is the YALE 420 A not the 420 D).
I’ve being looking around and the device indeed sends information back through MQTT, such as User ID, method of unlocking, time, etc. I intend to use all this information to create an automation which notifies me whenever there’s action at the door. For that I will need to retrieve the information from mqtt messages.

Now, here’s where I need help. I’m not familiar with processing MQTT messages and extracting the necessary data from topics, subtopics, and attributes. Additionally, the device sends multiple messages to the same topics (or else I think so). When looking at messages using an MQTT external software (MQTT Explorer) I noticed that the strings only stay available until the next message with empty values arrives. And that happens fast.

gregolin, 2023-07-21:
> 

Hi @cefranco76 
My device is working fine here for a long time, but now I'm trying to get the info/action when someone opens from inside, and I'm not having success.
Different than you entity, I can't read its state and opening it form inside, it seems none action is generated.
Do you have any other configuration for that?
Thanks.

Franco1212, 2023-09-08:
> Hi guys, that's my solution.
>
> I'm not an expert on that but I just realized that msg has on data[5] the action's User ID Then I've created an attribute called userid with the returning value of [...]

Hello, i'm using yours configuration, since a year ago in tho homes, its working fine, tkz..... but today, i'm trying to create a new pic code, and seems like som bug here, in my two homes having the same problem, when i put the User, comes a wrong msg that the user must be numbers, but i'm putting numbers... so anybody else has the same problem?

Montreal666, 2023-09-30:
> Hello, i'm using yours configuration, since a year ago in tho homes, its working fine, tkz..... but today, i'm trying to create a new pic code, and seems like som bug here, in my two homes having [...]

same problem....
reported here as well:
https://github.com/Koenkk/zigbee2mqtt/issues/18912

dudu631, 2024-04-05:
Guys version 1.36.1 [breaks legacy extend](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1).

I changed the code to the new version.

```
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");

const e = exposes.presets;

const fzLocal = {
  action_source_name: {
    cluster: "closuresDoorLock",
    type: "raw",
    convert: (model, msg, publish, options, meta) => {
      const lookup = {
        0: "password_unlock", //'unknown', // password 15/07
        1: "unlock", //'lock', unlock system 15/07
        2: "auto_lock", // 'unlock', // one touch lock and autolock 15/07
        3: "RFID_unlock", //'lock_failure_invalid_pin_or_id', // card 15/07
        4: "fingerprint_unlock", //'lock_failure_invalid_schedule',  // fingerprint 15/07
        5: "unlock_failure_invalid_pin_or_id",
        6: "unlock_failure_invalid_schedule",
        7: "one_touch_lock",
        8: "key_lock",
        9: "key_unlock",
        10: "auto_lock",
        11: "schedule_lock",
        12: "schedule_unlock",
        13: "manual_lock",
        14: "manual_unlock",
        15: "non_access_user_operational_event",
      };
      const value = lookup[msg.data[3]];
      return { action_source_name: value };
    },
  },
  action_source_user: {
    cluster: "closuresDoorLock",
    type: "raw",
    convert: (model, msg, publish, options, meta) => {
      return { action_source_user: msg.data[5] };
    },
  },
};

const deviceConfig = {
  zigbeeModel: ["YMC 420 D"],
  model: "YMC 420 D",
  vendor: "Yale",
  description: "YMC 420 D",
  exposes: [
    e.lock(),
    e.lock_action(),
    e.battery(),
    e.pincode(),
  ],
  fromZigbee: [
    fzLocal.action_source_user,
    fzLocal.action_source_name,
    fz.lock,
    fz.battery,
    fz.lock_operation_event,
    fz.lock_programming_event,
    fz.lock_pin_code_response,
    fz.lock_user_status_response,
  ],
  toZigbee: [
    tz.lock,
    tz.pincode_lock,
    tz.lock_userstatus,
  ],
  meta: {
    pinCodeCount: 250,
    configure: async (device, coordinatorEndpoint, logger) => {
      const endpoint = device.getEndpoint(1);
      await reporting.bind(endpoint, coordinatorEndpoint, [
        "closuresDoorLock",
        "genPowerCfg",
      ]);
      await reporting.lockState(endpoint);
      await reporting.batteryPercentageRemaining(endpoint);
    },
  },
};

module.exports = [deviceConfig];

```

Tested here and it seems to be working fine.

rmribeiro2024, 2024-04-05:
> Guys version 1.36.1 [breaks legacy extend](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1).
>
> I changed the code to the new version.
>
> ```
> const fz = [...]

Muito obrigado irmão. Estava penando aqui pra resolver depois desta atualização.
Caiu como uma luva. Valeu mesmo!

gregolin, 2024-04-07:
I'm having some issue related to cache, maybe. Although I sent the new code, the HA still keeps using the previous code (I'm sure about that because I changed the device name in the new code). When I ask for new devices, the device is listed as the old one, and as unsupported.
Does anyone know how could I force the HA to renew the scripts?
Thanks.

dudu631, 2024-04-07:
@gregolin the addon folder location changed. You should be replacing it on `config/zigbee2mqtt`. I did the same mistake a while ago trying to change the file on `share/zigbee2mqtt`

gregolin, 2024-04-07:
> @gregolin the addon folder location changed. You should be replacing it on `config/zigbee2mqtt`. I did the same mistake a while ago trying to change the file on `share/zigbee2mqtt`
> 
> Also make sure to restart the zigbee2mqtt addon.

Sorry, but config/zigbee2mqtt? Where is this folder? Is it readable by "File Editor"?
In my mind I had to add lines similar to these below in "/homeassistant/zigbee2mqtt/configuration.yaml", and after that, create these files in the same folder, not?

external_converters:
  - ymc420d.js
  - persiana.js

Thanks.

gregolin, 2024-04-08:
Hi everyone.
Does anyone know how could I refresh the external converter script? It seems still trying to get the old one, and every time I try to connect again, it removes all users, fingerprint and access code, and I don't need to say how much this is bad.
Thanks.

dudu631, 2024-04-08:
@gregolin yes you need to add the external_converters to the configuration inside the zigbee2mqtt folder and create the appropriate files in the same folder, so you are correct. That should be enough. I don't have the full view of what's happening in your scenario... There's not a cache when loading the converter, if you change the file and reload the addon, that's the one it should be getting.

I still think you might be changing some old zigbee2mqtt folder that's not the one used by your addon, thus not solving the problem.

gregolin, 2024-04-09:
It's strange for me.
The external file is being loaded correctly, because when I write a wrong file name, I see the error in the log. So, writing correctly and not error message in the log suppose me it's loaded ok.
The Z2M log says:

`Zigbee2MQTT:info  2024-04-08 21:35:12: MQTT publish: topic 'homeassistant/select/0x00158d0006dd799e/identify/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x00158d0006dd799e/set/identify","device":{"identifiers":["zigbee2mqtt_0x00158d0006dd799e"],"manufacturer":"Yale","model":"Automatically generated definition (YMC 420 D)","name":"0x00158d0006dd799e","sw_version":"02.01.06","via_device":"zigbee2mqtt_bridge_0x00124b001cd4a28c"},"device_class":"identify","enabled_by_default":false,"entity_category":"config","json_attributes_topic":"zigbee2mqtt/0x00158d0006dd799e","object_id":"0x00158d0006dd799e_identify","options":["identify"],"origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"unique_id":"0x00158d0006dd799e_identify_zigbee2mqtt"}'`

I don't know why this **Automatically generated definition (YMC 420 D)**. Does it make sense to you? Your log register is similar?
Thanks.

gregolin, 2024-04-09:
@Koenkk , please, after almost 2 years, could you help me here with the Yale YMC 420D?
After the last Z2M version, the previous external converter doesn't work anymore. Two users, as you can see above, are using a new version and it works fine. I'm still having problems to use it.
Considering this log attached, does it says anything to you? i have tried several joins and changes, but it seems my external converter isn't considered.
By the way I'm attaching the configuration.yaml and the external converter as well.
Could you help me here, please? I really don't know what could I do in order to put again this locker to work with HA.
Thanks a lot!

[log_yale.txt](https://github.com/Koenkk/zigbee2mqtt/files/14912285/log_yale.txt)
[configuration.yaml](https://github.com/Koenkk/zigbee2mqtt/files/14912301/configuration.txt)
[ymc420d_v2.js](https://github.com/Koenkk/zigbee2mqtt/files/14912302/ymc420d_v2.js.txt)

olianibelloto, 2024-05-22:
@gregolin have you managed to solve it? I am in the same situation here.

olianibelloto, 2024-05-22:
@

> It's strange for me. The external file is being loaded correctly, because when I write a wrong file name, I see the error in the log. So, writing correctly and not error message in the log suppose [...]

@gregolin have you managed to solve it? I am in the same situation here.

[35 more replies]