Koenkk/zigbee2mqtt #8953: New Install, first start failure

dpawson, 2021-10-02:
Zigbee2MQTT fails to start?
- https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

Similar to this. 

## What happened

New install zigbee2mqtt
Hit start. 

## What did you expect to happen

No errors

## How to reproduce it (minimal and precise)

Edit configuration. info / start
Log reads
> zigbee2mqtt@1.21.2 start
> node index.js
Zigbee2MQTT:error 2021-10-02 11:39:28: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-10-02 11:39:28: Failed to start zigbee
Zigbee2MQTT:error 2021-10-02 11:39:28: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-10-02 11:39:28: Exiting...
Zigbee2MQTT:error 2021-10-02 11:39:29: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.js:116:27)
    at start (/app/index.js:95:5)
[11:39:29] INFO: Handing over control to Zigbee2mqtt Core ...

## Debug info
Zigbee2MQTT version:    1.21.2-1 
Adapter hardware: Conbee II
Adapter firmware version:  ?? How to find? Device one week old, auto update, hence assume most recent.

Koenkk, 2021-10-03:
Try adding the following to your configuration.yaml;

```yaml
serial:
  adapter: deconz
```

(last point of https://www.zigbee2mqtt.io/information/supported_adapters.html#notes)

dpawson, 2021-10-03:
addition to configuration.yaml failed with

Setup failed for serial: No setup or config entry setup function defined.
11:53:06 AM – (ERROR) setup.py

dpawson, 2021-10-04:
Setup is 
rpi 4b with HA eprom.
Conbee II.

Question. Should I install deCONZ - is this needed to talk to Philips Hue devices?

closed 2021-11-12 by github-actions[bot]