Koenkk/zigbee2mqtt #8956: Report MQTTs connection errors in logfile

andreasbrett, 2021-10-02:
Could you please log when a MQTTs connection can not be established or implement a `max_retries` setting? Today i restarted z2m because of the new release. The container went up and could talk to zigbee devices but a) the frontend didn't load and b) I could not communicate with z2m via MQTT.

Checking the logs I couldn't find the issue as there was no sign of a bad connection to the MQTT server even when using debug log level:

info  2021-10-02 13:38:23: Connecting to MQTT server at mqtts://REDACTED.REDACTED.COM:8883
debug 2021-10-02 13:38:23: MQTT SSL/TLS: Path to CA certificate = /tls/ca.crt
debug 2021-10-02 13:38:23: Using MQTT login with username: zigbee2mqtt
debug 2021-10-02 13:38:23: Using MQTT client ID: 'zigbee2mqtt'

In my MQTT server's log I found z2m to re-connect over and over again. What I failed to do was update the root CA certificate for my MQTT server (see https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/). So this was not a z2m error but the logfile of z2m made it hard to find the error as everything looked fine.

I propose to either have a setting for maximum connection retries and/or log when a MQTT connection needs to be retried incl. the reason why (SSL verification error etc). z2m currently only handles onConnect and onMessage events but misses out on important events in the onError, onReconnect and onClose events.

Koenkk, 2021-10-03:
Can you check if this has been fixed in the latest-dev?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

andreasbrett, 2021-10-03:
Thanks for your quick implementation. Unfortunately it does not work. z2m still hangs in a loop trying to connect. The z2m frontend doesn't get startet at all which is a good sign that z2m is stuck at boot. Your implementation would also just log out the issue not stop z2m from looping nor describe the actual connection error.

What I tried with some dirty code work inside the docker container was to add an onError event handler like this:

1) add event handler assignment around here: https://github.com/Koenkk/zigbee2mqtt/blob/45a4978ca8a1db663d31028cbbb20ce9d492e8fa/lib/mqtt.ts#L79

`this.client.on('error', this.onError);`

2) implement the event handler to log out the actual error (and stop z2m from looping):

```
async onError(error) {
        logger.error('Error connecting to MQTT server: ' + error);
        # do something to stop z2m here as connection can't be established
        # i don't know how though... sorry
}
```

I'm not very experienced with async/await/promises and how the mqtt lib integrates with the main z2m code in order to quit z2m.

andreasbrett, 2021-10-03:
BTW: the event message thrown in an error is quite descriptive. For my case the actual logging was:

`Error connecting to MQTT server: certificate has expired`

which is on point.

Koenkk, 2021-10-03:
Could you check if the error is logged with the changes from https://github.com/Koenkk/zigbee2mqtt/pull/8988/files ?

andreasbrett, 2021-10-04:
> Could you check if the error is logged with the changes from https://github.com/Koenkk/zigbee2mqtt/pull/8988/files ?

I tried but I could not find the logging on the console. Also adding a dedicated message via `logger.error(err);` did not show up in the logs.

Koenkk, 2021-10-04:
Can you check again with the latest dev?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

andreasbrett, 2021-10-05:
Yes, that works and is how I would expect it to behave. 

```
info  2021-10-05 20:18:37: Connecting to MQTT server at mqtts://redacted.example.com:8883
debug 2021-10-05 20:18:37: MQTT SSL/TLS: Path to CA certificate = /tls/ca.crt
debug 2021-10-05 20:18:37: Using MQTT login with username: zigbee2mqtt
debug 2021-10-05 20:18:37: Using MQTT client ID: 'zigbee2mqtt'
error 2021-10-05 20:18:37: MQTT failed to connect: certificate has expired
error 2021-10-05 20:18:37: Exiting...
info  2021-10-05 20:18:37: Stopping zigbee-herdsman...
info  2021-10-05 20:18:38: Stopped zigbee-herdsman
```

closed 2021-10-05 by andreasbrett