Koenkk/zigbee2mqtt #8958: All sensor values are no more updated since last dev commit

ALaDoffe, 2021-10-02:
## What happened

Since the last dev commit 2fb59f8efa9d8e97ad34401dc2612ff3f55863f1 sensor values are no more updated.
We can see new line in log, new message in mosquitto but values are always the same.

Mosquitto
Client mosqsub|1776-raspberryp received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x00158d000349c149', ... (142 bytes))
{"battery":100,"humidity":52.46,"last_seen":"2021-10-02T15:17:53+02:00","linkquality":63,"pressure":1007.1,"temperature":24.47,"voltage":3005}

Z2M Log
debug Received Zigbee message from '0x00158d000349c149', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.weather"}' from endpoint 1 with groupID 0
info MQTT publish: topic 'zigbee2mqtt/0x00158d000349c149', payload '{"battery":100,"humidity":52.46,"last_seen":"2021-10-02T15:17:53+02:00","linkquality":63,"pressure":1007.1,"temperature":24.47,"voltage":3005}'

Same issue for all devices except lights and switches

## Debug info
Zigbee2MQTT version: latest dev  commit 2fb59f8efa9d8e97ad34401dc2612ff3f55863f1
Adapter hardware: CC2652P
Adapter firmware version:

NickDomoticz, 2021-10-03:
I have the same issue with non updating values (Xioami WSDCGQ11LM). However not all values are affected and still get updated.

snippem, 2021-10-04:
Had the same issue i removed the last seen comment  from my configuration .yaml and all the issues are gone.
What i experienced where aqara door sensors not responding and aqara climate sensors not updating.

Koenkk, 2021-10-05:
Can you provide the debug log of this?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

Koenkk, 2021-10-06:
@ALaDoffe besides `last_seen`, did you also set `debounce`?

closed 2021-10-08 by Koenkk