Koenkk/zigbee2mqtt #8959: External converters are ignored with 1.21.2

okastl, 2021-10-02:
## What happened
External converter, working fine with 1.21.1, stopped working with 1.21.2. It is ignored. The file is loaded, if I add a
throw new Error('Whoops!');
zigbee2mqtt will halt during startup.
## What did you expect to happen
It should work.
## How to reproduce it (minimal and precise)
Add an external converter as described in the documentation.
Unfortunately it may work at first, which makes reproducing the problem difficult.
## Debug info
Zigbee2MQTT version: 1.21.2
Adapter hardware: CC26X2R (slash)
Adapter firmware version: 20210708

systemiii, 2021-10-02:
I have the same issue

fiodorow, 2021-10-03:
I am experiencing the same issue with external converter [ptvo counter 2ch](https://modkam.ru/2021/04/29/schetchik-impulsov-zigbee/?fbclid=IwAR1DVzTK6l4qRMhy-OLhXQB5pr0baGxdv0i2v1ujYfZN43Cj7xN57Fk7p0I) which works fine on 1.21.1 but fails on 1.21.2. 

Log error that might be helpful:

(node:365) UnhandledPromiseRejectionWarning: AssertionError [ERR_ASSERTION]: Endpoint name 'l5' is given but device is unsupported
    at Zigbee.resolveEntityLegacy (/app/lib/zigbee.ts:393:31)
    at Groups.onStateChange (/app/lib/extension/groups.js:118:29)
    at EventEmitter.emit (events.js:400:28)
    at EventBus.emit (/app/lib/eventBus.ts:132:22)
    at State.set (/app/lib/state.js:90:23)
    at Controller.publishEntityState (/app/lib/controller.js:326:37)
    at Controller.start (/app/lib/controller.js:170:26)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at start (/app/index.js:95:5)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:365) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:365) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

PotatoMill, 2021-10-03:
I also have the same issue, it even seems like a switch that should not be affected by the external converters stops working when they are activated

Koenkk, 2021-10-03:
Is this issue still present in the latest dev branch (if yes please provide logs of this)? https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

okastl, 2021-10-03:
> Is this issue still present in the latest dev branch (if yes please provide logs of this)?

Yes, it is.
I have attached debug log (which doesn't show any error) and the converter.
TS130F _TZ3000_8kzqqzu4 isn't picked. The standard Tuya converter is used. If I comment out the standard Tuya converter, the device is "unsupported". If I put the converter in a new manufacturer.js file in zigbee_herdsman_converters/devices, it works fine.
[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/7274586/log.zip)
[QS-Zigbee-C01-2.zip](https://github.com/Koenkk/zigbee2mqtt/files/7274587/QS-Zigbee-C01-2.zip)

Koenkk, 2021-10-04:
@okastl can you check if this is fixed now in the latest-dev?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

okastl, 2021-10-04:
> @okastl can you check if this is fixed now in the latest-dev?

Yes, the problem is fixed in the latest dev. Thank you!

closed 2021-10-05 by Koenkk