Koenkk/zigbee2mqtt #8961: this.mqtt.emit() no longer supported in onZigbeeEvent() of user extension in 1.21.2

okastl, 2021-10-02:
## What happened
1.21.2 breaks existing user extensions, which use this.mqtt.emit() in onZigbeeEvent()
When called, the following error is thrown:
Failed to call 'StorenSwitch' 'onZigbeeEvent' (TypeError: this.mqtt.emit is not a function at StorenSwitch.onZigbeeEvent (/opt/zigbee2mqtt/dist/util/externally-loaded.js:51:27) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:427:81) at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:297:9))

The call is
this.mqtt.emit('message', {topic: `${this.mqttBaseTopic}/${target_device}/set`, message: cmd});
from an onZigbeeEvent() in a user extension.

The extension worked fine in 1.21.1

## What did you expect to happen
That it continues to work.

## How to reproduce it (minimal and precise)
Create a user extension, which uses this.mqtt.emit() from onZigbeeEvent()
e.g. https://github.com/Koenkk/zigbee2mqtt/discussions/7642
## Debug info
Zigbee2MQTT version: 1.21.2
Adapter hardware:  CC26X2R1 (slash)
Adapter firmware version: 20210708

okastl, 2021-10-02:
A workaround is to change
this.mqtt.emit('message', {topic: ${this.mqttBaseTopic}/${target_device}/set, message: cmd});
to
this.mqtt.onMessage(`${this.mqttBaseTopic}/${target_device}/set`, cmd);

Maybe this was changed intentionally

Koenkk, 2021-10-03:
This is indeed expected due to a large refactor. The following will be added to the next z2m release release notes;

[image]

https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/information/user_extensions.md has been updated already

closed 2021-10-03 by Koenkk