Koenkk/zigbee2mqtt #8963: Multiple problems since version 1.21.2-1

sygys, 2021-10-02:
First the mi smartplug gives the error:

Payload '{"auto_off":false,"consumer_connected":true,"consumption":135.39,"current":0,"energy":135.39,"led_disabled_night":false,"linkquality":72,"overload_protection":null,"power":0,"power_outage_memory":true,"state":"ON","temperature":23,"update":{"state":"idle"},"update_available":false,"voltage":226}' is not a Number

Since the last update.

Second of all the overload protection doesnt go further then 2200 watt while this plug can take loads up to 2300. as can be found here: https://cdn-reichelt.de/documents/datenblatt/R800/XIAOMI_XM200016_DB-DE.pdf

Also i have tried loads of things to get rid of the json click error. but even the legacy change in the configuration.yaml doesnt help.

Also i keep getting around 50 messages a day about invalid color mode. is there any news on when that is going to be fixed or what it is exactly?

Im reverting back to the previous version until the first 2 errors are fixed. I cant use the smartplug anymore when the max overload is set too low.

aegjoyce, 2021-10-03:
I too am getting lots of these errors, for me it's with a pair of Moes BHT-002 thermostats. Only started since it updated today:

```
Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:146 
Integration: MQTT (documentation, issues) 
First occurred: 20:41:45 (24 occurrences) 
Last logged: 20:45:00

Payload '{"child_lock":null,"current_heating_setpoint":18,"deadzone_temperature":null,"heat":"OFF","linkquality":255,"local_temperature":20.5,"max_temperature_limit":null,"preset":"hold","preset_mode":"hold","running_state":"idle","system_mode":"heat"}' is not a Number
Payload '{"child_lock":null,"current_heating_setpoint":21,"deadzone_temperature":null,"heat":"OFF","linkquality":255,"local_temperature":23,"max_temperature_limit":null,"preset":"hold","preset_mode":"hold","running_state":"idle","system_mode":"heat"}' is not a Number
```
Will try reverting to see if it fixes it.

callifo, 2021-11-02:
Same issue for me, since the update to 1.22

```
2021-11-02 21:24:45 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"battery":100,"device":{"applicationVersion":2,"dateCode":"20190219","friendlyName":"motion_sensor_powder_room","hardwareVersion":1,"ieeeAddr":"0x0017880109187b36","manufacturerID":4107,"manufacturerName":"Philips","model":"9290012607","networkAddress":63273,"powerSource":"Battery","softwareBuildID":"6.1.1.27575","stackVersion":1,"type":"EndDevice","zclVersion":1},"illuminance":0,"illuminance_lux":0,"last_seen":"2021-11-02T21:24:45+10:30","led_indication":null,"linkquality":120,"motion_sensitivity":null,"occupancy":false,"occupancy_timeout":null,"temperature":23.74,"update":{"state":"idle"},"update_available":false}' is not a Number
```

Seems to be happening specifically on that powder room sensor only, even though I have half a dozen of them in total.

closed 2021-12-10 by github-actions[bot]