Koenkk/zigbee2mqtt #8966: Reduce reporting interval on the TS0601 air quality sensor

CornerBit, 2021-10-03:
## What happened

The device reports sensor values too frequently at approximately 200ms intervals. Below is an example of the reports I see:
```
info MQTT publish: topic 'zigbee2mqtt/0x<snip>', payload '{"co2":1,"elapsed":202,"formaldehyd":358,"humidity":56,"last_seen":"2021-10-03T01:33:07.265Z","linkquality":24,"temperature":24,"voc":0}'
info MQTT publish: topic 'zigbee2mqtt/0x<snip>', payload '{"co2":1,"elapsed":199,"formaldehyd":358,"humidity":56,"last_seen":"2021-10-03T01:33:07.465Z","linkquality":18,"temperature":24,"voc":0}'
info MQTT publish: topic 'zigbee2mqtt/0x<snip>', payload '{"co2":1,"elapsed":202,"formaldehyd":358,"humidity":56,"last_seen":"2021-10-03T01:33:07.667Z","linkquality":27,"temperature":24,"voc":0}'
```

I'm not sure how to reduce the reporting interval on the device. I can see the `manuSpecificTuya` cluster but no attributes are listed for it under the `Reporting` tab in the UI. I noticed a few other people have similar reporting interval issues in #7299. Any assistance with this would be greatly appreciated!

## Debug info
Zigbee2MQTT version: 1.21.2
Adapter hardware: CC26X2R1

CornerBit, 2021-11-07:
Still seeing this issue in the latest version.

kkossev, 2021-11-08:
@CornerBit 
Even when paired to Tuya gateway,  this crappy device bombards the Zigbee network every second!
I have switched mine off until a device firmware update is available that will allow configuration of the reporting intervals (highly unlikely..)

CornerBit, 2021-11-10:
@kkossev thanks for confirming it's a bug on the device end. Hopefully they do release an update, but I have my doubts...

RuneNyhuus, 2021-11-19:
Yeah i have the same problem...
the zigbee2mqtt is running like crazy with this device turned on.

VikeDragon, 2021-11-19:
+ Same problems with T0601_air_quality round box

dedors, 2021-11-24:
My device never goes below 4 messages/sec and spams up to 10/s. It's pretty crazy. (TZE200_yvx5lh6k / round model)

harryohalloran, 2021-12-02:
> @CornerBit
> Even when paired to Tuya gateway, this crappy device bombards the Zigbee network every second!
> I have switched mine off until a device firmware update is available that will allow configuration of the reporting intervals (highly unlikely..)

Where do you even find the latest firmware updates?  I was having trouble locating on the Tuya site.  I'm having the same issue and it's causing problems with other devices based on the high amount of traffic on the network.

kkossev, 2021-12-03:
> Where do you even find the latest firmware updates? I was having trouble locating on the Tuya site. I'm having the same issue and it's causing problems with other devices based on the high amount of traffic on the network.

Firmware updates are available sometimes when the device is paired to Tuya gateway. But currently, there are no updates for TS0601.

There is something that I can't understand. The company (companies) that produce this crap should have become broke already - Tuya charges the manufacturers per the number of the API calls or messages sent to the cloud servers  - https://developer.tuya.com/en/docs/iot/membership-service?id=K9m8k45jwvg9j

aaamoeder, 2022-01-11:
also noticing this. Is there no way to limit this ? I have 2 and it's very taxiing on my gateway.

jamesanf, 2022-01-19:
also researching this too – i get an update every 2-3 seconds on every sensor in the device. it's not needed and is flooding my network. while i wait for a firmware update, is it at least possible for zigbee2mqtt to increase the interval it reports these values to HA?

aaamoeder, 2022-01-19:
> also researching this too – i get an update every 2-3 seconds on every sensor in the device. it's not needed and is flooding my network. while i wait for a firmware update, is it at least possible for zigbee2mqtt to increase the interval it reports these values to HA?

I have currently excluded these sensors from the recorder to relieve stress on the Database and have created filter sensors for the ones I like to have historical data for.
Not a fix, but a workable workaround.

jamesanf, 2022-01-19:
> 

That sounds sort of like what I need to do. Does by filter sensor do you mean a derivative sensor for the overactive ones? If so would you mind sharing your code? I'm interested in how you reduced the update rate of the new sensor

jamesanf, 2022-01-19:
> I have currently excluded these sensors from the recorder to relieve stress on the Database and have created filter sensors for the ones I like to have historical data for. Not a fix, but a workable [...]

I also wonder if you have tried to use the debounce feature in zigbee2mqtt's individual device.yaml settings? I've tried it, but anything above 'debounce: 2' kills all MQTT reporting altogether. But maybe I'm using it wrong?

aaamoeder, 2022-01-20:
> > 
> 
> That sounds sort of like what I need to do. Does by filter sensor do you mean a derivative sensor for the overactive ones? If so would you mind sharing your code? I'm interested in how you reduced the update rate of the new sensor

look into the [recorder ](https://www.home-assistant.io/integrations/recorder/)and [filter sensors](https://www.home-assistant.io/integrations/filter/#configuration-variables) ;-)

bartplessers, 2022-01-24:
> above 'debounce: 2' kills all MQ

same here. It's 'better" but still to much messages.

jamesanf, 2022-01-24:
> > above 'debounce: 2' kills all MQ
> 
> same here. It's 'better" but still to much messages.

It basically meant that the device was dead, except for a seemingly random (and seemingly inaccurate) update 2-3 times a day. In the end I gave up; I don't know why the debounce broke it, everything I read seemed to suggest that it would just reduce the messages not eliminate them. I experimented with a few different debounce values before scrapping the device entirely. It will sit in a drawer until somebody can find a good fix – hopefully soon!

All in all, not worth the effort for a cheap and probably not very accurate sensor...

aaamoeder, 2022-02-28:
is there any news on this ?
I've had to disconnect our sensors due to zigbee network problems caused by the sensors overloading the network apparently..

JoryHogeveen, 2022-04-08:
Same issue here, wen't through the available clusters and even the Tuya docs but couldn't find anything relative to this topic...

jamesanf, 2022-04-08:
> Same issue here, wen't through the available clusters and even the Tuya docs but couldn't find anything relative to this topic...
> 
> Related: [zigpy/zha-device-handlers#1470](https://github.com/zigpy/zha-device-handlers/issues/1470) Related: #8154 Related: #8220
> 
> @Koenkk Maybe reopen this topic as it's still an issue?

I also hit the same brick walls. After my last reply to this thread I told the supplier, who then just gave me the money back straight away no questions asked – that makes me think it's a pretty common/known issue. I'd still love to have it back in my network so do post the fix here if you find one. 

And if there's a better/working air quality sensor in a similar price band, I'm all ears.

closed 2022-04-16 by github-actions[bot]

Saharok14, 2023-06-12:
The same problem... Does anyone have a workaround?

fonsmeesters, 2023-07-05:
Same problem here. I have 4 of these sensors and my home assistant database is getting out of control....

kbickar, 2023-07-25:
Seeing the same issue with https://www.zigbee2mqtt.io/devices/TS0601_smart_human_presence_sensor_1.html getting a message every second

elupus, 2023-11-26:
Seem the issue also exist with ZHA integration in HA. So maybe not anything that can be done about it?

McGiverGim, 2023-12-29:
I've just bought this device :( and the problem remains...

I have added a "debounce" value of 100 but with "ignore debounce" for "temperature", in this way at least I spam a lot less HA, only when "temperature" changes... It continues being a lot but is a lot less than default. But this is only from z2m to HA, the device continues spamming the zigbee network.

Another question: in theory according to the manual:
- The value of fomaldehyde must be between 0 and 10, and I have values of more than 100 and the habitual must be about 0.02 or less.
- The value of VOC must be between 0 and 99.0, and I have values of 500 and the habitual must be about 0.1 or less.

Do you have this high values too? To me it seems the units are wrong and they must be converted, probably dividing by 1000.

amateurishere, 2024-01-19:
Any updates yet?

Koenkk, 2024-01-19:
AFAIK it is not possible to reduce the reporting interval of these TuYa devices, so not fixable from Z2M

kbickar, 2024-01-19:
Is it possible to rate limit the messages sent over MQTT from the device?

Koenkk, 2024-01-19:
Using the [`debounce`](https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#common-device-options) option.

RuneNyhuus, 2024-01-19:
debounce works great!

crowbarsolutions, 2024-06-20:
What value are you setting debounce to that shows success?

crowbarsolutions, 2024-08-12:
I ended up putting a smart outlet to power this sensor and setting up an automation in home assistant to turn it on for 5 seconds once an hour. It's primitive, but does the trick.

Just be aware to only use the latest value or max value when doing any reporting. Since in the case of my sensor it will start spitting out a default reading for a few seconds before starting to report the actual values in the air.

```
alias: Schedule - Throttle Carbon Dioxide Sensor
description: ""
trigger:
  - platform: time_pattern
    hours: /1
action:
  - service: switch.turn_on
    metadata: {}
    data: {}
    target:
      entity_id: switch.co2_sensor
  - delay:
      hours: 0
      minutes: 0
      seconds: 15
      milliseconds: 0
  - repeat:
      sequence:
        - delay:
            hours: 0
            minutes: 0
            seconds: 5
            milliseconds: 0
      until:
        - condition: template
          value_template: >-
            {{ states.sensor.carbon_dioxide_co2.last_changed >
            state_attr('automation.schedule_throttle_carbon_dioxide_sensor',
            'last_triggered') }}
  - service: switch.turn_off
    metadata: {}
    data: {}
    target:
      entity_id: switch.co2_sensor
mode: restart
```

ivanfmartinez, 2024-08-12:
> I ended up putting a smart outlet to power this sensor and setting up an automation in home assistant to turn it on for 5 seconds once an hour. It's primitive, but does the trick.

I made a small change in code to ignore reports from the "SPAMMER" devices. Not a clean job but easier to make

https://github.com/Koenkk/zigbee2mqtt/issues/17984#issuecomment-2267638647

Wheemer, 2024-08-20:
> > I ended up putting a smart outlet to power this sensor and setting up an automation in home assistant to turn it on for 5 seconds once an hour. It's primitive, but does the trick.
> 
> I made a small change in code to ignore reports from the "SPAMMER" devices. Not a clean job but easier to make
> 
> [#17984 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/17984#issuecomment-2267638647)

How can I implement this change?

ivanfmartinez, 2024-08-20:
> How can I implement this change?

I made the changes directly in the indicated file on my docker container. As there are a lot of cases like this and no recommended solution to be used. A better sollution will be to have it configurable like debouncer but I did not have time to check how to make all necessary changes in the UI.

ft276, 2024-08-23:
Is it possible to set up an automation that sets debounce to 0 for 2s every minute and the rest of the time sets it to whatever (even 0.01 doesn't allow any reporting with my TS0601)?
I can't see how to set such advanced settings except through the settings page :(

TomNOYB, 2024-10-27:
I second this ^
My tuya TS0601 Model
 [ZN2S-RS3E-DH] https://www.zigbee2mqtt.io/devices/ZN2S-RS3E-DH.html#zemismart-zn2s-rs3e-dh)
makes my network unusable when left chatting like crazy. Any debounce value solves the issue, but blocks all communication from the device.

CrazyVito11, 2024-12-28:
Maybe it could be possible to set a rate limiter on a device from Z2M?
This won't help with the amount of Zigbee messages, but it would help the HA server that is behind Z2M.

The device could send messages as fast as it wants to, but Z2M only allows a message to pass every X seconds, no matter what kind of update it is. This won't be a solution for something like a presence detection sensor, but it would be an acceptable solution for the air quality sensor.

CrazyVito11, 2024-12-28:
Just noticed that it seems like someone has already made a PR to pretty much implement what I said :tada: 

https://github.com/Koenkk/zigbee2mqtt/pull/24122

voidzero, 2025-02-13:
Thanks. Just noticed the spam from this crazy device.

tuongbui15, 2025-02-16:
You can try this config in YAML

devices:
  '0x00124b0025112233':  # Your device IEEE 
    friendly_name: 'Zemis dimmer 3 gang'
    debounce: 500  # only send MQTT after 500ms if any changes
    reporting:
      brightness_l1:
        min: 5
        max: 600
        change: 10
      brightness_l2:
        min: 5
        max: 600
        change: 10
      brightness_l3:
        min: 5
        max: 600
        change: 10
.... (add your entities)

GrumpyMeow, 2025-07-05:
Using the "debounce" option also does not work for me. It stops the reporting altogether. I can imagine the debounce-algorithm only allows messages after not receiving a message for a period of 500ms. But this is just a suspicion. Not sure though. I've now chosen to go for the "throttle" option with a value of 5seconds. This at least won't inflate my Home assistant database.
This seems like a partial solution as the Zigbee network is still being put under (some) pressure.

phyzical, 2025-10-20:
@GrumpyMeow did you find this still degraded network or was the debouncing enough to make it chill out mesh wise?

McGiverGim, 2025-11-15:
The debounce work for me in acceptable manner, but the new throttle is far better for this