Koenkk/zigbee2mqtt #8971: SNZB-02 fails to send temperature msTemperatureMeasurement failed (AREQ - ZDO - bindRsp after 10000ms)

tvrulesme, 2021-10-03:
## Debug info
Zigbee2MQTT version: version 1.21.2 (commit #44922120)
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20201127

Supported device SNZB-02 pairs successfully but shows the following error when it attempts to broadcast a temperature reading
`Zigbee2MQTT:error 2021-10-03 10:23:10: Configure failed: Error: Bind 0x00124b0022cf704d/1 msTemperatureMeasurement from '0x00124b00193691c1/1' failed (AREQ - ZDO - bindRsp after 10000ms)`

configuration.yaml:
```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.0.2
  user: mqtt_user
  password: Penguin1
serial:
  port: /dev/ttyACM0
devices:
  '0x00124b0022cf704d':
    friendly_name: '01-humidity'
    retain: true
```

logs:
```
Zigbee2MQTT:info  2021-10-03 10:12:49: Starting Zigbee2MQTT version 1.21.2 (commit #44922120)
Zigbee2MQTT:info  2021-10-03 10:12:49: Starting zigbee-herdsman (0.13.149)
Zigbee2MQTT:info  2021-10-03 10:12:58: zigbee-herdsman started (restored)
Zigbee2MQTT:info  2021-10-03 10:12:58: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:info  2021-10-03 10:12:58: Currently 0 devices are joined:
.......

info  2021-10-03 10:18:39: Device '0x00124b0022cf704d' joined
info  2021-10-03 10:18:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0022cf704d","ieee_address":"0x00124b0022cf704d"},"type":"device_joined"}'
info  2021-10-03 10:18:39: Starting interview of '0x00124b0022cf704d'
info  2021-10-03 10:18:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0022cf704d","ieee_address":"0x00124b0022cf704d","status":"started"},"type":"device_interview"}'
info  2021-10-03 10:18:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b0022cf704d"},"type":"device_connected"}'
info  2021-10-03 10:18:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b0022cf704d"},"type":"pairing"}'
info  2021-10-03 10:18:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0022cf704d","ieee_address":"0x00124b0022cf704d"},"type":"device_announce"}'
info  2021-10-03 10:18:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b0022cf704d"},"type":"device_announced"}'
info  2021-10-03 10:18:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[],"type":"groups"}'
info  2021-10-03 10:18:40: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"dateCode":"20201127","friendly_name":"Coordinator","ieeeAddr":"0x00124b00193691c1","lastSeen":1633252720100,"networkAddress":0,"softwareBuildID":"zStack12","type":"Coordinator"},{"description":"-","friendly_name":"0x00124b0022cf704d","ieeeAddr":"0x00124b0022cf704d","lastSeen":1633252720050,"networkAddress":59037,"type":"Unknown","vendor":"-"}]'
info  2021-10-03 10:18:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[{"dateCode":"20201127","friendly_name":"Coordinator","ieeeAddr":"0x00124b00193691c1","lastSeen":1633252720166,"networkAddress":0,"softwareBuildID":"zStack12","type":"Coordinator"},{"description":"-","friendly_name":"0x00124b0022cf704d","ieeeAddr":"0x00124b0022cf704d","lastSeen":1633252720050,"manufacturerID":0,"networkAddress":59037,"type":"EndDevice","vendor":"-"}],"type":"devices"}'
info  2021-10-03 10:18:45: Successfully interviewed '0x00124b0022cf704d', device has successfully been paired
info  2021-10-03 10:18:45: Device '0x00124b0022cf704d' is supported, identified as: SONOFF Temperature and humidity sensor (SNZB-02)
info  2021-10-03 10:18:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature and humidity sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Voltage of the battery in millivolts","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SNZB-02","supports_ota":false,"vendor":"SONOFF"},"friendly_name":"0x00124b0022cf704d","ieee_address":"0x00124b0022cf704d","status":"successful","supported":true},"type":"device_interview"}'
info  2021-10-03 10:18:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Temperature and humidity sensor","friendly_name":"0x00124b0022cf704d","model":"SNZB-02","supported":true,"vendor":"SONOFF"},"type":"pairing"}'
info  2021-10-03 10:18:45: Changed device specific options of '0x00124b0022cf704d' ({"retain":true})

......

Zigbee2MQTT:info  2021-10-03 10:23:00: Device '01-humidity' joined
Zigbee2MQTT:info  2021-10-03 10:23:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"01-humidity","ieee_address":"0x00124b0022cf704d"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-10-03 10:23:00: Configuring '01-humidity'
Zigbee2MQTT:info  2021-10-03 10:23:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"01-humidity"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-10-03 10:23:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"01-humidity","ieee_address":"0x00124b0022cf704d"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-10-03 10:23:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"01-humidity"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-10-03 10:23:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[],"type":"groups"}'
Zigbee2MQTT:info  2021-10-03 10:23:00: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"dateCode":"20201127","friendly_name":"Coordinator","ieeeAddr":"0x00124b00193691c1","lastSeen":1633252980641,"networkAddress":0,"softwareBuildID":"zStack12","type":"Coordinator"},{"dateCode":"20201026","description":"Temperature and humidity sensor","friendly_name":"01-humidity","hardwareVersion":1,"ieeeAddr":"0x00124b0022cf704d","lastSeen":1633252980584,"manufacturerID":0,"manufacturerName":"eWeLink","model":"SNZB-02","modelID":"TH01","networkAddress":35923,"powerSource":"Battery","type":"EndDevice","vendor":"SONOFF"}]'
Zigbee2MQTT:info  2021-10-03 10:23:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[{"dateCode":"20201127","friendly_name":"Coordinator","ieeeAddr":"0x00124b00193691c1","lastSeen":1633252980659,"networkAddress":0,"softwareBuildID":"zStack12","type":"Coordinator"},{"dateCode":"20201026","description":"Temperature and humidity sensor","friendly_name":"01-humidity","hardwareVersion":1,"ieeeAddr":"0x00124b0022cf704d","lastSeen":1633252980584,"manufacturerID":0,"manufacturerName":"eWeLink","model":"SNZB-02","modelID":"TH01","networkAddress":35923,"powerSource":"Battery","type":"EndDevice","vendor":"SONOFF"}],"type":"devices"}'
Zigbee2MQTT:error 2021-10-03 10:23:10: Configure failed: Error: Bind 0x00124b0022cf704d/1 msTemperatureMeasurement from '0x00124b00193691c1/1' failed (AREQ - ZDO - bindRsp after 10000ms)
Zigbee2MQTT:info  2021-10-03 10:23:10: Successfully configured '01-humidity'
Zigbee2MQTT:info  2021-10-03 10:27:01: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
```

qukla1000, 2021-10-03:
I have the same problem: https://github.com/Koenkk/zigbee2mqtt/issues/8979

yang360, 2021-10-04:
Same problem here as well.

sjorge, 2021-10-04:
Make sure to force remove the failed pairing before trying again, I have the best success with removing the battery and putting it back in, instead of holding the button.

tvrulesme, 2021-10-04:
> Make sure to force remove the failed pairing before trying again, I have the best success with removing the battery and putting it back in, instead of holding the button.

Yes I have done this multiple times as well as deleting my database file and state files. Also tried with the dev branch with the same result

qukla1000, 2021-10-04:
Me too. I tired add three new sensors and the result was the same all the time. A tried add multiple times also with force removing. I also deleted datatbase and reinstall zigbe2mqtt. I tired reflash CC26X2R1 but it didn't help. Also tried change USB cable to another.
I also added some tuya motion sensor - that works without any problem. The problem exists only with Sonoff SNZB-02.

sjorge, 2021-10-04:
Are you pairing through a router or at the coordinator? I have a zzhp-lite prototype and all my pairings of my SNZB-02's were done through routers.

tvrulesme, 2021-10-04:
> Are you pairing through a router or at the coordinator? I have a zzhp-lite prototype and all my pairings of my SNZB-02's were done through routers.

Assuming through coordinator. Is zzhp-lite hardware?

tvrulesme, 2021-10-04:
> Are you pairing through a router or at the coordinator? I have a zzhp-lite prototype and all my pairings of my SNZB-02's were done through routers.

So my hardware is listed above it's pretty much the same as this: https://uk.rs-online.com/web/p/communication-wireless-development-tools/1977102/?cm_mmc=UK-PLA-DS3A-_-google-_-CSS_UK_EN_Raspberry_Pi_%26_Arduino_%26_Development_Tools_Whoop-_-Communication+%26+Wireless+Development+Tools_Whoop-_-1977102&matchtype=&pla-306759658968&gclid=CjwKCAjwzOqKBhAWEiwArQGwaEw5EcmHj-D38MUQm1z3uc8DZWEatV99cjjqob-V-9l6ZEpbq44ZjhoC4YAQAvD_BwE&gclsrc=aw.ds

sjorge, 2021-10-04:
> Assuming through coordinator. Is zzhp-lite hardware?

Yeah zzh-lite is a CC2652P based coordinator, can you try pairing through a router instead? Most of my bulbs are Trådfri bulbs and a few Hue/Innr so I don't think the manufacturer should matter.

tvrulesme, 2021-10-04:
> > Assuming through coordinator. Is zzhp-lite hardware?
> 
> Yeah zzh-lite is a CC2652P based coordinator, can you try pairing through a router instead? Most of my bulbs are Trådfri bulbs and a few Hue/Innr so I don't think the manufacturer should matter.

OK now I am confused. So you mean take the SNZB-02 away from the CC2531 and try and use my wifi router?

qukla1000, 2021-10-04:
> Are you pairing through a router or at the coordinator? I have a zzhp-lite prototype and all my pairings of my SNZB-02's were done through routers.

I have only coordinator in my network.

qukla1000, 2021-10-04:
> OK now I am confused. So you mean take the SNZB-02 away from the CC2531 and try and use my wifi router?

Not wifi router. Zigbee router - if you have.

tvrulesme, 2021-10-04:
> Not wifi router. Zigbee router - if you have.

Ah no, I don't sorry

sjorge, 2021-10-04:
> OK now I am confused. So you mean take the SNZB-02 away from the CC2531 and try and use my wifi router?

> Not wifi router. Zigbee router - if you have.

Yes, zigbee router... usually something that always has power like a bulb.

sjorge, 2021-10-04:
> I have only coordinator in my network.

> Ah no, I don't sorry

Interesting, both of you don't have any routers in the network, I wonder if somehow due to this message don't get buffered as there are no routers to hold them.

@Koenkk unless I am mistaken that is how it usually works right, also why the IKEA blinds need the repeater or another router for the remote to be able to talk to the blinds... I wonder if that is an issue here too.

qukla1000, 2021-10-04:
> Interesting, both of you don't have any routers in the network, I wonder if somehow due to this message don't get buffered as there are no routers to hold them.
>
> @Koenkk unless I am mistaken that is [...]

I also have other installation in other house. But zigbee2mqtt version is 1.18.1-1. There is also one coordinator, other sensors and few SNZB-02. It works without any problem. Joining and reporting states form sensors works seamlessly. It also works with CC26X2R1 bout i don't know with version of software.

tvrulesme, 2021-10-04:
> I also have other installation in other house. But zigbee2mqtt version is 1.18.1-1. There is also one coordinator, other sensors and few SNZB-02. It works without any problem. Joining and reporting [...]

Went through some old hardware boxes and found a switch which is acting as a router. So cleaned out database, added switch, added SNZB-02 and no error messages **BUT** I do not see any data in the logs either. How often does yours generally publish?

tvrulesme, 2021-10-04:
Right, battery in and out and I am seeing data :)

`Zigbee2MQTT:info  2021-10-04 13:42:11: MQTT published: topic 'zigbee2mqtt/humidity', payload '{"linkquality":89,"voltage":2900}'
Zigbee2MQTT:info  2021-10-04 13:42:12: MQTT published: topic 'zigbee2mqtt/humidity', payload '{"battery":85,"linkquality":89,"voltage":2900}'
Zigbee2MQTT:info  2021-10-04 13:42:12: Successfully configured 'humidity'
Zigbee2MQTT:info  2021-10-04 13:42:17: MQTT published: topic 'zigbee2mqtt/humidity', payload '{"battery":85,"linkquality":89,"temperature":23.74,"voltage":2900}'
Zigbee2MQTT:info  2021-10-04 13:42:17: MQTT published: topic 'zigbee2mqtt/humidity', payload '{"battery":85,"humidity":81.51,"linkquality":89,"temperature":23.74,"voltage":2900}'
Zigbee2MQTT:info  2021-10-04 13:42:27: MQTT published: topic 'zigbee2mqtt/humidity', payload '{"battery":85,"humidity":75.94,"linkquality":89,"temperature":23.74,"voltage":2900}'
Zigbee2MQTT:info  2021-10-04 13:42:37: MQTT published: topic 'zigbee2mqtt/humidity', payload '{"battery":85,"humidity":68.46,"linkquality":102,"temperature":23.74,"voltage":2900}'
Zigbee2MQTT:info  2021-10-04 13:42:47: MQTT published: topic 'zigbee2mqtt/humidity', payload '{"battery":85,"humidity":64.67,"linkquality":92,"temperature":23.74,"voltage":2900}'
Zigbee2MQTT:info  2021-10-04 13:42:58: MQTT published: topic 'zigbee2mqtt/humidity', payload '{"battery":85,"humidity":63.18,"linkquality":99,"temperature":23.74,"voltage":2900}'`

qukla1000, 2021-10-04:
> Went through some old hardware boxes and found a switch which is acting as a router. So cleaned out database, added switch, added SNZB-02 and no error messages **BUT** I do not see any data in the [...]

So it looks similar. I can add sonoff, it shows in zigbe2mqtt but sensor don't publish any status even after pushing button.
Of course while i adding this sensor i have error like:
`sTemperatureMeasurement failed (AREQ - ZDO - bindRsp after 10000ms)`
Not always but most of the time.

sjorge, 2021-10-04:
@tvrulesme certainly interesting, so a router in the mesh does seem to help. 
Mine don't immediately send data either, it goes something like this
1. force remove device
2. remove battery
3. enable join via a bulb in the room with the sensor
4. insert battery 
5. (check if the red led blinked, if not, hold the button until it does)
6. wait for z2m to pair successfully
7. (timeout error, back to 1.)

Then I think around ~15 min later it starts sending data, I think that might be some build in default when it first wakes up. Then it seems to follow whatever is configured in the reporting. Removing the battery and inserting it again once it is paired does seem to immediately have it publish battery and voltage, and the rest follow shortly after.

> So it looks similar. I can add sonoff, it shows in zigbe2mqtt but sensor don't publish any status even after pushing button. Of course while i adding this sensor i have error like: `sTemperatureMeasurement failed (AREQ - ZDO - bindRsp after 10000ms)` Not always but most of the time.

I'm guessing it will never send temperature if msTemperatureMeasurement reporting setup gave a timeout. I think that's also the first one it tries, so it probably never got around to setting up the others either, not having any data would make sense then.

I never had luck manually configuring the reporting on these devices, but you could also try that.
[image]

This is what the default reporting looks like.

Koenkk, 2021-10-04:
@tvrulesme if I understand correctly, the device works fine (temperature etc is updated), the only error here is the configure failing?

@sjorge if that is the case we can add a try/catch to the configure (to suppress this error) as it seems not all devices need this.

qukla1000, 2021-10-04:
> @tvrulesme if I understand correctly, the device works fine (temperature etc is updated), the only error here is the configure failing?
> 
> @sjorge if that is the case we can add a try/catch to the configure (to suppress this error) as it seems not all devices need this.

@Koenkk For me device don't work but shows in zigbe2mqtt. Devie don't raport there state. Error shows most f the time but not always. Pushing button on sensor don't update status. Device also isn't reporting its state after while.

tvrulesme, 2021-10-04:
> @tvrulesme if I understand correctly, the device works fine (temperature etc is updated), the only error here is the configure failing?
> 

Yes this is correct @Koenkk, works with a router but not stand alone

Koenkk, 2021-10-04:
@tvrulesme any chance you could provide a sniff when pairing via both router (working situation) and non-working situation (coordinator). https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

tvrulesme, 2021-10-04:
> @tvrulesme any chance you could provide a sniff when pairing via both router (working situation) and non-working situation (coordinator). https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Yeah absolutely. Will get on it tomorrow

takthetank, 2021-10-04:
I have the same problem with SNZB-02 sensors, while other devices work fine.

tvrulesme, 2021-10-05:
> @tvrulesme any chance you could provide a sniff when pairing via both router (working situation) and non-working situation (coordinator). https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Really sorry @Koenkk I got stuck at this point
```
pi@raspberrypi:~/flash_cc2531 $ /snap/bin/cc-tool.run -e -w sniffer_fw_cc2531.hex 
  CC Debugger device not found
```
This is on a raspberry pi.

tvrulesme, 2021-10-05:
Brief update. Sadly the resolution above does not appear to be consistent. Tried a complete clean down removing all devices, adding the switch first followed by the SNZB-02 and again I see this error:

`Zigbee2MQTT:error 2021-10-05 11:25:59: Configure failed: Error: Bind 0x00124b0022cf704d/1 msTemperatureMeasurement from '0x00124b00193691c1/242' failed (AREQ - ZDO - bindRsp after 10000ms)
Zigbee2MQTT:info  2021-10-05 11:25:59: Successfully configured '0x00124b0022cf704d'` 

Please shout if there is anything useful I can send through

sjorge, 2021-10-05:
You only enabled joining through the switch right?

tvrulesme, 2021-10-05:
> You only enabled joining through the switch right?

Sorry @sjorge how would I do that?

sjorge, 2021-10-05:
Easiest is using the frontend and hitting the `V` icon next to `Permit join (All)` then selecting the switch:
[image]

tvrulesme, 2021-10-05:
> That should prevent the sensor from trying to use the coordinator.
[image]
Only seeing Disable join. Is that right?

tvrulesme, 2021-10-05:
[image]
Lined to coordinator. So adding it to router before must have just been luck

sjorge, 2021-10-05:
Might have been because joining was still enabled from before you selected the switch?

tvrulesme, 2021-10-05:
@sjorge have tried everything I can think of and I can't get it to join the switch instead of composer.

tvrulesme, 2021-10-05:
Ok, finally got it to join the router (by physically moving both a long way)

[image]
But the error message persists

```
Zigbee2MQTT:error 2021-10-05 13:31:59: Configure failed: Error: Bind 0x00124b0022cf704d/1 msTemperatureMeasurement from '0x00124b00193691c1/1' failed (AREQ - ZDO - bindRsp after 10000ms)
Zigbee2MQTT:info  2021-10-05 13:31:59: Successfully configured '0x00124b0022cf704d'
Zigbee2MQTT:info  2021-10-05 13:32:07: Starting network scan (includeRoutes 'false')
Zigbee2MQTT:info  2021-10-05 13:32:09: Network scan finished
```

sjorge, 2021-10-05:
😞 so the earlier successful attempt was just random then

tvrulesme, 2021-10-05:
> 😞 so the earlier successful attempt was just random then

Unfortunately yes. So sorry to raise hopes

Koenkk, 2021-10-06:
I've ordered the device to further debug this issue.

takthetank, 2021-10-07:
Hi @Koenkk I managed to sniff through windows, please find attached the z2m and sniffing logs.
[SNZB-02_logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/7299485/SNZB-02_logs.zip)

Please tell me if you need something more!

phobosdigg, 2021-10-07:
The same problem with this sensor.
The problem appeared after I changed the pan_id to 6755 in the zigbee2mqtt settings due to other problems.
Up to this point, the sensors worked fine.
What pan_id do you use in the settings zigbee2mqtt?

takthetank, 2021-10-07:
I haven't set or changed the pan_id!

grfernandes54, 2021-10-09:
I Tried to add 4 SNZB-02 and have the same problem, all sensors show N/A at the exposes

[image]

nickmcgill, 2021-10-09:
I have just downloaded the latest-dev docker image, and I can now pair the SNZB-02, no errors appear to be logged, but, to get it to report correctly I need to reconfigure it.
The reconfiguration must be done before renaming the device.
I hope this is of some help.

deatondg, 2021-10-09:
@maxbause What on Earth? I've been having this same issue for days. I recently bought 2 new sensors, and they wouldn't work although my existing ones would re-pair just fine. This method totally worked for me. Very easy, and very strange.

maxbause, 2021-10-09:
I'm happy that the method works for you too @degreal and @deatondg! :)

@sjorge I also think that this problem has something to do with the state of the sensor, but I haven't had the time yet to further investigate.

nicolaferraro, 2021-10-10:
> Hey guys, I had the same problem: The sensor was connected, but I couldn't get any date out of it.
> 
> I've found a sloppy fix, but it worked for me:
> 
> 1. Make sure that the sensor is still connected to the bridge
> 2. Now open the back of the sensor and remove the button cell battery
> 3. In your zigbee2mqtt web interface, force remove the sensor
> 4. Take a piece of metal (I used a screwdriver tip) and short out the two battery contacts of the sensor (see attached file)
> 5. Now put the battery back in and wait for the sensor to connect
>    5.1 Note: Don't press the button on the sensor, let it connect by itself
> 
> After that, all of my 5 sensor worked perfectly. [image]

Hi @maxbause, I tried it but the sensor does not seem to connect back in step 5 (unless I hold the button again for a few secs). Should I just plug the battery and leave it alone for some minutes or reconnection happens immediately?

Does the light flash when you plug the battery again?

garykrige, 2021-10-10:
I am also not managing to connect back (I'm using the latest dev docker image). I've left the device to connect for 20 minutes, but still no signs of connection without using the button - at which point it fails.

valdeira888, 2021-10-10:
> Same problem here with my new SNZB-02 (I'm trying to add two of them but same error for both), here's the log line:
> 
> `Configure failed: Error: Bind 0x00124b0023adcfe3/1 msTemperatureMeasurement from '0x00124b0022813510/1' failed (AREQ - ZDO - bindRsp after 10000ms)`
> 
> I have another SNZB-02 in my network (added months ago) and it works fine.

Same here, bought 2 and both are not working, even with the screw driver fix.

phobosdigg, 2021-10-10:
One of the sensors was connected using the maxbause method, but the other two sensors are not connected.

Koenkk, 2021-10-12:
@takthetank well that is interesting since nothing has been changed in the device definition since then.

Any chance you could provide me the herdsman debug log when pairing on 1.19.1 (working) and on 1.21.2 (non-working)

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Koenkk, 2021-10-14:
@takthetank I see temperature reports incoming from an unknown device:

```
2021-10-13T21:17:20.818766586Z   zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '20312', skipping... +0ms
```

Do you have an other device which does temperature measurements in your network?

If not, maybe for some reason this device uses a different network address when pairing vs sending messages. Can you try:
- Force remove the SNZB-02
- Remove battery, and make sure the SNZB-02 is completely empty, press the reset button until it shows no signs of life anymore
- Pair the device again

snippem, 2021-10-14:
@Koenkk bought 3 of these sensors as posted above the device pairs and then the following error apears.

Configure failed: Error: Bind 0x00124b0023205202/1 msTemperatureMeasurement from '0x00124b00214f3f95/1' failed (AREQ - ZDO - bindRsp after 10000ms)

Allready updated to the latest dev and did as you discribed:

Force remove the SNZB-02
Remove battery, and make sure the SNZB-02 is completely empty, press the reset button until it shows no signs of life anymore
Pair the device again

but no luck the device does pair but only shows null on al the values and shows the error when it pairs.
Have tried on a second pi4 and a cc2652 running hassos.
And the sensor does pair but also shows null on all the values and the same error. When i did the force remove then short circuit the sensor and pair again that works on that setup. It shows all of the values and it works as it should.
But this trick does not work on the setup that controlls my home. The only difference is that the zigbee stick is a cc2652RB and 124 devices on the network that the zigbee stick controlls.

Koenkk, 2021-10-14:
@snippem 

> The only difference is that the zigbee stick is a cc2652RB and 124 devices on the network that the zigbee stick controlls.

Try pairing it very close to the coordinator, I think this device is a bit tricky over multiple hops.

dsduarte, 2021-10-14:
> @snippem not sure how to fix it now, will check once I receive my SNZB-02.

I don't know if it helps, but I've deleted all my devices, installed the new edge version and after pairing using CC2531, SNZB-02 was reporting temp and humidity but the other parameteres like LQI and battery not.
I've deleted all again a reinstalled the edge version using sonoff zb bridge over TCP and all devices are reporting all of their parameters ok!

salvq, 2021-10-17:
Just followed tipd from @maxbause and it just started working, values are being received finally

```
zigbee2mqtt    | Zigbee2MQTT:info  2021-10-17 13:43:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00226b0cfb', payload '{"battery":23,"humidity":75.85,"linkquality":81,"temperature":28.2,"voltage":2800}'
zigbee2mqtt    | Zigbee2MQTT:info  2021-10-17 13:44:01: MQTT publish: topic 'zigbee2mqtt/0x00124b00226b0cfb', payload '{"battery":23,"humidity":75.85,"linkquality":81,"temperature":28.2,"voltage":2800}'
```

pierrepinon, 2021-10-19:
@maxbause procedure works too, thx 🙏

t0ny-peng, 2021-10-19:
Can confirm that the solution provided by maxbause works perfectly for my 3 sensors.

snippem, 2021-10-19:
What are the network sizes with the maxbause way to add the sensor?
Because that way also works for me but only on a test setup with a few devices.
When I try to add them with the maxbause  way to a 124 device network it doesn't work anymore.
Then I get the same errors as dicribed in in the title of this topic.

snippem, 2021-10-21:
> @maxbause your method works for me for my 8 sensors
> 
> For those having trouble, it's not clear from the picture but when your shorting the contacts what you are trying to short is the metal piece in the middle to the metal piece on the side. 

Please give more details about you setup.
Because I and others already stated.
The method works but how many routers and child devices are on the network.
As I stated before the method works but with my network with 124 routers and child devices it does not.

[66 more replies]

closed 2022-04-17 by Koenkk