Koenkk/zigbee2mqtt #8979: Sonoff SNZB-02 joins with error. No states reporting.

qukla1000, 2021-10-03:
## What happened
No possibility of correct connection and reports state of Sonoff SNZB-02 sensors
After parring  sensors don't report states, even when i push the button. Sometimes devices join successful but no reporting state or shows errors like this: https://hastebin.com/unegukoran.apache and still does not reporting state.
Others sensors for example like motion sensor works correctly.

Tried with with 3 sensors multiple times.
I also decided to change channel but this won't help as well.
premit_join: true
## What did you expect to happen
Successful joining SONOFF SNZB-02 with report states.

## How to reproduce it (minimal and precise)
Try add to the network SNZB-02

## Debug info
Zigbee2MQTT version: 1.21.2-1 
Adapter hardware: CC26X2R1II
Adapter firmware version: 20210708

mgabetta, 2021-10-06:
Exact same thing here!
`Configure failed: Error: Bind 0x00124b0023adcfe3/1 msTemperatureMeasurement from '0x00124b0022813510/1' failed (AREQ - ZDO - bindRsp after 10000ms)`
The strange is that I have another SNZB-02 in my network (added some month ago) and it works like a charm!

Koenkk, 2021-10-06:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/8971

closed 2021-10-06 by Koenkk