Koenkk/zigbee2mqtt #8997: Alarm Keypads - panelStatus initial value

robertalexa, 2021-10-04:
Creating this new issue to keep it all contained in one place rather than hijacking other threads.

With the operation of keypads, and the recent PRs in toZigbee and device specific code, everything behaves as expected.

The only flaw spotted so far is to do with panelStatus. For context, when a keypad is pressed it will ask the server for the status of the alarm.
[image]

When the server sets the state (either from server or accepting a request from keypad), it also sets the status in the globalStore
[image]

However there are situations where the globalStore value is basically undefined, and thus there is no value to be sent to the keypad on request.

The issue arises in 2 scenarios (as confirmed by @stefangries as I do not have a keypad anymore; also seen with a few other users while they were setting the devices up):
 - when the device is paired up for the first time (which i can see by reading the code)
 - when z2m restarts (which was mentioned by Stefan - and it looks like globalStore is volatile)

###Solution 1

The 1st scenario but be easily rectified at device level (see above screenshot). If the value is `undefined` then pass `disarm` (without setting the globalStore value - it will get naturally set with the first usage by user action)

Technically this would also resolve the 2nd scenario too, but it raises some other concern - what happens if the alarm is set (either home or away) at the time z2m restarts. For example HA or Node-Red will have the alarm `arm_all_zones`, but when you press a key on the keypad it will not reflect the truth, that is because nothing set a new arm command and thus the panelStatus in globalStore was not updated. Again, technically, this will work fine, with the downside that the keypad will show the wrong state for the alarm until the next user action that changes the state of the alarm.

###Solution 2

The alternative would be to set the alarm state in a non volatile way, so instead of setting it to globalStore, we write it to a file, thus surviving a restart. I do not know if z2m does this already for anything else, or if this would be a first.

###Conclusion
Do we accept the compromise of solution 1 or is there a better way?

@Koenkk and @sjorge can i please get your opinion on this?

@stefangries if and when we reach a plan of action, would you be willing to do some testing with some code we will provide?

sjorge, 2021-10-04:
It would make sense to save this value persistently I think, I know z2m saves a bunch of stuff in database.db, not sure if there is a way to push something in there manually, but Koenkk probably knows if there is.

robertalexa, 2021-10-04:
That is my thought too.

I have just had another click of another possible problem. With the current code, the panelStatus gets saved in globalStore against `entity`
[image]

Which would then mean if a user wants to have more than 1 keypad, the house alarm state is bound to one device, not both. If we replace `entity` with a fixed value, like say `house_alarm`, then this will work, but z2m will then only support 1 single alarm system per implementation (which is not necessarily a bad thing). Thoughts?

stefangries, 2021-10-04:
> @stefangries if and when we reach a plan of action, would you be willing to do some testing with some code we will provide?

Sure.

sjorge, 2021-10-04:
> That is my thought too.
>
> I have just had another click of another possible problem. With the current code, the panelStatus gets saved in globalStore against `entity` [image]
>
> Which would then mean [...]

Interesting, but that would cause issues for people with multiple alarms right?

e.g. 

Alarm 1 (house):
- keypad 1 (front door)
- keypad 2 (back door)
- bunch of contact sensors hooked up via ha

Alarm 2 (detached garage):
- keypad 3 (garage door)
- garage door sensor hooked up via ha

That would be 2 different alarms with different keypads and one with multiple keypads even.

robertalexa, 2021-10-04:
You are correct :) Hence my doubt

> but z2m will then only support 1 single alarm system per implementation (which is not necessarily a bad thing).

At the minute Alarm 1 with 1 keypad + Alarm 2 with 1 keypad works fine.

Adding a 2nd keypad to either of them is possible, but only if the automation server updates both keypads when the status changes (via server or via 1 of them). So basically store the status in "mirror" between 2 entities. 

Hm... maybe it is not as big of a problem as i thought it is? It just needs a bit more attention on the user's side. Or maybe it is worth adding this as a note on all the supported keypads?

robertalexa, 2021-10-04:
Further thoughts. saving the panelStatus in a non-volatile way could actually be part of something bigger? For example, in the configuration file:

- alarm_group_1:
     - keypad_1
     - keypad_2
- alarm_group_1:
     - keypad_1

Then save the panelStatus with key `alarm_group_1` (will need to change code to find out what group the keypad belongs to)

sjorge, 2021-10-04:
> You are correct :) Hence my doubt
>
>
> At the minute Alarm 1 with 1 keypad + Alarm 2 with 1 keypad works fine.
>
> Adding a 2nd keypad to either of them is possible, but only if the automation server [...]

Might be the most flexible if the user/external tool just manages the state, since well... it is the thing that should know the state and probably should restore it.

It can listen for topic z2m publishes to on startup and just restore the state. It shouldn't matter then how many alarm systems/keypads per alarm system there are.

robertalexa, 2021-10-04:
> It can listen for topic z2m publishes to on startup and just restore the state. It shouldn't matter then how many alarm systems/keypads per alarm system there are.

Is this what you mean? `MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'`

sjorge, 2021-10-04:
> Android [4].

Yes

-- 
~ sjorge

Links:
------
[1] 
https://github.com/Koenkk/zigbee2mqtt/issues/8997#issuecomment-933507328
[2] 
https://github.com/
[3] 
[4]

stefangries, 2021-10-04:
@robertalexa For me I already found a solution by automatically responding messages with empty action and action_transaction with a correct state message.

robertalexa, 2021-10-04:
If i understand your message correctly, it means you are waiting until the keypad asks for an update? I guess that can create some unpleasant delay. But if it works for you that is great.

I wonder if maybe I should just leave it as it is, and update the documentation to cover for:
- set arm_mode after pairing (to have an initial state)
- set arm_mode after z2m restart
- set arm_mode to secondary keypads for multi-keypad setups

stefangries, 2021-10-04:
> If i understand your message correctly, it means you are waiting until the keypad asks for an update?
Yes.
 
> I wonder if maybe I should just leave it as it is, and update the documentation to cover for:
> 
> * set arm_mode after pairing (to have an initial state)
> * set arm_mode after z2m restart
> * set arm_mode to secondary keypads for multi-keypad setups

If the user hasn't to cover these cases in automations manually, it would be great. If Zigbee2MQTT can't handle this, the documentation should be updated.

robertalexa, 2021-10-04:
The only one possible would be the initial state after pairing. For the rest, that is not possible. This is because z2m is purely a "translator" and has not decision making rights in the alarm flow (e.g. it is not the brain of the operation, just the messenger).

Will find some time to update the documentation for all the keypads and they i will close this issue.

guerocp, 2021-10-04:
@stefangries, I get it connected as you and robertalexa told me, sending the server command, but, I cant make it change the status, it publish when I enter a code, but then it blinks red again, I think because it doenst receive an answer from the server, sorry about this, because this topic is for problems with the keypad not for configuration, but can you share me your configuration and automatitations you made in home assistant to get it working? Im kind off confused and can't make it work, thanks

robertalexa, 2021-10-04:
@guerocp could you please share with us what are the commands you are sending. E.g. what mqtt payloads are you sending.

After pairing you need to send an initial command setting the alarm state. This is names arming from server in the documentation.

After that, the next time you press a button on thr keypad it will behave correctly.

If all is well, you can then use the keypad to make a request. The server will need to respond with 2 commands. One to respond to the request, after you check the pin is correct, this has to include the same arm mode as requested by the keypad and the same transaction number. Then the second command is actually changing the alarm state, so basically like the above but without transaction number. You can also find example for this in the documentation.

Hope this helps, if not, let me know what are the commands you are sending and maybe provide some logs.

guerocp, 2021-10-05:
> @guerocp could you please share with us what are the commands you are sending. E.g. what mqtt payloads are you sending.
>
> After pairing you need to send an initial command setting the alarm state. [...]

I already send the initial command, and let my keypad working, but I don´t know how to integrate with home assistant, how to let it work, receiving mqtt command from home assistant, I will upload an screenshot, so you can can see the mqtt messages, Thanks
[image]

stefangries, 2021-10-05:
@robertalexa Do you think it would be useful if I convert my automation in Home Assistant into a blueprint and post it in the HA Community Forum? Then you could also link it from the documentation. My automation is able to synchronize Z2M messages with a Home Assistant [alarm panel](https://www.home-assistant.io/integrations/manual).

robertalexa, 2021-10-05:
@guerocp I understand, so there is nothing wrong, you just need help learning how to use it.

@stefangries By all means do it :) In essence you will be helping... well.. the entire world. So it is definitely something useful. I agree that the HA forum is the best place to write that article, as it caters to HA more than it does to z2m, but we can then have a link in z2m documentation pointing to your blueprint/guide.

stefangries, 2021-10-05:
> @stefangries By all means do it :) In essence you will be helping... well.. the entire world. So it is definitely something useful. I agree that the HA forum is the best place to write that article, as it caters to HA more than it does to z2m, but we can then have a link in z2m documentation pointing to your blueprint/guide.
> 
> LE: realistically your guide will work with any zigbee intermediary right? (z2m, zha, deconz) So basically you could be helping pretty much everyone?

I think it will only work with Z2M as it partially relies on raw mqtt messages. I'll do my best to work on it until the weekend.

robertalexa, 2021-10-05:
Yes you are right. Reading values can be done straight from HA entities, but responding needs to be mqtt as payloads to the keypad.

stefangries, 2021-10-09:
Here is a Blueprint that should solve virtually all problems by synchronising the Keypad State with an Alarm Control Panel in Home Assistant. It is also able to send the correct state directly to the keypad initially, so that the user does not have to send MQTT messages manually.

https://community.home-assistant.io/t/zigbee2mqtt-sync-keypad-and-with-alarm-control-panel-states/345311

stefangries, 2021-10-09:
> 1st one - does that happen when you do it via mqtt only, skipping all integrations? 

I've been thinking about it. It's probably a bug in the logic of the Blueprint. It is triggered by the last "default" block, which always responds to some messages with the current state and thus overwrites the invalid code command. I may have to work on this again.

Do you have any idea which criteria can be used to recognise a keypad (after pairing / restart) that does not yet have a state?

> 2nd one - i am no expert when it comes to HA alarm panel, but I think that might be because of your "panel_arming" trigger. That always executes "exit_delay" which is incorrect for "arm_home". Would it not be possible to split it in triggers? "panel_arming_away" and "panel_arming_home", and then do exit_delay only on the away one?

Unfortunately, the alarm panel does not distinguish whether it switches to armed_home or armed_away. In both cases, the intermediate state is arming. Therefore, when a user uses the panel, it is not possible to distinguish what is likely to follow. With the keypad, that might be possible somehow, that's true.

Btw, what would be the correct intermediate state when switching to arm_home if it is not exit_delay?

stefangries, 2021-10-09:
Okay, I think now I know how to solve this. Will check tomorrow. Thanks for your help and ideas!