Koenkk/zigbee2mqtt #8998: WSDCGQ11LM does not pair but it does pair with my test setup

snippem, 2021-10-04:
I was running the dev version when i had the problem that my sensors values where stuck.
I removed the WSDCGQ11LM and wanted to add the sensor again.
This is where the problem started i cant add the sensor to my setup.
Tested the sensor with my test setup with the same hardware and the sensor works. The test setup runs on a different zigbee channel.
Tried switching my wifi channel and tried setting it up with the sensor close to the antenna noting works.
Switched the CC2652RB for the CC2652 from my test setup and this does not help either.
What am i doing wrong haha or what can i do to different.

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2652RB
Adapter firmware version: 1.21.2

cd334, 2021-10-04:
Same here. For me both xiaomi temperature senor does not pair. The exact types: WSDCGQ11LM and WSDCGQ01LM.
Zigbee2MQTT version: 1.21.2
Adapter hardware: CC2652R
Coordinator firmware version: 20210708

In the log there is nothing.
Other Xiamomi devices pairing good.

snippem, 2021-10-04:
okay did some more investigation every time i pres the little button on the sensor to pair.
I saw another device in the log send a message.
So i removed the router that created the log input and then tried to pair the sensor.
And the sensor joined without any issues the removed router was a HUE bulb.
Dont know why this happend but this solved the problem

cd334, 2021-10-05:
I have also resolved the problem. The battery was very weak, and with new battery the pairing was OK.
It can be closed from my side.

closed 2021-11-12 by github-actions[bot]

MattWestb, 2021-11-12:
Most Philips HUE lights is not removing children and neighbors that have timed out or leaved the network also after mouths and only resetting the light is deleting  the bad addresses reported from the light (therefor is all HUE in the black boy now).
This can making bad things with routing in the network.

I was have problem joining WSDCGQ11LM thru some Zigbee 3 routers then forcing them using them but letting joining one old Zigbee HA router and after its pared i kitting the bad router and pressing the button it jumping to on good Zigbee 3 router.