Koenkk/zigbee2mqtt #9004: Cannot access the Web UI in Home Assistant

nickagian, 2021-10-04:
# Bug Report
## What happened
The Web UI in Home Assistant doesn't work (502 Bad Gateway)
## What did you expect to happen
To access the Web UI
## How to reproduce it (minimal and precise)
After fresh installation and any time afterwards (even after restarting the add-on)
## Debug Info
Zigbee2MQTT version: 1.21.2
Adapter hardware: CC26X2R1
Adapter firmware version: 3.x.0

BenOoi, 2021-10-05:
Make sure the frontend is configured in configuration.yml and is started. 
https://www.zigbee2mqtt.io/information/frontend.html

After that check your log.txt in data/log/\<timestamp\>/log.txt. Mine says:

```info  2021-10-04 21:40:39: Started frontend on port 0.0.0.0:8080```

nickagian, 2021-10-05:
I think the frontend is enabled as it should (have specified only the port to 8099). But I don't see this entry you are talking about in the logs. In there I only see, after I have restarted the add-on, that "Failed to call 'Frontend' 'stop' (TypeError: Cannot read property 'clients' of null...". Also at another day's log I see "Error while starting zigbee-herdsman".

I noticed in the supervisor logs that each time I try to launch the web UI it says that

"ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]"

Shouldn't it actually be the local IP?

BenOoi, 2021-10-06:
Can you tell a little more about your setup? Are you using docker or run it directly on Linux? If you dont regocnise the IP then it sounds like docker. Also you talk about an addon. So are you using the hassio device? I do not have experience with that but it could be using docker underneath (just a wild guess).

Als check if port 8099 is not used by anything else (try another port), and make sure zigbee-herdsman can load. Make sure it can find your usb device. I have noticed after replugging in the USB it might change port numbers.

BenOoi, 2021-10-07:
can you share your configuration.yml and also the top 20ish lines from log.txt? (make sure you first go to the latest \<timestamp\> directory).

nickagian, 2021-10-07:
Well obviously... that's why I didn't see anything.

Here my log.txt at debug level. I don't really see any problem... 

```
debug 2021-10-08 00:18:33: Loaded state from file /config/zigbee2mqtt/state.json
info  2021-10-08 00:18:33: Logging to console and directory: '/config/zigbee2mqtt/log/2021-10-08.00-18-31' filename: log.txt
debug 2021-10-08 00:18:33: Removing old log directory '/config/zigbee2mqtt/log/2021-10-04.16-59-01'
info  2021-10-08 00:18:33: Starting Zigbee2MQTT version 1.21.2 (commit #unknown)
info  2021-10-08 00:18:33: Starting zigbee-herdsman (0.13.149)
debug 2021-10-08 00:18:33: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
info  2021-10-08 00:18:34: zigbee-herdsman started (resumed)
info  2021-10-08 00:18:34: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
debug 2021-10-08 00:18:34: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b0022a4c967","panID":6754}
info  2021-10-08 00:18:34: Currently 0 devices are joined:
info  2021-10-08 00:18:34: Zigbee: disabling joining new devices.
info  2021-10-08 00:18:34: Connecting to MQTT server at mqtt://localhost:1883
debug 2021-10-08 00:18:34: Using MQTT login with username: my_user
```

And yes, the add-on is in home assistant.