Koenkk/zigbee2mqtt #9005: Schneider Electric S520513W Wiser Odace, command nothing happens

jerguitariste, 2021-10-04:
I bought a light intensity dimmer
Schneider Electric S520513W Wiser Odace
zigbee2mqtt.io

https://www.zigbee2mqtt.io/devices/WDE002334.html

the variator is perfectly recognized in zigbeelinker
on the other hand the commands do not work

the z2m dialogue to the device which is not working, but the device dialogue to z2m is working.

I tried including 4 times, but I have the same problem

in Z2m
I have  error message :

Publish 'get' 'state' to '0x680ae2fffe37fcf4' failed: 'Error: Read 0x680ae2fffe37fcf4/21 genOnOff(["onOff"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE)
[image]

jerguitariste, 2021-10-17:
no answer?

closed 2021-11-26 by github-actions[bot]

kevincaradant, 2022-10-22:
Hi

Is it solved ? I want to buy them but I saw some threads and you with this issue

Thanks

jerguitariste, 2022-10-22:
bonjour

Oui il fonctionne bien
mais il faut bien insister lors de l'inclusion dans jeedom

merci

kevincaradant, 2022-10-22:
Bonjour, 

Dans jeedom ? Je pensais que le probleme c’était que un soucis de communication entre Z2M et le module Schneider ? 

J'utilise Home Assistant dans mon cas

kevincaradant, 2022-10-22:
Autre question, je n'arrive pas a trouver la réponse à cela mais il fonctionne AVEC neutre et SANS neutre ou que SANS neutre ? 

J'ai du neutre et je souhaiterais eviter des scintillements des spots ect, surout que je crois qu'on est limité à 5 spots maxi.

Merci :)

jerguitariste, 2022-10-22:
Oui j'avais des soucis pour la commande on/off mais ça fonctionne maintenant
il fonctionne sans neutre

kevincaradant, 2022-10-22:
OK Merci,
Utilises tu un bypass ?
Tu le mets sur quoi comme ampoule ? et de combien de watt ? Désolé de poser autant de question mais cela m'interesse bcp ;)
 Ca fonctionne bien sans scintillement ? 
 J'aurais plus de question après cela :D

jerguitariste, 2022-10-22:
pas de soucis je n'utilise pas de bypass
j'utilise des led osram je crois 5 ou 6 watt, et il y a pas de scintillement

kevincaradant, 2022-10-22:
ok tu as 6w en tout ou un certain nombre de spots sur le meme module ? J'ai cru lire qque part 5 ampoules max par module

Merci en tout cas pour tous ces renseignements :)

kevincaradant, 2022-10-22:
Ah si dernière question .... sorry, est-ce que tu as pu brancher ces interrupteurs en mode va et vient ? ou cela ne fonctionne pas.
je me demandais ce que l'on pouvait prendre en va et vient compatible entre eux. Si tu n'es pas dans ce cas, cela ne fait rien ;)

jerguitariste, 2022-10-22:
Jai 3 spots de 6watt qui sont piloté
Et jai pas de va et viens

Stilnight, 2023-03-25:
> […](#)
> -- Mr OLLAGNON Jerome 80 c chemin de la vigne 38360 Noyarey tel 0683542719

Salut,

J'ai le même message que toi sur Z2M, pourrais tu donner un peu de détail comment tu as réussi à le faire focntionner?

Merci!