Koenkk/zigbee2mqtt #9007: Schneider Electric S520567W Wiser Odace

bprime29, 2021-10-04:
Hi!
I boufht the switch for roller shutter Schneider Electric S520567W Wiser Odace. The device isn't supported by I tried by following the documentation "How to...".
It works but not very well.
My config:
`const definition = {
    zigbeeModel: ['NHPB/SHUTTER/1'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'S520567W', // Vendor model number, look on the device for a model number
    vendor: 'Schneider Electric', // Vendor of the device (only used for documentation and startup logging)
    description: 'Wiser Odace Volet roulant', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.cover_position_tilt, fz.command_cover_close, fz.command_cover_open, fz.command_cover_stop], // We will add this later
    toZigbee: [tz.cover_position_tilt, tz.cover_state], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.cover_position()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    meta: {coverInverted: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1) || device.getEndpoint(5);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
};`

The main problem concerns the positioning of the shutter (%).

Do you have any recommendations? Will you integrate this model?

Thanks!

KKox-26, 2021-11-02:
Hello, 

I bought the same reference. The main problem is the default timer of the roller. It is fixed at 2 minutes and the command to change it is unknown. You can fixed it manually in your home automation software. You take the percentage number corresponding to the time for opening your roller and you fix this number as 100 % with some rules.

Its is not the best but it works perfectelly.

If you need i can give you my automation for HA.

bprime29, 2021-11-03:
Hello,

Thanks for your answer.
I don't really see how to do it. Can you give me your configuration?

Bertrand

KKox-26, 2021-11-03:
So this is not a zigbee2mqtt issue. So i created a post to help in french on [hacf.fr](https://forum.hacf.fr/t/solution-pour-bien-integrer-linterrupteur-volet-roulant-connecte-zigbee-3-0-wiser-odace-blanc-s520567w/7035).

I hope it can help you

bprime29, 2021-11-03:
Merci! Je test ça ce soir :)

closed 2021-12-12 by github-actions[bot]

gerdur, 2023-01-10:
> Hello,
>
> I bought the same reference. The main problem is the default timer of the roller. It is fixed at 2 minutes and the command to change it is unknown. You can fixed it manually in your home [...]

Hello, I have purchased 3 of these devices S520567W from Schneider Electric. Pairing in ZigBee2MQTT under HomeAssistant worked, but until now, I am not able to control these devices via ZigBee2MQTT/HomeAssistant. Can you please support me to get these devices running remotely? Many thanks, Gerhard