Koenkk/zigbee2mqtt #9009: Support new manufacturer for TS0202

vladispavlov, 2021-10-05:
## Information about the device + link

Mini Zigbee PIR on AliExpress:
[https://www.aliexpress.com/item/1005001977463358.html](https://www.aliexpress.com/item/1005001977463358.html)
<details>

[image]
</details>

Looks like RH3040
https://www.zigbee2mqtt.io/devices/RH3040.html
but in z2m defined as TS0202
https://www.zigbee2mqtt.io/devices/TS0202.html
Manufacturer name: _TZ3000_ykwcwxmz
which is not supported

Works with the following external converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TZ3000_ykwcwxmz'}],
    model: 'TS0202',
    vendor: 'TuYa',
    description: 'Motion sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery()],
};

module.exports = definition;
```
<details>

[image]
</details>
However, the battery value is displayed as N/A.
<details>

[image]
</details>
In the Tuya app value was displayed as an exact percentage.
<details>

[image]
[image]
</details>

## data/database.db entry of the device
`{"id":11,"type":"EndDevice","ieeeAddr":"0xa4c13890bdbd753a","nwkAddr":21309,"manufId":4417,"manufName":"_TZ3000_ykwcwxmz","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TZ3000_ykwcwxmz","powerSource":3,"zclVersion":3,"appVersion":67,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"currentZoneSensitivityLevel":1,"iasCieAddr":"0x00124b0022ab741f","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1633385129127}`

danieledwardgeorgehitchcock, 2021-10-05:
It looks as though the device does not have a `genPowerCfg` cluster which is where the fz.battery converter gets its data from.

I cannot see a `manuSpecificTuya` cluster either so, I am guessing that the device may only send a low power alarm through the `ssIasZone` cluster..?

If you have the native TuYa zigbee gateway, you should be able to see what is able to be reported by the device

vladispavlov, 2021-10-06:
I parsed the logs and found the following related to the sensor:
<details>

```
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0202"}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_ykwcwxmz"}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'genBasic', data '{"appVersion":67}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":0}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":0}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00124b0022ab741f","zoneState":1}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'attributeReport', cluster 'ssIasZone', data '{"61441":1}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'attributeReport', cluster 'ssIasZone', data '{"currentZoneSensitivityLevel":1}' from endpoint 1 with groupID 0
Received Zigbee message from '0xa4c13890bdbd753a', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
```
</details>

I can't confirm that battery level contains in `genBasic`. I'll wait for more new logs from sensor.

I started to search converters from zigbee with similar cluster data and found the next:
- `ZB003X_attr`
- `ias_occupancy_alarm_1`
- `ias_enroll`

And there are the next related converters to zigbee:
- `ZB003X`

So I also used exposes from ZB003X:
- `exposes.enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high']).withDescription('PIR sensor sensitivity'),`
- `exposes.enum('keep_time', ea.STATE_SET, ['0', '30', '60', '120', '240']).withDescription('PIR keep time in seconds'),`

So i created the new converter which includes mentioned converters and now it looks like this:
```
let fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

fz = Object.assign(fz, {
    ts0202_enroll: {
        cluster: 'ssIasZone',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const zoneState = msg.data.zoneState;
            const iasCieAddr = msg.data.iasCieAddr;
            return {
                enrolled: zoneState !== 0,
                ias_cie_address: iasCieAddr,
            };
        },
    },
});

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TZ3000_ykwcwxmz'}],
    model: 'TS0202',
    vendor: 'TuYa',
    description: 'Motion sensor',
[... 27 lines cut ...]
        "manufacturerID": 4417,
        "manufacturerName": "_TZ3000_ykwcwxmz",
        "model": "TS0202",
        "networkAddress": 53504,
        "powerSource": "Battery",
        "stackVersion": 0,
        "type": "EndDevice",
        "zclVersion": 3
    },
    "enrolled": true,
    "keep_time": 30,
    "last_seen": "2021-10-06T02:20:01.327Z",
    "linkquality": 123,
    "occupancy": null,
    "sensitivity": "medium",
    "tamper": null
}
```

Now I can change the "sensitivity" and "keep time" parameters, but I didn't see that they somehow affect the sensor.
[... cut ...] hen trying to read data using dev console or make some motion or change previously mentioned params. This is a problem that i found also on previous converter. Also when I get this error, the sensor is not sending any information, it just seems to freeze, there are not even link quality updates and of course there are no any logs.

I'm using z2m 1.21.2 with [egony's cc2652p](https://github.com/egony/cc2652p_E72-2G4M20S1E/wiki/Home-EN) coordinator revision 20210813.

Any ideas on how to fix this?

closed 2021-12-06 by github-actions[bot]

vladispavlov, 2022-02-20:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3885