Koenkk/zigbee2mqtt #9010: Xiaomi RTCGQ11LM did not responding correctly after upgrade

SargonofAssyria, 2021-10-05:
Yesterday I upgraded from 1.21.1-dev commit: 5829663b to 1.21.2-dev commit: 7ccf781b
After the upgrade ALL of my RTCGQ11LM did not respond properly to a "physical" trigger.
The mqtt response was always 'occupancy:false'
After I went back to my previous version everything was fine again.

towerhand, 2021-10-05:
I can confirm, I have the same issue since yesterday.

Koenkk, 2021-10-05:
First make sure you are not affected by https://github.com/Koenkk/zigbee2mqtt/issues/8663, if not, please provide the debug log when triggering occupancy.

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

ALaDoffe, 2021-10-05:
Seems to be same as #8958
Can you try removing last seen option if set ?

SargonofAssyria, 2021-10-05:
I can confirm that I am not using CH341 driver. I am using this in production so I cannot easily go back the latest and do some testing or debugging. Not sooner than tomorrow, otherwise ....

towerhand, 2021-10-05:
Running `uname -r`  results in 5.10.63-v7+ on my raspberry pi 3+, so I would think #8663 is not the issue. Right?

towerhand, 2021-10-05:
`Debug Received Zigbee message from 'Kitchen Motion Sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Kitchen Motion Sensor', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20170627","friendlyName":"Kitchen Motion Sensor","hardwareVersion":1,"ieeeAddr":"0x00158d0003554e19","manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ11LM","networkAddress":51706,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"illuminance":7,"illuminance_lux":7,"last_seen":1633455591924,"linkquality":78,"occupancy":false,"temperature":23,"voltage":3035}'
Info MQTT publish: topic 'zigbee2mqtt/Kitchen Motion Sensor', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20170627","friendlyName":"Kitchen Motion Sensor","hardwareVersion":1,"ieeeAddr":"0x00158d0003554e19","manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ11LM","networkAddress":51706,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"illuminance":7,"illuminance_lux":7,"last_seen":1633455597066,"linkquality":78,"occupancy":false,"temperature":23,"voltage":3035}'
Debug Received Zigbee message from 'Kitchen Motion Sensor', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":852}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Kitchen Motion Sensor', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20170627","friendlyName":"Kitchen Motion Sensor","hardwareVersion":1,"ieeeAddr":"0x00158d0003554e19","manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ11LM","networkAddress":51706,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"illuminance":7,"illuminance_lux":7,"last_seen":1633455597088,"linkquality":78,"occupancy":false,"temperature":23,"voltage":3035}'
Debug Received Zigbee message from 'Kitchen Motion Sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Kitchen Motion Sensor', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20170627","friendlyName":"Kitchen Motion Sensor","hardwareVersion":1,"ieeeAddr":"0x00158d0003554e19","manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ11LM","networkAddress":51706,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"illuminance":7,"illuminance_lux":7,"last_seen":1633455597088,"linkquality":78,"occupancy":false,"temperature":23,"voltage":3035}'
Info MQTT publish: topic 'zigbee2mqtt/Mudroom Motion Sensor', payload '{"battery":100,"device":{"applicationVersion":5,"friendlyName":"Mudroom Motion Sensor","ieeeAddr":"0x00158d00044589f0","manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ11LM","networkAddress":12720,"powerSource":"Battery","type":"EndDevice"},"illuminance":3,"illuminance_lux":3,"last_seen":1633455597357,"linkquality":null,"occupancy":false,"temperature":23,"voltage":3355}'
Debug Received Zigbee message from 'Mudroom Motion Sensor', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":2}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Mudroom Motion Sensor', payload '{"battery":100,"device":{"applicationVersion":5,"friendlyName":"Mudroom Motion Sensor","ieeeAddr":"0x00158d00044589f0","manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ11LM","networkAddress":12720,"powerSource":"Battery","type":"EndDevice"},"illuminance":3,"illuminance_lux":3,"last_seen":1633455597383,"linkquality":75,"occupancy":false,"temperature":23,"voltage":3355}'
Debug Received Zigbee message from 'Mudroom Motion Sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Mudroom Motion Sensor', payload '{"battery":100,"device":{"applicationVersion":5,"friendlyName":"Mudroom Motion Sensor","ieeeAddr":"0x00158d00044589f0","manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ11LM","networkAddress":12720,"powerSource":"Battery","type":"EndDevice"},"illuminance":3,"illuminance_lux":3,"last_seen":1633455597383,"linkquality":78,"occupancy":false,"temperature":23,"voltage":3355}'
Info MQTT publish: topic 'zigbee2mqtt/Kitchen Motion Sensor', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20170627","friendlyName":"Kitchen Motion Sensor","hardwareVersion":1,"ieeeAddr":"0x00158d0003554e19","manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ11LM","networkAddress":51706,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"illuminance":7,"illuminance_lux":7,"last_seen":1633455602148,"linkquality":78,"occupancy":false,"temperature":23,"voltage":3035}'
Debug Received Zigbee message from 'Kitchen Motion Sensor', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":852}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Kitchen Motion Sensor', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20170627","friendlyName":"Kitchen Motion Sensor","hardwareVersion":1,"ieeeAddr":"0x00158d0003554e19","manufacturerID":4151,"manufacturerName":"LUMI","model" [... cut ...]
Info MQTT publish: topic 'zigbee2mqtt/Kitchen Motion Sensor', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20170627","friendlyName":"Kitchen Motion Sensor","hardwareVersion":1,"ieeeAddr":"0x00158d0003554e19","manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ11LM","networkAddress":51706,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"illuminance":7,"illuminance_lux":7,"last_seen":1633455649901,"linkquality":84,"occupancy":false,"temperature":23,"voltage":3035}'`

towerhand, 2021-10-05:
FYI, I rolled back from latest-dev to latest tag and the motion sensors work perfectly again, hopefully the logs help.

Koenkk, 2021-10-05:
I cannot reproduce this, can you provide your `configuration.yaml`?

towerhand, 2021-10-05:
****configuration.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: HIDDEN
  user: HIDDEN
  password: HIDDEN
  include_device_information: true
  keepalive: 60
frontend:
  port: HIDDEN
  host: HIDDEN
  auth_token: HIDDEN
serial:
  port: /dev/ttyUSB0
advanced:
  pan_id: 6756
  channel: 25
  cache_state: true
  cache_state_persistent: true
  cache_state_send_on_startup: true
  log_level: info
  log_directory: data/log/%TIMESTAMP%
  log_file: log.txt
  log_rotation: true
  rtscts: false
  network_key: HIDDEN
  last_seen: epoch
  elapsed: false
  report: true
  homeassistant_discovery_topic: homeassistant
  homeassistant_status_topic: homeassistant/status
  legacy_api: false
devices: devices.yaml
groups: groups.yaml

****devices.yaml
'0x00158d0003554e19':
  friendly_name: Kitchen Motion Sensor
  retain: true
  qos: 1
  debounce: 0.2
  occupancy_timeout: 6
'0x00158d00044589f0':
  friendly_name: Mudroom Motion Sensor
  retain: true
  qos: 1
  debounce: 0.2
  occupancy_timeout: 6
'0x00158d00044b11ba':
  friendly_name: Foyer Motion Sensor
  retain: true
  qos: 1
  debounce: 0.2
  occupancy_timeout: 6

Koenkk, 2021-10-06:
Can you try removing the `debounce`?

towerhand, 2021-10-06:
I’ll try tomorrow, today I’m away from home.

towerhand, 2021-10-06:
Didn’t have time to try it yesterday, but I’ll try that tomorrow too.

SargonofAssyria, 2021-10-07:
Tnx, will test tomorrow and let you know.