Koenkk/zigbee2mqtt #9012: Invalid value for number...  (range -65535.0 - 65535.0)

Mr-Groch, 2021-10-05:
## What happened
After updating to Z2M 1.21.2 I see lot of errors in HA log:

```
Invalid value for number.licznik_pradu_1_l2: 444740 (range -65535.0 - 65535.0)
Invalid value for number.licznik_pradu_1_l2: 444773 (range -65535.0 - 65535.0)
Invalid value for number.licznik_pradu_1_l2: 444808 (range -65535.0 - 65535.0)
Invalid value for number.licznik_pradu_1_l2: 444837 (range -65535.0 - 65535.0)
```

Device is DIY counter based on ptvo firmware

## What did you expect to happen
No errors

## How to reproduce it (minimal and precise)
After each PTVO DIY couner report

## Debug info
Zigbee2MQTT version: 1.21.2
Adapter hardware: CC26X2R1
Adapter firmware version: 20210708

rindlerblabla, 2021-10-05:
Maybe @ptvoinfo can help you out in some way?

ptvoinfo, 2021-10-05:
@Mr-Groch, @rindlerblabla I cannot help here. It is a problem with HA integration. I don't know why z2m sets this limited range for an analog value.

rindlerblabla, 2021-10-05:
@ptvoinfo  Thanks for your reply!

Mr-Groch, 2021-11-05:
Issue still exists in Z2M 1.22.0 and HA 2021.11.0

Koenkk, 2021-11-05:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-11-13 by github-actions[bot]

tj57, 2022-04-21:
In Zigbee2MQTT 1.25.0 I still get this error:
Invalid value for number.gaz_l2: 26046 (range 0.0 - 100.0)

I'm using ptvo device.
Whether this is a problem on the HA side or Zigbee2MQTT ?
Is there a workaround available?

vitoskin, 2023-07-19:
> In Zigbee2MQTT 1.25.0 I still get this error: Invalid value for number.gaz_l2: 26046 (range 0.0 - 100.0)
> 
> I'm using ptvo device. Whether this is a problem on the HA side or Zigbee2MQTT ? Is there a workaround available?

Was the issue fixed?
I've the same in Zigbee2MQTT 1.32.1 - "Invalid value for number.0x00124b00198fde97_l3: 3127 (range 0.0 - 100.0)"

tj57, 2023-08-15:
Probably not, I still see this error in the logs.