Koenkk/zigbee2mqtt #9019: Set state on groups does not work after upgrade

BenOoi, 2021-10-05:
## What happened
I upgraded from 1.20 to 1.21.2 on my pi. After this the groups no longer respond to input. When activating a group member, I do get a response from the group saying it is also changed state

## What did you expect to happen
When I turn on a group, all members turn on.

## How to reproduce it (minimal and precise)
Before upgrade all groups functioned as expected. After upgrade the mqtt message is sent:

```json
zigbee2mqtt/Woonkamer/set {"tst":"2021-10-05T12:11:37.913127Z+0200","topic":"zigbee2mqtt/Woonkamer/set","qos":0,"retain":0,"payloadlen":15,"payload":{"state":"ON"}}
```

But there is no further response. Lights stay off and Home Assistant puts the switch back to off. When setting a member:

```json
zigbee2mqtt/Hanglamp/set {"tst":"2021-10-05T12:12:19.386196Z+0200","topic":"zigbee2mqtt/Hanglamp/set","qos":0,"retain":0,"payloadlen":15,"payload":{"state":"ON"}}
zigbee2mqtt/Woonkamer {"tst":"2021-10-05T12:12:19.598960Z+0200","topic":"zigbee2mqtt/Woonkamer","qos":0,"retain":0,"payloadlen":31,"payload":{"brightness":127,"state":"ON"}}
zigbee2mqtt/Hanglamp {"tst":"2021-10-05T12:12:19.599581Z+0200","topic":"zigbee2mqtt/Hanglamp","qos":0,"retain":0,"payloadlen":99,"payload":{"brightness":127,"linkquality":45,"power_on_behavior":null,"state":"ON","update":{"state":"idle"}}}
```

I also get response from the group. So seems to be working half way.

Also reusing the group ID when recreating groups gives a weird response. I removed all members from groupID 4, recreated the group with the same name and ID and could not add any members. This ID seems to be unusable now.

## Debug info
Zigbee2MQTT version: 1.21.2 (commit #44922120) zigbee-herdsman (0.13.149)
Adapter hardware: CC26X2R1 slae.sh
Adapter firmware version: 20210120 zStack3x0

Koenkk, 2021-10-12:
Can you provide the debug log of that?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

BenOoi, 2021-10-13:
For some reason I cannot reproduce now. Initial problem has been fixed so this can be closed.