Koenkk/zigbee2mqtt #9028: 1.21.2 has breaking changes for Green Power devices (e.g. Philips Hue Tap)

davidjb, 2021-10-06:
## What happened

On upgrading to v1.21.2, I've found the action values for Green Power devices have changed as a result of https://github.com/Koenkk/zigbee-herdsman-converters/pull/2980, meaning all automations won't work until they're updated to use the changed values.

The change was easy enough for make via find-and-replace, it was more that the breaking change was unexpected. 

For the Philips Hue Tap, the mapping is:

* `toggle` → `press_1`
* `recall_scene_0` → `press_2`
* `recall_scene_1` → `press_3`
* `recall_scene_2` → `press_4`

## What did you expect to happen

Changelog would ideally mention breaking changes and, if following SemVer, then the major version number should be increased.

## How to reproduce it (minimal and precise)

Be using a previous version and update to `>=1.21.2`.

## Debug info
Zigbee2MQTT version: 1.21.2

rindlerblabla, 2021-10-06:
Similar that 1.21(?) broke some functionality of the keypads https://github.com/Koenkk/zigbee2mqtt/issues/8630.

I also think that it should be more obvious when functionality is lost, maybe add another headline for the release updates - "Breaking changes".

Koenkk, 2021-10-06:
> Changelog would ideally mention breaking changes and, if following SemVer, then the major version number should be increased.

You are totally right, this should indeed be marked as a breaking change however I completely missed this (mainly because it wasn't meant to be breaking in the first place, details: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2885)

davidjb, 2021-10-06:
No worries, thanks @Koenkk.  I mostly opened this to make folks aware of the situation in case they experienced the same situation.

I really appreciate all your efforts -- and fwiw these underlying changes are really helpful.  They makes the Tap switches more usable in z2m and easier to work with as the action values more logical/memorable.

davidjb, 2021-10-06:
@Koenkk Sounds good 👍  Perhaps in the interim you could edit the release description for the current release (https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.21.2) and mention the known issue / breaking change?  Linking to this issue (e.g. the mapping of values) would probably help others - assuming they're reading the release notes 😄

closed 2021-12-01 by github-actions[bot]