Koenkk/zigbee2mqtt #9029: Add verbose logging on startup to validate MQTT server connectivity

ktoms71, 2021-10-06:
## Is your feature request related to a problem? Please describe
I have been trying tirelessly to get the latest pull to connect to my mqtt server.   Syntax and values in my configuration.yaml are correct and have been tested in other containers.    But Im not able to connect and there is no notification or logging or anything.   I have tested various permutations of systan in the config file, also tested connections to various other mqtt servers within containers on the same docker platform.   I have have also noticed that with the latest release, the "Started frontend on port 0.0.0.0:9442" comment is not written to the log.   But in prior releases(1.21.1) and earlier, this comment is logged, even though connectivity to the mqtt server is not taken place.     I did not want to submit this request as a problem, but there is something at issue.    As mentioned, there is possibly something at issue in node.js that is prohibiting from connecting to an mqtt server.   Ive made some posts in the discord channel under the release-branch section.     Any feedback would be greatly appreciated and Im really looking foward to utilizing this great tool.   Thanks in advance.

Koenkk, 2021-10-06:
Looks similar to https://github.com/Koenkk/zigbee2mqtt/issues/8956 , latest-dev produces better errors now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

ktoms71, 2021-10-10:
Hi Koenkk,   I have tried the dev and all of the other released going back about 8 months and still having issues with z2m not being at all responsive to connecting to mqtt servers.     below are links to my configuration file, the docker run syntax and lastly the log files from z2m.   

configuration.yaml  https://pastebin.com/VvheEXYC
docker run https://pastebin.com/Tkp4r0ZL
logs   ---> https://pastebin.com/wyrtLMCh

ktoms71, 2021-10-11:
> @ktoms71 can you provide the log when running with:
>
> docker run -d --name='zigbee2mqtt' --net='bridge' -e TZ="America/Chicago" -e DEBUG=* -e HOST_OS="Unraid" -p '9442:9442/tcp' -v '/mnt/user/appdata/zigbee2mqtt':'/app/data':'rw' --device='/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2191707-if00' 'koenkk/zigbee2mqtt'
> dc2bedd65426b23df58ed5dac4d8543fe1657b6eb5291830ea7414c7bdcffde5

Disregard my last message.  I'll test the run command that you provide and I'll provide the logs.  Thanks, Kevin.

Koenkk, 2021-10-16:
@chrishae looks related to deconz, what does this error mean?

closed 2021-12-04 by github-actions[bot]

mpolivanov, 2022-02-10:
I have the same issue here aftre update to 1.23.0. After restart I had no access to web interface anymore, so I restarted with debug enabled and got multiple "no free slots. Delay sending of APS Request" each second into the log. At some point, probably 30 minutes later, following appeared and zigbee2mqtt container stopped:

2022-02-10T22:10:01.591Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-02-10T22:10:01.650Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-02-10T22:10:01.711Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-02-10T22:10:01.770Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-02-10T22:10:01.830Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0x7 seqNr: 33
2022-02-10T22:10:01.830Z zigbee-herdsman:deconz:driver too many timeouts - restart serial connecion
2022-02-10T22:10:01.830Z zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0
2022-02-10T22:10:01.831Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-02-10T22:10:01.860Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 34
2022-02-10T22:10:01.891Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-02-10T22:10:01.952Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-02-10T22:10:02.011Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-02-10T22:10:02.071Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-02-10T22:10:02.132Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-02-10T22:10:02.148Z zigbee-herdsman:deconz:driver Serialport opened
2022-02-10T22:10:02.150Z zigbee-herdsman:deconz:driver Error writing serial Port: ENODEV: no such device, write
Error: ENODEV: no such device, write

Started the container again and the messages are gone now.

I have a virtual system running in Proxmox where all my containers are placed. The whole setup worked fine for two months.