Koenkk/zigbee2mqtt #9031: Moes TuYa TS0601 _TZE200_cwnjrr72 HY368

Scalabanana, 2021-10-06:
I can't get this Thermostat to work. Joining it to the zigbee Network reports it back as unsupported device. Browsing Github I found this type of device should be already in the supported devices list.
I also tried to get it to work with an external converter, but this gave me another problem:
`> zigbee2mqtt@1.18.1 start /app
> node index.js
(node:401) UnhandledPromiseRejectionWarning: internal/modules/cjs/loader.js:883
  throw err;
  ^
Error: Cannot find module '../lib/exposes'
Require stack:
- /app/lib/util/utils.js
- /app/lib/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:880:15)
    at Function.Module._load (internal/modules/cjs/loader.js:725:27)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at /app/lib/util/externally-loaded.js:1:17
    at Script.runInContext (vm.js:143:18)
    at Script.runInNewContext (vm.js:148:17)
    at Object.runInNewContext (vm.js:303:38)
    at loadModuleFromText (/app/lib/util/utils.js:156:8)
    at loadModuleFromFile (/app/lib/util/utils.js:162:12)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:401) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:401) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.`
I am using zigbee2mqtt addon in HassOS version 1.18.1.

Meljohn0719, 2022-01-03:
Hi im having issues on my zigbeemqtt. its showing that my new device is not supported by the zigbeemqtt. [
[image]
[image]
](url)

Can someone help me with this..

Scalabanana, 2022-01-03:
Which z2m version are you running?

Meljohn0719, 2022-01-04:
> Which z2m version are you running?

Hello @Scalabanana im currently using this version. 
[image]