Koenkk/zigbee2mqtt #9058: MiJia light intensity sensor pairs but fails the rest (GZCGQ01LM)

Ferrari2, 2021-10-07:
Hi

I am using zigbee2mqtt under Home Assistant and the pairing seems OK (interview completed = true) and I can see the light sensor device in the zigbee2mqtt UI.

Unfortunately the light sensor doesn't work and I see this error in the log:
error Failed to configure '0x04cf8cdf3c7ccd0c', attempt 2 (Error: Bind 0x04cf8cdf3c7ccd0c/1 genPowerCfg from '0x00124b001949c4d8/1' failed (SRSP - ZDO - bindReq after 6000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))

MiJia light intensity sensor FW: _2019www._

I am running zigbee2mqtt v1.21.2-1

The light sensor works fine many versions ago...

Best ragards
Bjarke

silversmarty, 2021-10-12:
same thing here : device is paired, but nothing else works 

[image]

Logs:

> Zigbee2MQTT:error 2021-10-12 07:29:33: Failed to configure 'z_lumi', attempt 1 (Error: Bind 0x04cf8cdf3c81a4e9/1 genPowerCfg from '0x00212effff074a3d/1' failed (waiting for response TIMEOUT)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:757:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:500:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/xiaomi.js:1251:13)
    at Configure.configure (/app/lib/extension/configure.ts:119:13)
    at Configure.onMQTTMessage_ (/app/lib/extension/configure.ts:60:21))

m-novitsky, 2021-11-04:
It's the same with me. The three sensors are pairing but not working. Error - as in the first message.

Ferrari2, 2021-11-04:
Yes, the same here using Zigbee2MQTT v1.22.0-2...

m-novitsky, 2021-11-12:
The problem was solved by changing the coordinator channel to 15. And by re-pairing all devices :(

Ferrari2, 2021-11-13:
OK - better than nothing :-)
Where did you exactly change the coordinator channel to 15? My question is just because I can't find the spot :-)

m-novitsky, 2021-11-13:
zigbee2mqtt/data/configuration.yaml
[image]

Ferrari2, 2021-11-13:
It seem like I have an issue concerned the adapter settings as it remains on ch. 11...

Zigbee2MQTT:error 2021-11-14 00:00:44: Configuration is not consistent with adapter state/backup!
Zigbee2MQTT:error 2021-11-14 00:00:44: - PAN ID: configured=6754, adapter=6754
Zigbee2MQTT:error 2021-11-14 00:00:44: - Extended PAN ID: configured=00124b001949c4d8, adapter=dddddddddddddddd
Zigbee2MQTT:error 2021-11-14 00:00:44: - Network Key: configured=01030507090b0d0f00020406080a0c0d, adapter=01030507090b0d0f00020406080a0c0d
Zigbee2MQTT:error 2021-11-14 00:00:44: - **Channel List: configured=15, adapter=11**
Zigbee2MQTT:error 2021-11-14 00:00:44: Please update configuration to prevent further issues.
Zigbee2MQTT:error 2021-11-14 00:00:44: If you wish to re-commission your network, please remove coordinator backup at /config/zigbee2mqtt/coordinator_backup.json.
Zigbee2MQTT:error 2021-11-14 00:00:44: Re-commissioning your network will require re-pairing of all devices!
Zigbee2MQTT:error 2021-11-14 00:00:44: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-11-14 00:00:44: Failed to start zigbee
Zigbee2MQTT:error 2021-11-14 00:00:44: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-11-14 00:00:44: Exiting...

m-novitsky, 2021-11-14:
The key phrase is "If you wish to re-commission your network, please remove coordinator backup at /config/zigbee2mqtt/coordinator_backup.json." Do it :)

DJTerentjev, 2021-11-18:
Same issue for me.

DJTerentjev, 2021-11-18:
Looks like this can help:
https://github.com/Koenkk/zigbee2mqtt/issues/8931

closed 2021-12-26 by github-actions[bot]