Koenkk/zigbee2mqtt #9089: Lidl HG06467 gives color_mode key error in Home Assistant

groenmarsmannetje, 2021-10-09:
## What happened

The following error is shown in Home Assistant log multiple times

```Logger: homeassistant.util.logging
Source: util/logging.py:105 
First occurred: 11:25:07 (2 occurrences) 
Last logged: 11:30:06

Exception in state_received when handling msg on 'zigbee2mqtt/kerstverlichting_2': '{"brightness":56,"color":{"b":56,"h":270,"s":75,"x":0.4603,"y":0.1948},"color_temp":9,"effect":null,"last_seen":"2021-10-09T11:25:07+02:00","linkquality":75,"state":"OFF"}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 37, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 308, _color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 247, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'
```

## What did you expect to happen

No errors in logging

## How to reproduce it (minimal and precise)

Errors start appearing when you restart Home Assistant

## Debug info
Zigbee2MQTT version: 1.21.2
Adapter hardware: ZZH
Adapter firmware version: 20210708

stefangries, 2021-10-13:
Hey @Koenkk, seems like your posted link is wrong. Which is the correct one? I'm experiencing the same issue.

stefangries, 2021-11-02:
Hey @Koenkk , seems like this fix did'n fix the problem :-(
I'm still getting the error in 1.22.0-2.

`2021-11-02 20:57:15 ERROR (MainThread) [homeassistant.util.logging] Exception in state_received when handling msg on 'zigbee2mqtt/Lichterkette': '{"brightness":254,"color":{"b":254,"h":60,"s":75},"effect":null,"last_seen":"2021-11-02T20:57:15+01:00","linkquality":255,"state":"ON"}'`

Koenkk, 2021-11-02:
@stefangries can you provide the z2m debug log after setting the color? 
See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

stefangries, 2021-11-02:
Wenn man in ausgeschaltetem Zustand das erste mal den Schalter betätigt erscheint der folgende log. Der Schalter schnellt aber sofort wieder auf Aus zurück.

```
Debug Received MQTT message on 'zigbee2mqtt/Lichterkette/set' with data '{"state": "ON"}'
Debug Publishing 'set' 'state' to 'Lichterkette'
Info MQTT publish: topic 'zigbee2mqtt/Lichterkette', payload '{"brightness":254,"color":{"b":254,"h":60,"s":75},"effect":null,"last_seen":"2021-11-02T22:38:11+01:00","linkquality":255,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/Lichterkette', payload '{"brightness":254,"color":{"b":254,"h":60,"s":75},"effect":null,"last_seen":"2021-11-02T22:38:11+01:00","linkquality":255,"state":"ON"}'
```

If you press the switch again, it also changes to on. However, the device was already on.

```
Debug Received MQTT message on 'zigbee2mqtt/Lichterkette/set' with data '{"state": "ON"}'
Debug Publishing 'set' 'state' to 'Lichterkette'
Info MQTT publish: topic 'zigbee2mqtt/Lichterkette', payload '{"brightness":254,"color":{"b":254,"h":60,"s":75},"effect":null,"last_seen":"2021-11-02T22:38:12+01:00","linkquality":255,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Lichterkette', payload '{"brightness":254,"color":{"b":254,"h":60,"s":75},"effect":null,"last_seen":"2021-11-02T22:38:12+01:00","linkquality":255,"state":"ON"}'
```

@Koenkk Do you need more information?

Koenkk, 2021-11-03:
Found the problem, can you check if it has been fixed?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

stefangries, 2021-11-04:
I tried the Edge Addon (Image zigbee2mqtt/zigbee2mqtt-edge-amd64:edge, sha256:42431b7f7d9a10b598ea18cd7d827615503052ff9f65ae29d01ae0768cd4ca95, Created  2021-11-03 21:28:32). No difference, sill not working.

From what I have seen in your [commit](https://github.com/Koenkk/zigbee-herdsman-converters/commit/d9b7cba96b5daf9a79dee8d45b055c00373a0ba6), I'd expected to see the "color_mode" in the MQTT messages, which is not the case.

```
Debug Received MQTT message on 'zigbee2mqtt/Lichterkette/set' with data '{"state": "ON"}'
Debug Publishing 'set' 'state' to 'Lichterkette'
Info MQTT publish: topic 'zigbee2mqtt/Lichterkette', payload '{"brightness":254,"color":{"b":254,"h":131,"s":100},"effect":null,"last_seen":"2021-11-04T08:51:00+01:00","linkquality":255,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/Lichterkette', payload '{"brightness":254,"color":{"b":254,"h":131,"s":100},"effect":null,"last_seen":"2021-11-04T08:51:00+01:00","linkquality":255,"state":"ON"}'
```

```
2021-11-04 08:51:00 ERROR (MainThread) [homeassistant.util.logging] Exception in state_received when handling msg on 'zigbee2mqtt/Lichterkette': '{"brightness":254,"color":{"b":254,"h":131,"s":100},"effect":null,"last_seen":"2021-11-04T08:51:00+01:00","linkquality":255,"state":"ON"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 37, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 318, in state_received
    self._update_color(values)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 257, in _update_color
    color_mode = values["color_mode"]
KeyError: 'color_mode'
```

Koenkk, 2021-11-04:
Try setting the color once

stefangries, 2021-11-04:
> Try setting the color once

That fixed the problem indeed, thank you!

Koenkk, 2021-11-04:
Great!

closed 2021-11-04 by Koenkk