Koenkk/zigbee2mqtt #9092: Large delay (minutes) xiaomi door sensors (MCCGQ11LM)

jeroenvdpoll, 2021-10-09:
## Background
Hello, 

- I use several xiaomi sensors in combination with zigbee2mqt (and home assistant) for over a year with no issues. 
- Suddenly (cannot find a reason, like an update or something) my door sensors (total 4) respond really slow (around 5 minutes respond time).
- All other (xiaomi) zigbee2mqtt devices do work normally (motion sensors, buttons, smoke sensors).
- After around 5 minutes the states do change.
- Rebooting system, rejoining devices, joining a fresh out of the box door sensor all didn't help
- Looks related to only the door sensors

I really have no clue where to look or what to do. Rejoining the door sensors didn't do anything.

Can anybody help?

Regards,
Jeroen

## What happened
When I open or close a door I cannot see any state change in:
1. MQTT
2. Zigbee2mqtt web dashboard
3. Home Assistant

After around 3 - 6 minutes the states do change

## What did you expect to happen
See state changes immediately instead of minutes

## How to reproduce it (minimal and precise)
See above. 

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC2652R
Adapter firmware version: unsure

jeroenvdpoll, 2021-10-09:
> Sounds like #8663

Hi Koen, I did check the kernel: I have kernel 5.4.0-87-generic so that shouldn't be affected. Also doesn't explain why it is only happening to the doorsensors and the issue popped up without any updated on the underlying (ubuntu) OS.

Any other idea's what to check?

Koenkk, 2021-10-09:
5.4.0-87 is affected:

> Ubuntu Focal: 5.4.0-85 and greater (not fixed yet as of 5.4.0-89)

Soildk, 2021-10-10:
Thanks.. I tried to follow the guide on https://github.com/daverobertson63/CH341SER, but I'm failing on several steps with my Ubuntu server. I'm a complete Linux noob, so I don't have any idea what to do.

bachoo786, 2021-10-15:
> > Sounds like #8663
> 
> Hi Koen, I did check the kernel: I have kernel 5.4.0-87-generic so that shouldn't be affected. Also doesn't explain why it is only happening to the doorsensors and the issue popped up without any updated on the underlying (ubuntu) OS.
> 
> Any other idea's what to check?

I am in the same boat as you @jeroenvdpoll I am running Ubuntu 20.04.3 LTS with the kernel 5.4.0-88-generic. All my xiaomi devices are slow in responding to z2m. 

I tried updating the CH431 driver per @Koenkk's recommendation but it is not straightforward. I also tried to update my kernel to version which is not affected but I cannot as there is no clear guidance on how to do it.

Its really frustrating as I am thinking of reinstalling everything from scratch.

Koenkk, 2021-10-15:
I will close this, it's not something that is caused by z2m and the issue + workaround has already been covered in https://github.com/Koenkk/zigbee2mqtt/issues/8663 . @daverobertson63 could you maybe make some tutorial on how to use your drivers?

jeroenvdpoll, 2021-10-16:
@daverobertson63. I realize that my comment was directed to @bachoo786 and not you. Mistake from my side.