Koenkk/zigbee2mqtt #9099: Trying to add support for Mercator ikuü globe - S9E27LED9W-RGB-Z

Vertikar, 2021-10-10:
Hi all,

Trying to add support for a Mercator ikuü globe, have followed the Tuya guide, but can only see the link quality being exposed. It seems to be a TS0505B, so in theory it just needs the manufacturer name added to tuya.js, but can't do this as easily with docker. 

Here's a snippet of what I'm seeing with debug logging enabled:

```
debug 2021-10-10 11:42:13: Received Zigbee message from '0x60a423fffe6b3478', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":153,"currentHue":0,"currentSatur
ation":254,"currentX":0,"currentY":0}' from endpoint 1 with groupID 0
debug 2021-10-10 11:42:13: No converter available for 'S9E27LED9W-RGB-Z' with cluster 'lightingColorCtrl' and type 'attributeReport' and data '{"colorTemperature":153,"currentHue":0,"curren
tSaturation":254,"currentX":0,"currentY":0}'
info  2021-10-10 11:42:13: MQTT publish: topic 'zigbee2mqtt/0x60a423fffe6b3478', payload '{"last_seen":"2021-10-10T11:42:13+11:00","linkquality":15}'
debug 2021-10-10 11:42:14: Received Zigbee message from '0x60a423fffe6b3478', type 'attributeReport', cluster 'genBasic', data '{"appVersion":82}' from endpoint 1 with groupID 0
info  2021-10-10 11:42:14: MQTT publish: topic 'zigbee2mqtt/0x60a423fffe6b3478', payload '{"last_seen":"2021-10-10T11:42:14+11:00","linkquality":15}'
debug 2021-10-10 11:42:14: Received Zigbee message from '0x60a423fffe6b3478', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2021-10-10 11:42:14: No converter available for 'S9E27LED9W-RGB-Z' with cluster 'genOnOff' and type 'attributeReport' and data '{"onOff":1}'
info  2021-10-10 11:42:14: MQTT publish: topic 'zigbee2mqtt/0x60a423fffe6b3478', payload '{"last_seen":"2021-10-10T11:42:14+11:00","linkquality":21}'
```

## Information about the device + link

https://www.ikuu.com.au/product/classic-colour-globe-e27-rgb-cct-2/

modelID: 'TS0505B',
manufacturerName: '_TZ3000_x2fqbdun'
model: 'S9E27LED9W-RGB-Z',
vendor: 'Mercator ikuü',
description: 'Classic Colour Globe E27 RGB + CCT',

## data/database.db entry of the device

```
{
  "id": 27,
  "type": "Router",
  "ieeeAddr": "0x60a423fffe6b3478",
  "nwkAddr": 7653,
  "manufId": 4098,
  "manufName": "_TZ3000_x2fqbdun",
  "powerSource": "Mains (single phase)",
  "modelId": "TS0505B",
  "epList": [
    1,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 269,
      "inClusterList": [
        0,
        3,
        4,
        5,
        6,
        4096,
        8,
        768
      ],
      "outClusterList": [
        25,
        10
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "TS0505B",
            "manufacturerName": "_TZ3000_x2fqbdun",
            "powerSource": 1,
            "zclVersion": 3,
            "appVersion": 82,
            "stackVersion": 0,
            "hwVersion": 1,
            "dateCode": ""
          }
        },
        "genLevelCtrl": {
          "attributes": {
            "currentLevel": 254
          }
        },
        "genOnOff": {
          "attributes": {
            "onOff": 1
          }
        },
        "lightingColorCtrl": {
          "attributes": {
            "currentHue": 0,
            "currentSaturation": 254,
            "currentX": 0,
            "currentY": 0,
            "colorTemperature": 153,
            "colorCapabilities": 25,
            "colorTempPhysicalMin": 153,
            "colorTempPhysicalMax": 500
          }
        },
        "genPowerCfg": {
          "attributes": {}
        },
        "genDeviceTempCfg": {
          "attributes": {}
        },
        "genOnOffSwitchCfg": {
          "attributes": {}
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b00216000c8",
          "endpointID": 1
        },
        {
          "cluster": 8,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b00216000c8",
          "endpointID": 1
        },
        {
          "cluster": 768,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b00216000c8",
          "endpointID": 1
        },
        {
          "cluster": 0,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b00216000c8",
          "endpointID": 1
        }
      ],
      "configuredReportings": [
        {
          "cluster": 6,
          "attrId": 0,
          "minRepIntval": 0,
          "maxRepIntval": 300,
          "repChange": 0
        },
        {
          "cluster": 8,
          "attrId": 0,
          "minRepIntval": 3,
          "maxRepIntval": 300,
          "repChange": 1
        }
      ],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 82,
  "stackVersion": 0,
  "hwVersion": 1,
  "dateCode": "",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {
    "reporting": 1,
    "configured": 2
  },
  "lastSeen": 1633824357724
}
```

External converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0505B',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_x2fqbdun'
        },
    ],
    model: 'S9E27LED9W-RGB-Z',
    vendor: 'Mercator ikuü',
    description: 'Classic Colour Globe E27 RGB + CCT',
    extend: extend.light_onoff_brightness_colortemp_color(),
    meta: {applyRedFix: true, enhancedHue: false},
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.setTime, // Add this if you are getting no converter for 'commandSetTimeRequest'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
```

Vertikar, 2021-10-15:
Any suggestions on how to sort this out? More logs I need to collect or anything?

Koenkk, 2021-10-16:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-10-16 by Koenkk