Koenkk/zigbee2mqtt #9107: homecloud (YUYA) temperature sensor TY-SZT not supported in zigbee2mqtt

raf610, 2021-10-10:
I have Zigbee2mqtt on raspberry pi zero with CC2531 coordinator.
I bought the nice temperature sensor T&H TY-SZT (model n. TT001-ZA) but I can't pair it in any way on zigbee2mqtt.

it looks like zigbee doesn't see it despite trying to reset it and remove the battery.

Has anyone solved the issue?

thank you

raf610, 2021-10-16:
no it is not!
I can't find this device in the list of supported devices yet

[image]

janpieper, 2021-10-17:
Yeah, the picture in the zigbee2mqtt documentation looks wrong. Mine looks exactly like yours.

---

[image]

---

[image]

raf610, 2021-10-17:
So do you think that to solve the problem I have to replace CC2531 with CC2652P?
It seems strange to me that when I try to pair the devices, Zigbee2mqtt doesn't see anything

closed 2021-11-25 by github-actions[bot]