Koenkk/zigbee2mqtt #9114: NEO-PD07 datapoint 113 & type 'commandSetTimeRequest' not parsed

Masterz69, 2021-10-11:
## What happened
Recently added device Neo Coolcam NAS-PD07 working fine, but some errors appears:

```
log_2021-10-03.13-05-18_1.txt:warn  2021-10-11 06:11:48: zigbee-herdsman-converters:NEO-PD07: NOT RECOGNIZED DP #113 with data {"status":0,"transid":0,"dp":113,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[2]}}
log_2021-10-03.13-05-18_1.txt:warn  2021-10-11 06:11:48: zigbee-herdsman-converters:NEO-PD07: NOT RECOGNIZED DP #113 with data {"status":0,"transid":0,"dp":113,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[2]}}
log_2021-10-03.13-05-18_1.txt:warn  2021-10-11 06:11:48: zigbee-herdsman-converters:NEO-PD07: NOT RECOGNIZED DP #113 with data {"status":0,"transid":0,"dp":113,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[4]}}
log_2021-10-03.13-05-18_1.txt:warn  2021-10-11 06:11:54: zigbee-herdsman-converters:NEO-PD07: NOT RECOGNIZED DP #113 with data {"status":0,"transid":0,"dp":113,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[2]}}
log_2021-10-03.13-05-18_1.txt:warn  2021-10-11 06:11:59: zigbee-herdsman-converters:NEO-PD07: NOT RECOGNIZED DP #113 with data {"status":0,"transid":0,"dp":113,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[2]}}
log_2021-10-03.13-05-18_1.txt:warn  2021-10-11 06:12:05: zigbee-herdsman-converters:NEO-PD07: NOT RECOGNIZED DP #113 with data {"status":0,"transid":0,"dp":113,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[4]}}
```
^^ see at least two values for 'data'. But have no idea about meaning. 
#7915 also have no description for DP 113.

As well time to time appears:
```
log_2021-10-03.13-05-18_2.txt:debug 2021-10-09 16:39:29: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":29441}'
log_2021-10-03.13-05-18_2.txt:debug 2021-10-09 17:40:14: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":29697}'
log_2021-10-03.13-05-18_2.txt:debug 2021-10-09 18:04:49: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":768}'
log_2021-10-03.13-05-18_2.txt:debug 2021-10-09 19:04:05: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":1280}'
log_2021-10-03.13-05-18_2.txt:debug 2021-10-09 20:04:50: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":1536}'
...
log_2021-10-03.13-05-18_1.txt:debug 2021-10-10 12:18:13: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":5632}'
log_2021-10-03.13-05-18_1.txt:debug 2021-10-10 13:17:34: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":5888}'
log_2021-10-03.13-05-18_1.txt:debug 2021-10-10 14:18:18: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":6144}'
log_2021-10-03.13-05-18_1.txt:debug 2021-10-10 15:19:03: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":6400}'
```
payloadSize value increase +256 in all cases, except transition over 0, looks like.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.21.2-1

danieledwardgeorgehitchcock, 2021-10-11:
Hi,

Regarding the `commandSetTimeRequest` error, have you seen the `onEvent` section [here](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html#2-adding-your-device)?

Regarding data values, these are of `Enum` data type which is usually some sort of setting button i.e. language that will be translated from a fixed numeric value to a textual value.

It is a lot easier to debug the devices with the TuYa gateway / app first to determine what is being communicated to / from the device, and then implementing the same in Zigbee2MQTT - is that a possibility for you?

Masterz69, 2021-10-11:
Moved to https://github.com/Koenkk/zigbee-herdsman-converters/issues/3152