Koenkk/zigbee2mqtt #9130: Binding Xiaomi QBKG12LM to INNR RS 228 T "Nothing to bind"

EmmanuelLM, 2021-10-12:
## What happened

1. Xiaomi QBKG12LM is known as "Kitchen switch 2" and INNR RS 228 T is "Nook light 1" 
2. decoupled the left switch (Xiaomi) via the interface -> successful
3. Trying to bind the switch to the lightbulb with mqtt Payload {"from":"Kitchen switch 2/left","to":"Nook light 1"}

## What did you expect to happen

Expecting the switch to bind to, instead getting:

MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"from":"Kitchen switch 2","to":"Nook light 1"},"type":"device_bind_failed"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":[],"from":"Kitchen switch 2/left","to":"Nook light 1"},"error":"Nothing to bind","status":"error"}'
Error Nothing to bind 

## How to reproduce it (minimal and precise)
As above

## Debug info
Zigbee2MQTT version:  1.21.2-1 
Adapter hardware: ZZH
Adapter firmware version: latest as at today

Koenkk, 2021-10-12:
Indeed a limitation of the device, it is not capable of sending genOnOff commands to devices.

ars4l4n, 2023-09-23:
> Indeed a limitation of the device, it is not capable of sending genOnOff commands to devices.

does that mean they can't be bound? I have the same error but with the Xiaomi Mija Smart Button and a Philips Hue White Bulb E27

Koenkk, 2023-09-23:
> > Indeed a limitation of the device, it is not capable of sending genOnOff commands to devices.
> 
> does that mean they can't be bound? I have the same error but with the Xiaomi Mija Smart Button and a Philips Hue White Bulb E27

yes