Koenkk/zigbee2mqtt #9136: Error in zigbe2mqtt log " Failed to configure "

Palthe, 2021-10-13:
I have the following error in the log of zigbee2mqtt:

Zigbee2MQTT:error 2021-10-13 02:08:07: Failed to configure 'Light switch - 3de floor - hallway stairs', attempt 1 (Error: Bind 0x00158d0005cfeab4/2 genOnOff from '0x00124b0023932de1/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))
Zigbee2MQTT:error 2021-10-13 02:43:58: Failed to configure 'Light switch - ground floor - stairs', attempt 1 (Error: Bind 0x00158d0005cfe9b5/2 genPowerCfg from '0x00124b0023932de1/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))

The devices what are giving the problem are Xiaomi "Aqara D1 single key wireless wall switches"

I do not know if it is lated but I receive the following error message repeatively in my supervisor log. The ip address of the docker container of zigbee2mqtt is 172.30.33.3. 

21-10-13 02:02:41 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.3:8099 ssl:default [Connect call failed ('172.30.33.3', 8099)]

## Debug info
Zigbee2MQTT version: 1.21.2-1
Adapter hardware: CC2652RB zigbee2mqtt stick 
Adapter firmware version: do not know, cannot find it, it appears not to be in the database file

Palthe, 2021-10-14:
Thank you for you response. 

When pressing the reconfigure button on the fronted, before clicking for action the following error is given. The units do appear to work. I also noticed that all Xiaomi batterij powered devices have the same problem and "fload" my logs with errors. I andded the database file [database.log](https://github.com/Koenkk/zigbee2mqtt/files/7343795/database.log). 

Any other ideas?

Zigbee2MQTT:error 2021-10-14 07:52:22: Failed to configure 'Environmental sensor - ground floor - living room', attempt 2 (Error: Bind 0x54ef441000106676/1 genPowerCfg from '0x00124b0023932de1/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))
(node:329) UnhandledPromiseRejectionWarning: Error: Write 0x00158d00
0456c75d/1 genBasic({"65520":{"value":[170,16,5,65,71,1,1,16,1],"type":65}}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:390:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:329) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:329) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

Palthe, 2021-10-15:
Yes, you are correct. I am seeing now errors with all battery devices. I tried to do the same with this device. 

The D1 switch gives the same error after I tried to re-configure:

Zigbee2MQTT:error 2021-10-15 06:43:49: Failed to configure 'Light switch - 3de floor - hallway stairs', attempt 3 (Error: Bind 0x00158d0005cfeab4/2 genPowerCfg from '0x00124b0023932de1/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))

A few times per day the connection drops and none of my devices work. I suspect that this has to do with the following error: 

21-10-13 02:02:41 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.3:8099 ssl:default [Connect call failed ('172.30.33.3', 8099).

172.30.33.3:8099 is the docker of M2mqtt

lawful76, 2024-10-15:
I have the same issue with smart light switch [](https://www.zigbee2mqtt.io/devices/TS0011.html).
The switch works but there're errors after every restart.

[23:32:38] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-10-15 23:32:44] error: 	zh:zstack:znp: Failed to determine if path is valid: 'Error: spawn udevadm ENOENT'
[2024-10-15 23:33:08] error: 	z2m: Failed to configure '0xa4c1388b4bd24903', attempt 1 (Error: Bind 0xa4c1388b4bd24903/1 genOnOff from '0x00124b0024c8c177/0' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:5965:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:205:21)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[2024-10-15 23:36:29] error: 	z2m: Failed to configure '0xa4c1388b4bd24903', attempt 2 (Error: Bind 0xa4c1388b4bd24903/1 genOnOff from '0x00124b0024c8c177/0' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:5965:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:205:21)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[2024-10-15 23:36:56] error: 	z2m: Failed to configure '0xa4c1388b4bd24903', attempt 3 (Error: Bind 0xa4c1388b4bd24903/1 genOnOff from '0x00124b0024c8c177/0' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:5965:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:205:21)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))