Koenkk/zigbee2mqtt #9165: Weirdness with user extension events being called twice

sjorge, 2021-10-15:
## What happened
I've updated my restore state extension published here: https://github.com/Koenkk/zigbee2mqtt/discussions/7642#discussion-3388806 to work with the changes required on the dev branch.

After some initial struggling I managed to get it to work again (yay!), the code is also a bit cleaner than before I think.

However all actions are now done twice!

```
Zigbee2MQTT:debug 2021-10-15 18:14:07: Device 'light/office/desk/bulb' announced itself
Zigbee2MQTT:info  2021-10-15 18:14:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"light/office/desk/bulb","ieee_address":"0xec1bbdfffe4318dc"},"type":"device_announce"}'
Zigbee2MQTT:debug 2021-10-15 18:14:07: Applying initial state '{"state":"ON"}' to 'light/office/desk/bulb' (0xec1bbdfffe4318dc)
Zigbee2MQTT:info  2021-10-15 18:14:07: MQTT publish: topic 'zigbee2mqtt/light/office/desk/bulb/set', payload '{"state":"ON"}'
Zigbee2MQTT:debug 2021-10-15 18:14:07: Applying initial state '{"state":"ON"}' to 'light/office/desk/bulb' (0xec1bbdfffe4318dc)
Zigbee2MQTT:info  2021-10-15 18:14:07: MQTT publish: topic 'zigbee2mqtt/light/office/desk/bulb/set', payload '{"state":"ON"}'
Zigbee2MQTT:debug 2021-10-15 18:14:07: Received MQTT message on 'zigbee2mqtt/light/office/desk/bulb/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2021-10-15 18:14:07: Publishing 'set' 'state' to 'light/office/desk/bulb'
Zigbee2MQTT:debug 2021-10-15 18:14:07: Received MQTT message on 'zigbee2mqtt/light/office/desk/bulb/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2021-10-15 18:14:07: Publishing 'set' 'state' to 'light/office/desk/bulb'
Zigbee2MQTT:debug 2021-10-15 18:14:08: Received Zigbee message from 'light/office/desk/bulb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
```

As you see the bulb announce itself, this results in the event being published over mqtt. (So far so good), we can see that the onDeviceAnnounce function from the extension also gets called based on the logging:
```
Zigbee2MQTT:debug 2021-10-15 18:14:07: Applying initial state '{"state":"ON"}' to 'light/office/desk/bulb' (0xec1bbdfffe4318dc)
```

The problem is, it gets called twice! This should not be happening, I double checked but I don't think I am registering the event handler twice so I'm confused.

## What did you expect to happen
The event to be called only once.

## How to reproduce it (minimal and precise)
Loading the user extension and configure a bulb to have a `restore_state` device settings block, then unplug/replug the bulb.

## Debug info
Zigbee2MQTT version:     1.21.2-dev commit: 35719370
Adapter hardware: zzhp-lite
Adapter firmware version: 20210708

sjorge, 2021-10-16:
No change, still triggeres twice:
```
Zigbee2MQTT:debug 2021-10-16 11:14:12: Device 'light/office/desk/bulb' announced itself
Zigbee2MQTT:info  2021-10-16 11:14:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"light/office/desk/bulb","ieee_address":"0xec1bbdfffe4318dc"},"type":"device_announce"}'
Zigbee2MQTT:debug 2021-10-16 11:14:13: Applying initial state '{"state":"ON"}' to 'light/office/desk/bulb' (0xec1bbdfffe4318dc)
Zigbee2MQTT:info  2021-10-16 11:14:13: MQTT publish: topic 'zigbee2mqtt/light/office/desk/bulb/set', payload '{"state":"ON"}'
Zigbee2MQTT:debug 2021-10-16 11:14:13: Applying initial state '{"state":"ON"}' to 'light/office/desk/bulb' (0xec1bbdfffe4318dc)
Zigbee2MQTT:info  2021-10-16 11:14:13: MQTT publish: topic 'zigbee2mqtt/light/office/desk/bulb/set', payload '{"state":"ON"}'
```

Koenkk, 2021-10-17:
What if you change:

```js
// attach to events
this.eventBus.on('deviceAnnounce', this.onDeviceAnnounce.bind(this), this.constructor.name);
```

to 

```js
// attach to events
this.onDeviceAnnounce = this.onDeviceAnnounce.bind(this);
this.eventBus.on('deviceAnnounce', this.onDeviceAnnounce, this.constructor.name);
```

sjorge, 2021-10-17:
I also noticed that start() seems to be called twice!
```
Zigbee2MQTT:info  2021-10-17 15:04:15: Starting device state restore
Zigbee2MQTT:info  2021-10-17 15:04:15: Starting device state restore
```

As this also shows up in the log.

closed 2021-10-17 by sjorge