Koenkk/zigbee2mqtt #9172: Zigbee2mqtt - Errors when sending commands to devices

johannesbec, 2021-10-16:
## What happened
Hi guys,

I have a network of ~40 devices running with a cc2531 adapter for more than 2 years. 
A week ago the stick (?) crashed and I replaced it with a slae cc2652 usb stick, due to many reports about cc2531 crashing and having issues.

what I did before changing to the cc2652:
I updated the version from 1.18 to 1.21.1 
I flashed the cc2652 with the current coordinator firmware dated 20210708
Replaced the serial port in the config
I started zigbee2mqtt - whenever I send a command, I got the following error:

Error Publish 'set' 'state' to '0x000d6ffffec92d64' failed: 'Error: Command 0x000d6ffffec92d64/1 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

What I tried to solve this issue then:
- I removed the coordinator_backup.json and re-flashed the stick (taking into account to repair all my devices)
- changed pan_id and network-security-key
- upgraded from 1.21.1 to 1.21.2 
- and repeated to remove coordinator__backup and reflashing several times

Current behaviour: 
 zigbee2mqtt starts, I can pair devices. However, as soon as I want to switch a bulb on/off (or sending any command), above error is reported

Error Publish 'set' 'state' to '0x000d6ffffec92d64' failed: 'Error: Command 0x000d6ffffec92d64/1 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

## What did you expect to happen

Zigbee2mqtt accepting commands and the bulb to react accordingly

## How to reproduce it (minimal and precise)

start zigbee2mqtt with slae cc2652

## Debug info
Zigbee2MQTT version: 1.21.2
Adapter hardware: CC2652RB
Adapter firmware version: 20210708_coordinator (Koenkk)

Logs:
https://pastebin.com/RX5JyRGt

wimb0, 2021-11-04:
I'm having the same issue. Z2M only works correctly for a short while.
Using a slae.sh cc2652rb stick with firmware 20210708
My OS is Ubuntu 20.04 with Zigbee2mqtt running in docker.

beng90, 2021-11-12:
I have the same problem on CC2652P, before on CC2531.
Communication works only a while. When I try to switch device around 10 times right away HA get a lag and wait 6 seconds. After that it sometimes execute last command. Hour later it is worst and it hardly works at all.

I hope that someone resolved this issue.
I got a tip that kernel upgrade may fix it but it seems to be really difficult in HA distribution.

Koenkk, 2021-11-14:
Could you check wether this still happens with the following firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

beng90, 2021-11-18:
> Could you check wether this still happens with the following firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

Unfortunately it doesn't help. I tried this with temperature sensor and wall switch. Initially everything works fine but after hour light state doesn't refresh in HA, temperature as well. Next day last update of temperature sensor is day before.
I've tried this coordinator in MacOS using VirtualBox and there is no problem. The problem is in Raspberry Pi 3b on Hassio OS.

Koenkk, 2021-11-19:
@beng90 if the problem only occurs on the pi it could be a power issue, make sure that:
- the cc2652p is connected through a usb extension cable (to prevent interference)
- you use a good power supply
- any other usb devices (especially have stuff like ssds) are connected via a powered usb hub

codingcatgirl, 2021-11-24:
I'm sorry to say that this problem is also occuring to me. Just like @johannesbc, i replaced a CC2531 with a CC2552P, but i'm now getting the same error after running the stick for a some time. Today it happened even just a few hours in. This has happened multiple times, and can – unsurprisingly, i guess – be resolved by unplugging and replugging the stick. I have reinstalled my entire zigbee2mqtt instance, which did not help.

I'm using an Raspberry Pi 4 with an original Raspberry Pi power supply and the Zigbee USB stick is the only USB device connected. I will try using a usb extension cable to see if it helps, but i'm not optimistic.

beng90, 2021-12-25:
> @beng90 if the problem only occurs on the pi it could be a power issue, make sure that:
> 
> * the cc2652p is connected through a usb extension cable (to prevent interference)
> * you use a good power supply
> * any other usb devices (especially heavy stuff like ssds) are connected via a powered usb hub

I have checked this solutions excluding third point instead this I used different disk case and it helped for few days. Disconnecting was much rare than before, it worked around two days and then i must restart coordinator. However it doesn't fix my problem. 
Now I've changed hardware to Dell wyse and there is no problems. I couldn't fix it on Rpi.

wimb0, 2022-01-06:
Somehow the issue was gone for a while, but it seems to have returned after updating to 1.22.2 and firmware 20211217.
Rebooting and unplug the stick only works sometimes.

closed 2022-02-14 by github-actions[bot]