Koenkk/zigbee2mqtt #9175: IKEA TRÅDFRI E1745 not detected

Z3NOX, 2021-10-16:
## What happened
I bought a new IKEA TRÅDFRI motion sensor (E1745), which should be supported due to [the list of supported devices](https://www.zigbee2mqtt.io/devices/E1525_E1745.html). However, when I want to pair it it is shown as unsupported device. I'm a bit confused here. Am I doing something wrong or is the problem based on a newer model/firmware of the device that is not detected yet? 

```
Zigbee2MQTT:debug 2021-10-16 19:54:50: Received Zigbee message from 'motion_sensor', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":537011747,"imageType":4552,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2021-10-16 19:54:50: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'IKEA of Sweden'
Zigbee2MQTT:warn  2021-10-16 19:54:50: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

## What did you expect to happen
The device should be detected correctly as it seems to be supported.

## How to reproduce it (minimal and precise)
Just pair the device by pressing the link button 4 times. (Have done this a couple of times)

## Debug info
Zigbee2MQTT version: 1.21.2
Adapter hardware: CC26X2R1 (slaesh)
Adapter firmware version: 20210708

zsbagyura, 2021-11-21:
I have the same issue with E1745, I cannot use it. The log is full with messages like this:
Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'IKEA of Sweden' 

How can I help to solve this issue?

wilcomir, 2021-11-22:
Hey @zsbagyura for me it eventually worked - restarting zigbee2mqtt helped; also try to leave the sensor without batteries for a prolonged period of time so that it "resets".
I have no idea of what is going wrong, but with the suggestions above and some trial and error I managed to connect the sensors in the end.

zsbagyura, 2021-11-22:
Thanks for the help. Indeed the reboot of zigbee2mqtt and the battery trick solved the issue.

Z3NOX, 2021-11-22:
Wow, this also helped on my site. Eventually I had to try this multiple times until it worked.
Also be aware: You have to take out both batteries in order for the sensor to reset. As they are wired in parallel, the sensor will still be functional with only one battery…

In the debug log the device joined and left a view times before the interview was successful:
```
Nov 22 10:12:01 :info  2021-11-22 10:12:01: Device '0xb4e3f9fffeb302e0' joined                                                                                                                                     
Nov 22 10:12:01 :info  2021-11-22 10:12:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeb302e0","ieee_address":"0xb4e3f9fffeb302e0"},"type":"device_joined"}'
Nov 22 10:12:01 :info  2021-11-22 10:12:01: Starting interview of '0xb4e3f9fffeb302e0'                                                                                                                             Nov 22 10:12:01 :info  2021-11-22 10:12:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeb302e0","ieee_address":"0xb4e3f9fffeb302e0","status":"started"},"type":"device_interview"}'                                                                                                                                                                                                Nov 22 10:12:01 :info  2021-11-22 10:12:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xb4e3f9fffeb302e0"},"type":"device_connected"}'
 Nov 22 10:12:01 :info  2021-11-22 10:12:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xb4e3f9fffeb302e0"},"type":"pairing"}'
Nov 22 10:12:03 :debug 2021-11-22 10:12:03: Device '0xb4e3f9fffeb302e0' announced itself
Nov 22 10:12:03 :info  2021-11-22 10:12:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeb302e0","ieee_address":"0xb4e3f9fffeb302e0"},"type":"device_announce"}' Nov 22 10:12:03 :info  2021-11-22 10:12:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xb4e3f9fffeb302e0"},"type":"device_announced"}'
Nov 22 10:12:11 :warn  2021-11-22 10:12:11: Device '0xb4e3f9fffeb302e0' left the network
Nov 22 10:12:11 :info  2021-11-22 10:12:11: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeb302e0","ieee_address":"0xb4e3f9fffeb302e0"},"type":"device_leave"}'    Nov 22 10:12:11 :info  2021-11-22 10:12:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xb4e3f9fffeb302e0"},"type":"device_removed"}'
Nov 22 10:12:43 :info  2021-11-22 10:12:43: Device '0xb4e3f9fffeb302e0' joined                                                                                                                                     
Nov 22 10:12:43 :info  2021-11-22 10:12:43: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeb302e0","ieee_address":"0xb4e3f9fffeb302e0"},"type":"device_joined"}'
Nov 22 10:12:43 :info  2021-11-22 10:12:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xb4e3f9fffeb302e0"},"type":"device_connected"}'
Nov 22 10:12:45 :debug 2021-11-22 10:12:45: Device '0xb4e3f9fffeb302e0' announced itself
Nov 22 10:12:45 :info  2021-11-22 10:12:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeb302e0","ieee_address":"0xb4e3f9fffeb302e0"},"type":"device_announce"}'
Nov 22 10:12:45 :info  2021-11-22 10:12:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xb4e3f9fffeb302e0"},"type":"device_announced"}'
Nov 22 10:12:56 :warn  2021-11-22 10:12:56: Device '0xb4e3f9fffeb302e0' left the network
Nov 22 10:12:56 :info  2021-11-22 10:12:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeb302e0","ieee_address":"0xb4e3f9fffeb302e0"},"type":"device_leave"}'
Nov 22 10:12:56 :info  2021-11-22 10:12:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xb4e3f9fffeb302e0"},"type":"device_removed"}'
Nov 22 10:13:24 :info  2021-11-22 10:13:24: Device '0xb4e3f9fffeb302e0' joined
Nov 22 10:13:24 :info  2021-11-22 10:13:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeb302e0","ieee_address":"0xb4e3f9fffeb302e0"},"type":"device_joined"}'
Nov 22 10:13:24 :info  2021-11-22 10:13:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xb4e3f9fffeb302e0"},"type":"device_connected"}'
Nov 22 10:13:27 :debug 2021-11-22 10:13:27: Device '0xb4e3f9fffeb302e0' announced itself
Nov 22 10:13:27 :info  2021-11-22 10:13:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeb302e0","ieee_address":"0xb4e3f9fffeb302e0"},"type":"device_announce"}'
Nov 22 10:13:27 :info  2021-11-22 10:13:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xb4e3f9fffeb302e0"},"type":"device_announced"}'
[... cut ...] ame":"0xb4e3f9fffeb302e0","ieee_address":"0xb4e3f9fffeb302e0","status":"successful","supported":true},"type":"device_interview"}'
Nov 22 10:13:46 :info  2021-11-22 10:13:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload {"message":"interview_successful","meta":{"description":"TRADFRI motion sensor","friendly_name":"0xb4e3f9fffeb302e0","model":"E1525/E1745","supported":true,"vendor":"IKEA"},"type":"pairing"}'
Nov 22 10:14:15 :info  2021-11-22 10:14:15: Configuring '0xb4e3f9fffeb302e0'
```

closed 2021-11-22 by Z3NOX

juanpescador, 2022-01-16:
I was having issues too, what worked for me was to cause the motion sensor to send a Zigbee message by physically moving the sensor (i.e. triggering motion). Similar to how some battery powered switches need clicking during pairing.

The tricky part is that the motion sensor only detects motion every once in a while, a minute and a half I think.

dlavrantonis, 2022-07-23:
why is this closed? This is far from a proper solution. I'm having exactly the same issues and I'm trying to pair 10 sensors of this model that I bought after seeing that they are compatible. Whether the sensors will pair or not is something completely random and far from straightforward. If theres no smooth way to pair them, i believe they shall be marked as incompatible - if you have lots of sensors to pair it will take you hours

wilcomir, 2022-09-05:
Hello @dlavrantonis - not sure if this is still helpful for you. I ran in some kind of an issue re pairing a couple of devices this evening, and I think that @juanpescador is right.

Try the following:

- start from a dead device - remove batteries for at least 60 s or so
- make sure that joining your zigbee network is allowed
- insert the batteries with the device facing down, ideally on something soft such a sofa - the device should not spot you
- press the connect button four times
- wait for a notification to show up in the logs that the joining is in progress
- point the device towards you so that it sees you and sends a message on the network

This proved to be surprisingly reliable to me. If you still have problems and are willing to test this out, if this works for you as well I can submit a PR to update the documentation with these additional notes.

dlavrantonis, 2022-09-06:
hello @wilcomir, unfortunately I have returned the sensors so I have no way to test this at the moment. If someone has verified this, such a note will be very helpful for potential buyers

richi115, 2022-10-05:
I bought an E1745 a few days ago and the only way to pair it was using the procedure @wilcomir described before, you should add it to the devices notes.

FrAllard, 2022-11-09:
> Hello @dlavrantonis - not sure if this is still helpful for you. I ran in some kind of an issue re pairing a couple of devices this evening, and I think that @juanpescador is right.
>
> Try the [...]

Thank you very much! I just forked the official document to try to have your solution added to the document. I added a last step, is to keep the device pointed at you and keeping moving to keep the device active sending commands. It takes less time to fully configure this way. From my experience.