Koenkk/zigbee2mqtt #9176: Mark group as unavailable when all members are unavailable

definitio, 2021-10-16:
## Is your feature request related to a problem? Please describe
I have a group of light bulbs. It stays on when bulbs turned off by switch.

## Describe the solution you'd like
I want availability support for groups.

eloo, 2021-10-29:
I second that request.
It seems with the currently setup is easy to setup a proper behaviour of a group of lights. 

E.g. when the lights are switched off the entity in homeassistant and also the last message in mqtt is "State: On".
Nevertheless that all members are offline. Further the last_seen is also not propagated to the group.

So it would be really nice if we could get an attribute which matches the real state of the group. 

Thanks

Lerbaek, 2021-11-19:
I third this.
While it could be easily worked around by just creating logical groups in Home Assistant instead, I find that using zigbee2mqtt groups is the only way to have the bulbs act simultaneously, and so, for those of us with physical wall switches, this is actually quite the issue.

eloo, 2021-12-20:
AFAIK this is still open

eloo, 2022-01-20:
not stale

eloo, 2022-02-20:
not stale

deviantintegral, 2022-02-23:
I was able to get close to this, in that the Sengled bulbs I have can report when they lose power. I set up the following reporting on each bulb, which causes the group to turn off and reset itself when they are turned back on.

Unfortunately, the group still stays available, and HA acts as if commands are successful. With ZHA, the group becomes marked unavailable (after 10 minutes?) so you know you can't send it commands.

While I have most of my physical switches set up with guards, so they are never off, I do have one guest bedroom where I want the physical switch to work so guests can't possibly be woken by a misclick.

[image]

Koenkk, 2022-03-14:
This has now been implemented ([docs](https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/guide/configuration/device-availability.md#groups))

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-14 by definitio

eloo, 2022-03-15:
@Koenkk seems to work as expected 👍 
thanks

PR #11626 (2022-03-14): Availability for groups
Fixes #9176

TODO:
- [x] Test
- [x] Wait for next z2m release (1 March)