Koenkk/zigbee2mqtt #9185: Zigbee2MQTT Version 1.21.2-dev & Homeassistant Generic Thermostat & Hama 00176593 & Zigbee2Mqtt do not work - need help!

zibous, 2021-10-17:
## What happened
The control takes place via Lovelace thermostat. Inital, when I restart home assistant, then the temperature is set for all thermostats via the script. 

**Ater a certain Time, however, this is always reset the `current_heating_setpoint` to 16 degrees ???**

```
init_climate_thermostat:
  alias: Heizungsthermostate einstellen
  sequence:
    - service: mqtt.publish
      data_template:
        topic: conbee2mqtt/thermostat-bad/set/current_heating_setpoint
        payload_template: "{{ states('input_number.thermostat_bad') | float }}"
    - service: mqtt.publish
      data_template:
        topic: conbee2mqtt/thermostat-office/set/current_heating_setpoint
        payload_template: "{{ states('input_number.thermostat_office') | float }}"
    - service: mqtt.publish
      data_template:
        topic: conbee2mqtt/thermostat-wc/set/current_heating_setpoint
        payload_template: "{{ states('input_number.thermostat_wc') | float }}"
    - service: mqtt.publish
      data_template:
        topic: conbee2mqtt/thermostat-wohnzimmer1/set/current_heating_setpoint
        payload_template: "{{ states('input_number.thermostat_wohnzimmer1') | float }}"
    - service: mqtt.publish
      data_template:
        topic: conbee2mqtt/thermostat-wohnzimmer2/set/current_heating_setpoint
        payload_template: "{{ states('input_number.thermostat_wohnzimmer2') | float }}"
    - service: mqtt.publish
      data_template:
        topic: conbee2mqtt/thermostat-kitchen/set/current_heating_setpoint
        payload_template: "{{ states('input_number.thermostat_kitchen') | float }}"
    - service: mqtt.publish
      data_template:
        topic: conbee2mqtt/thermostat-kindenzimmer1/set/current_heating_setpoint
        payload_template: "{{ states('input_number.thermostat_kindenzimmer1') | float }}"
    - service: mqtt.publish
      data_template:
        topic: conbee2mqtt/thermostat-kinderzimmer2/set/current_heating_setpoint
        payload_template: "{{ states('input_number.thermostat_kindenzimmer2') | float }}"
    - service: mqtt.publish
      data_template:
        topic: conbee2mqtt/thermostat-schlafzimmer/set/current_heating_setpoint
        payload_template: "{{ states('input_number.thermostat_schlafzimmer') | float }}"
    - service: mqtt.publish
      data_template:
        topic: conbee2mqtt/thermostat-fitnessraum/set/current_heating_setpoint
        payload_template: "{{ states('input_number.thermostat_fitnessraum') | float }}"

```

## What did you expect to happen
**Is it due to the Zigbee2MQTT or the Homeassistant configuration?** 

```- platform: "mqtt"
  name: Thermostat Bad
  temperature_unit: "C"
  initial: 25.00
  temp_step: 0.5
  min_temp: "5"
  max_temp: "30"
  json_attributes_topic: conbee2mqtt/thermostat-bad
  temperature_command_topic: "conbee2mqtt/thermostat-bad/set/current_heating_setpoint"
  availability_topic: "conbee2mqtt/bridge/state"
  current_temperature_topic: "conbee2mqtt/thermostat-bad"
  current_temperature_template: "{{ value_json.local_temperature }}"
  temperature_state_topic: "conbee2mqtt/thermostat-bad"
  temperature_state_template: "{{ value_json.current_heating_setpoint }}"
  mode_state_topic: "conbee2mqtt/thermostat-bad"
  mode_state_template: "{{ value_json.system_mode }}"
  mode_command_topic: "conbee2mqtt/thermostat-bad/set/system_mode"
  modes:
    - "off"
    - "heat"
    - "auto"
  action_topic: "conbee2mqtt/thermostat-bad/set/system_mode"
  action_template: "{% set values = {'idle':'off','heat':'heating','auto':'auto'} %}{{ values[value_json.running_state] }}"
  away_mode_command_topic: "conbee2mqtt/thermostat-bad/set/value_json.away_mode"
  away_mode_state_topic: "conbee2mqtt/thermostat-bad"
  away_mode_state_template: "{{ value_json.away_mode }}"
  payload_on: "true"
  payload_off: "false"
  retain: true

```

## How to reproduce it (minimal and precise)
[image]

[image]

[image]

## Debug info
Zigbee2MQTT version: Version 1.21.2-dev
Adapter hardware: Conbee II
Adapter firmware version: latest

```
{"anti_scaling":"ON",
  "away_mode":"OFF",
  "battery_low":false,
  "child_lock":"UNLOCK",
  "current_heating_setpoint":16,
  "device":{
    "applicationVersion":85,
    "dateCode":"",
    "friendlyName":"thermostat-wohnzimmer2",
     "hardwareVersion":1,
     "ieeeAddr":"0x847127fffe0954af",
     "manufacturerID":4098,
      "manufacturerName":"_TZE200_yw7cahqs",
      "model":"SEA801-Zigbee/SEA802-Zigbee",
      "networkAddress":57759,
      "powerSource":"Battery",
      "stackVersion":0,
      "type":"EndDevice",
      "zclVersion":3
     },
    "elapsed":234,
    "frost_detection":"ON",
    "last_seen":"2021-10-17T08:51:28+02:00",
    "linkquality":215,
    "local_temperature":20.8,
    "local_temperature_calibration":0,
    "preset_mode":"none",
    "system_mode":"auto",
    "window_detection":"ON"
}
```

zibous, 2021-10-31:
After a long search I found the information in the Hama manual. it looks like this If the "AUTO" mode is selected, the thermostat is set to the "Outward" mode!

```
modes:
    - "off"
    - "heat"
    - "auto"
```

** Modes `auto` = `Outward mode`  !!!

> If this mode is active, the set temperature is automatically dialled down to 16°**

machv, 2021-11-18:
I think removing the schedule stored in the TRV from factory following [this](https://github.com/Koenkk/zigbee2mqtt/issues/7080#issuecomment-857700403) might also help as by default, in `auto` mode it would fall back to that 16°C setting at specified times.

zibous, 2021-11-19:
@machv , @Koenkk 

Thanks for this information 👍 ,  exactly what I've been missing so far. But unfortunately I don't manage to apply this. If I transfer the settings via the mqtt command, it works without errors, but nothing changes.

**What am I doing wrong ?** 

```json
{"availability":[
    {"topic":"conbee2mqtt/bridge/state"},
    {"topic":"conbee2mqtt/thermostat-office/availability"}
  ],
  "availability_mode":"all",
  "away_mode_command_topic":"conbee2mqtt/thermostat-office/set/away_mode",
  "away_mode_state_template":"{{ value_json.away_mode }}",
  "away_mode_state_topic":"conbee2mqtt/thermostat-office",
  "current_temperature_template":"{{ value_json.local_temperature }}",
  "current_temperature_topic":"conbee2mqtt/thermostat-office",
  "device":{"identifiers":["zigbee2mqtt_0x5c0272fffe8568d1"],
  "manufacturer":"Saswell",
  "model":"Thermostatic radiator valve (SEA801-Zigbee/SEA802-Zigbee)","name":"thermostat-office"},
  "json_attributes_topic":"conbee2mqtt/thermostat-office",
  "max_temp":"30",
  "min_temp":"5",
  "mode_command_topic":"conbee2mqtt/thermostat-office/set/system_mode",
  "mode_state_template":"{{ value_json.system_mode }}",
  "mode_state_topic":"conbee2mqtt/thermostat-office","modes":["off","heat","auto"],
  "name":"thermostat-office",
  "temp_step":0.5,
  "temperature_command_topic":"conbee2mqtt/thermostat-office/set/current_heating_setpoint",
  "temperature_state_template":"{{ value_json.current_heating_setpoint }}",
  "temperature_state_topic":"conbee2mqtt/thermostat-office",
  "temperature_unit":"C",
  "unique_id":"0x5c0272fffe8568d1_climate_conbee2mqtt"
  }
```

`transitionTime` - Is it minutes since midnight?
`dayofweek` - first day is Sunday, right?
`mode`- 3 "auto"

**Sample set weekly schedule for monday**

- 00:00 - 05:00 19.00 °C
- 05:00 - 09:00 20.00 °C
- 17:00 - 00:00 25:00 °C

mqtt: `conbee2mqtt/thermostat-office/set`
Payload:
```json
{
  "weekly_schedule": {
    "monday": {
      "dayofweek": 2,
      "mode": 3,
      "numoftrans": 4,
      "transitions": [
        {
          "heatSetpoint": "19.00",
          "transitionTime": 0
        },
        {
          "heatSetpoint": "25.00",
          "transitionTime": 300
        },
        {
          "heatSetpoint": "20.00",
          "transitionTime": 540
        },
        {
          "heatSetpoint": "25.00",
          "transitionTime": 1020
        }
      ]
    }
    
  }
}
```

The query always returns an error:
mqtt: `conbee2mqtt/thermostat-office/get`

```
payload: {"weekly_schedule": ""}
---> ERROR:
Publish 'set' 'weekly_schedule' to 'thermostat-office' failed: 'Error: Invalid mode: 3 for device thermostat-office'
No converter available for 'get' 'weekly_schedule' ()
```

I can't see any change in the dashboard. 
[image]

If I make a mistake, how can I clear the  weekly_schedule settings? 

Thanks for your time and help.

closed 2021-12-28 by github-actions[bot]

gomble, 2022-01-01:
> @machv , @Koenkk
>
> Thanks for this information 👍 , exactly what I've been missing so far. But unfortunately I don't manage to apply this. If I transfer the settings via the mqtt command, it works [...]

did you find a solution ?