Koenkk/zigbee2mqtt #9187: Blitzwolf SHP-13 not pairing - unsupported device

hrosak88, 2021-10-17:
Hi! I have several zigbee plugs blitzwolf shp-13. All worked well, but now some of them stopped working. I cannot pair them again, it shows unsupported device :-(

Version: zigbee2mqtt stable 1.21.2-1
Coordinator: slaesh CC2652RB

Thank you for your help.

closed 2021-11-25 by github-actions[bot]

Nenodema, 2022-03-04:
Have you fixed your issue? I’m encountering the same issue.

hrosak88, 2022-03-04:
> Have you fixed your issue? I’m encountering the same issue.
> 
> I have created a new issue for this #11703

Yes, you must update coordinator to newest firmware. Works like a charm 🙂

Nenodema, 2022-03-04:
> > Have you fixed your issue? I’m encountering the same issue.
> > I have created a new issue for this #11703
> 
> Yes, you must update coordinator to newest firmware. Works like a charm 🙂

Thanks for your reply. Please check my post, it seems that i have a new version of the SHP-13 which is currently not supported.