Koenkk/zigbee2mqtt #9189: add product ME81H.31 ZIGBEE

PHIL21120, 2021-10-17:
Hello, would it be possible to add this product? Regards

https://fr.aliexpress.com/item/1005002432774642.html?spm=a2g0s.9042311.0.0.27426c37meQz6U
[image]
[image]

..

{"id":37,"type":"Router","ieeeAddr":"0xcc86ecfffef896e3","nwkAddr":42247,"manufId":4098,"manufName":"_TZE200_ye5jkfsb","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005���(\u0012���(\u0012","65506":31,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE200_ye5jkfsb","powerSource":1,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1634404082118} 
..

PHIL21120, 2021-11-09:
Bonjour, Une petite relance!!

tolwi, 2021-11-23:
```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzDataPoints = {
    avattoSwitch: 1,
    avattoMode: 2,
    avattoTempSet: 16,
    avattoTempCurrent: 24,
    avattoChildLock: 40,
    avattoTempCorrection: 27,
    avattoSensor: 43,
};

const tzConverters = {
    avatto_thermostat_child_lock: {
        key: ['child_lock'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, fzDataPoints.avattoChildLock, value === 'lock');
        },
    },
    avatto_thermostat_preset: {
        key: ['preset'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointEnum(entity, fzDataPoints.avattoMode, value === 'manual' ? 0 : 1);
        },
    },
    avatto_thermostat_sensor: {
        key: ['sensor'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointEnum(entity, fzDataPoints.avattoSensor, value === 'in' ? 0 : 1);
        },
    },
    avatto_thermostat_mode: {
        key: ['system_mode'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointEnum(entity, fzDataPoints.avattoSwitch, value === 'heat');
        },
    },

    avatto_thermostat_temp_correction: {
        key: ['local_temperature_calibration'],
        convertSet: async (entity, key, value, meta) => {
            const temp = Math.round(value);
            await tuya.sendDataPointValue(entity, fzDataPoints.avattoTempCorrection, temp);
        },
    },
    avatto_thermostat_temp_set: {
        key: ['current_heating_setpoint'],
        convertSet: async (entity, key, value, meta) => {
            const temp = Math.round(value);
            await tuya.sendDataPointValue(entity, fzDataPoints.avattoTempSet, temp);
        },
    },

};

const fzConverters = {
    avatto_floor_thermostat: {
        cluster: 'manuSpecificTuya',
        type: ['commandGetData', 'commandSetDataResponse'],
        convert: (model, msg, publish, options, meta) => {
            const dp = msg.data.dp;
            const value = tuya.getDataValue(msg.data.datatype, msg.data.data);
            switch (dp) {
            case fzDataPoints.avattoSwitch:
                return {system_mode: value ? 'heat' : 'off'};
            case fzDataPoints.avattoChildLock: // mode
                return {child_lock: value ? 'lock' : 'unlock'};
            case fzDataPoints.avattoTempSet:
                return {current_heating_setpoint: value};
            case fzDataPoints.avattoTempCurrent:
                return {local_temperature: value};
            case fzDataPoints.avattoTempCorrection:
                return {local_temperature_calibration: value};
            case fzDataPoints.avattoMode:
                return {preset: value === 0 ? 'manual' : 'program'};
            case fzDataPoints.avattoSensor:
                return {sensor: value === 0 ? 'in' : 'out'};
            default:
                meta.logger.warn(`zigbee-herdsman-converters:AVATTO: Unrecognized DP #${dp} with data ${JSON.stringify(msg.data)}`);
            }
        },
    },
};

module.exports = [
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ye5jkfsb'}],
        model: 'ME81AH',
        vendor: 'TuYa',
        description: 'Wall-mount thermostat',
        whiteLabel: [
            {vendor: 'AVATTO', model: 'ME81AH'},
        ],
        fromZigbee: [fzConverters.avatto_floor_thermostat],
        toZigbee: [
            tzConverters.avatto_thermostat_child_lock,
            tzConverters.avatto_thermostat_preset,
            tzConverters.avatto_thermostat_mode,
            tzConverters.avatto_thermostat_temp_set,
            tzConverters.avatto_thermostat_temp_correction,
            tzConverters.avatto_thermostat_sensor,
        ],
        exposes: [
            e.child_lock(),
            e.local_temperature(),
            exposes.climate()
                .withSetpoint('current_heating_setpoint', 5, 30, 1, ea.STATE_SET)
                .withLocalTemperatureCalibration(ea.STATE_SET)
                .withSystemMode(['off', 'heat'])
                .withPreset(['manual', 'program'])
                .withSensor(['in', 'out'], ea.STATE_SET),
        ],
        onEvent: tuya.onEventSetLocalTime,
    },
];

```

tmaniac, 2021-11-23:
I've received this thermostat from AliExpress as well.
Allthought the manufacturer is different i have added the device to Zigbee2MQTT.

as far as i can see it works; still having a few issues though;
`Zigbee2MQTT:error 2021-11-23 21:58:17: No converter available for 'local_temperature' ()
Zigbee2MQTT:error 2021-11-23 21:58:24: No converter available for 'get' 'system_mode' ()
Zigbee2MQTT:error 2021-11-23 21:59:07: No converter available for 'local_temperature' ()
Zigbee2MQTT:error 2021-11-23 21:59:11: No converter available for 'local_temperature' ()
Zigbee2MQTT:error 2021-11-23 22:04:30: No converter available for 'get' 'system_mode' ()
Zigbee2MQTT:warn  2021-11-23 22:06:40: zigbee-herdsman-converters:ali: Unrecognized DP #108 with data {"status":8,"transid":0,"dp":108,"datatype":0,"fn":0,"data":{"type":"Buffer","data":[6,0,20,8,0,15,11,30,15,12,30,15,17,0,22,22,0,15,8,0,22,23,0,15]}}
Zigbee2MQTT:warn  2021-11-23 22:14:10: zigbee-herdsman-converters:ali: Unrecognized DP #108 with data {"status":8,"transid":0,"dp":108,"datatype":0,"fn":0,"data":{"type":"Buffer","data":[6,0,20,8,0,15,11,30,15,12,30,15,17,0,22,22,0,15,8,0,22,23,0,15]}}
Zigbee2MQTT:warn  2021-11-23 22:14:28: zigbee-herdsman-converters:ali: Unrecognized DP #36 with data {"status":8,"transid":0,"dp":36,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[1]}}
Zigbee2MQTT:warn  2021-11-23 22:14:28: zigbee-herdsman-converters:ali: Unrecognized DP #36 with data {"status":8,"transid":0,"dp":36,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[1]}}
Zigbee2MQTT:warn  2021-11-23 22:14:28: zigbee-herdsman-converters:ali: Unrecognized DP #45 with data {"status":8,"transid":0,"dp":45,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}}
Zigbee2MQTT:warn  2021-11-23 22:16:30: zigbee-herdsman-converters:ali: Unrecognized DP #104 with data {"status":8,"transid":0,"dp":104,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[3]}}
Zigbee2MQTT:warn  2021-11-23 22:16:30: zigbee-herdsman-converters:ali: Unrecognized DP #19 with data {"status":8,"transid":0,"dp":19,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,40]}}
Zigbee2MQTT:warn  2021-11-23 22:16:30: zigbee-herdsman-converters:ali: Unrecognized DP #26 with data {"status":8,"transid":0,"dp":26,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,10]}}
Zigbee2MQTT:warn  2021-11-23 22:16:30: zigbee-herdsman-converters:ali: Unrecognized DP #106 with data {"status":8,"transid":0,"dp":106,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,1]}}
Zigbee2MQTT:warn  2021-11-23 22:16:30: zigbee-herdsman-converters:ali: Unrecognized DP #103 with data {"status":8,"transid":0,"dp":103,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[1]}}
Zigbee2MQTT:warn  2021-11-23 22:18:43: zigbee-herdsman-converters:ali: Unrecognized DP #108 with data {"status":8,"transid":0,"dp":108,"datatype":0,"fn":0,"data":{"type":"Buffer","data":[6,30,21,8,0,15,11,30,15,12,30,15,17,0,21,22,0,15,8,0,22,23,0,15]}}
Zigbee2MQTT:warn  2021-11-23 22:19:06: zigbee-herdsman-converters:ali: Unrecognized DP #104 with data {"status":0,"transid":1,"dp":104,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[1]}}
Zigbee2MQTT:warn  2021-11-23 22:19:06: zigbee-herdsman-converters:ali: Unrecognized DP #19 with data {"status":0,"transid":1,"dp":19,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,40]}}
Zigbee2MQTT:warn  2021-11-23 22:19:06: zigbee-herdsman-converters:ali: Unrecognized DP #26 with data {"status":0,"transid":1,"dp":26,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,10]}}
Zigbee2MQTT:warn  2021-11-23 22:19:06: zigbee-herdsman-converters:ali: Unrecognized DP #106 with data {"status":0,"transid":1,"dp":106,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,1]}}
Zigbee2MQTT:warn  2021-11-23 22:19:06: zigbee-herdsman-converters:ali: Unrecognized DP #103 with data {"status":0,"transid":1,"dp":103,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[1]}}
Zigbee2MQTT:warn  2021-11-23 22:19:57: zigbee-herdsman-converters:ali: Unrecognized DP #108 with data {"status":0,"transid":1,"dp":108,"datatype":0,"fn":0,"data":{"type":"Buffer","data":[6,30,21,8,0,15,11,30,15,12,30,15,17,0,21,22,0,15,8,0,21,22,0,15]}}`

i'm guessing i'm missing some datapoints or something.

closed 2021-12-31 by github-actions[bot]