Koenkk/zigbee2mqtt #9197: Eurotronic Spirit zigbee thermostat - missing entity?

bence111, 2021-10-18:
### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/9195

<div type='discussions-op-text'>

<sup>Originally posted by **bence111** October 18, 2021</sup>
Hello,

I'm using an Eurotronic Spirit thermostat integrated into (dev type SPZB0001, zigbee_manufacturer is Eurotronic).
Some parameters (battery, eurotronic_valve_position, eurotronic_trv_mode and climate (including current temp, heating setpoing etc) are published to Mosquitto MQTT server and appearing in the Home assistant enity list.
There is one parameter (what I really need): **pi_heating_demand** (the current position of the valve), which is **not appearing in the entity list** therefore I can not check/display it in HA.
When browsing with MQTT explorer, it seems that is listed in homeassistant/sensor/DEVNAME topic, but no entity is created for it.
Mosquitto discovering is enabled.
Mosquitto version 6.0.1
Zigbee2MQtt version 1.21.2-1
HA version 2021.10.5
I already tried couple of things, e.g.:
-remove the device, add it again
-remove, then stop mosquitto, restart HA, delete old entities and join the device again

Is this pi_heating_demand is working for someone as entity?

How can I add MQTT value manually to entities?

Or what can be the problem with this eurotronic device in zigbee2mqtt?</div>

Morphy99, 2021-11-22:
Answered in the discussion, please close this issue.

Koenkk, 2021-11-22:
With https://github.com/Koenkk/zigbee2mqtt/pull/9759 `pi_heating_demand` is now exposed to HA. Changes will be available in the next release on 1 December.

closed 2021-11-22 by Koenkk